Koenkk/zigbee2mqtt #24047: [New device support]: HZC T11_ZG / Namron 4512783

ohagane, 2024-09-21:
### Link

https://www.hzc-electric.com/pro_view-119.html

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0x385cfbfffec805fe","nwkAddr":45190,"manufId":4714,"manufName":"Namron AS","powerSource":"Unknown","modelId":"4512783","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,6,513,516,1029,1794,2820,4096],"outClusterList":[3,25,1030],"clusters":{"hvacUserInterfaceCfg":{"attributes":{"tempDisplayMode":0}},"genOnOff":{"attributes":{"onOff":1}},"hvacThermostat":{"attributes":{"32768":1,"32770":0,"32772":0,"32773":60,"32774":0,"32778":1,"32787":500,"32797":0,"32799":0,"32800":10957,"32801":10957,"32802":1,"32803":0,"32804":0,"pICoolingDemand":0,"pIHeatingDemand":0,"runningState":0,"systemMode":4,"programingOperMode":1,"occupiedHeatingSetpoint":1850,"localTemp":2350}},"msRelativeHumidity":{"attributes":{"measuredValue":5900}},"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":1,"acPowerMultiplier":1,"acCurrentDivisor":1,"acCurrentMultiplier":1,"rmsCurrent":0,"activePower":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xbc026efffefe0022","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0xbc026efffefe0022","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20240117","swBuildId":"1.11","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1726917047875}

### Zigbee2MQTT version

1.40.1

### Comments

Other links:
https://www.elektroimportoren.no/namron-zigbee-edge-termostat-hvit/4512783/Product.html
https://www.elektroimportoren.no/namron//4512783-84-Zigbee-User-Guide.pdf

### External definition

```shell
const {onOff, humidity, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['4512783'],
    model: '4512783',
    vendor: 'Namron AS',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), humidity(), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The only thing that works is On/Off and humidity. 
After adding external definition i get this error in the log.

z2m: Failed to configure '0x385cfbfffec7f1fd', attempt 3 (Error: ZCL command 0x385cfbfffec7f1fd/1 haElectricalMeasurement.read(["acVoltageDivisor","acVoltageMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:443:29) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1772:33) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:193:21) at Configure.configure (/app/lib/extension/configure.ts:125:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

stave1989, 2024-09-26:
I hope support for this thermostat comes fast 🥰

Teeseeone, 2024-10-03:
Just got myself two of this also, hoping for support 🤓👌

ohagane, 2024-10-05:
Have been playing around with the setup file and got some of the functions working.
Now working as a thermostat in HA.

Follow the guide for adding External converters.
Save this file as 4812783.js in your zigbee2mqtt folder and add the external converter to your config file.
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-1-extending-the-external-definition

[4512783.txt](https://github.com/user-attachments/files/17265349/4512783.txt)

stave1989, 2024-10-05:
Thank you for the work. I can see that current and energy sensor does not work yet. And in HA it say off when not idling. 
Hope you share the external converter again if you tweek even more 😊

stave1989, 2024-10-06:
A correction, it is rapporting energy, but only in whole numbers, 12,13,14 nothing in between.

ohagane, 2024-10-06:
[4512783.txt](https://github.com/user-attachments/files/17269425/4512783.txt)

Energy is reporting KWH every 5 min, according to config.

Have made some updates to the config and it should now read states from thermostat at startup.
But sill only have basic functions of the thermostat, since I'm not able to read the "Private Thermostat" cluster.

I'm not a programmer, so just copy and pasting from other configs.
But trying to get my thermostats to work and decided to share my findings.

Remember to hit Reconfigure on the device after changing the config.

stave1989, 2024-10-06:
Really appreciate your work. see it reports energy but is there any way to get it to show more decimals? now it only shows integers.
[image]

stave1989, 2024-10-06:
I thought the same, but had to ask. strange that it only reports whole numbers when there are several decimal places on the thermostat.

Teeseeone, 2024-11-15:
I tried to add the converter, but can you control the thermostat, change temp etc?

erazor666, 2024-11-30:
Just started to look into this device after I purchased 3 of them. Not gotten very much done, this file however can be useful to someone digging into this device possibly:

Masssssy, 2024-12-10:
Also have one of these. I'm a complete zigbee n00b however but might try to look into this. 

The configs posted in the thread seems to work decently with basic funtionality dialed in somewhat but getting quite a bit of red warning/error popups. Also e.g voltage is missing.

Teeseeone, 2025-01-09:
How long will it take to get this device to be offically supported?

bskjon, 2025-01-15:
> How long will it take to get this device to be offically supported?

You could give it a try yourself 😜 
This device is pure pain. 
From what i see the behaviour is all over the place, and its the first time I'm actually getting somewhere, thanks to the schema.
Since i haven't seen anyone opening a PR/MR on it, i started myself.
[image]
Current progress.
Screen timeout is not in the schema either, so I'm playing around with my device. Rip powerbill

bskjon, 2025-01-16:
[image]
Hmm, mine does show decimals now 🤔

stave1989, 2025-01-17:
@bskjon 
Can you send me the external converter you currently are using? :)

erazor666, 2025-01-17:
> [@bskjon](https://github.com/bskjon) Can you send me the external converter you currently are using? :)

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512784'],
    model: '4512784',
    vendor: 'Namron',
    description: 'Edge Thermostat',
    extend: [

    ],
    fromZigbee: [
        fz.humidity,
        fz.thermostat,
        fz.hvac_user_interface,
        fz.metering, 
        fz.electrical_measurement,
        fz.namron_hvac_user_interface,
    ], 
    toZigbee: [
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_local_temperature_calibration,
        tz.thermostat_system_mode,
        tz.thermostat_temperature_display_mode,
        tz.thermostat_programming_operation_mode,
        tz.namron_thermostat_child_lock,
    ], 
    exposes: [
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 35, 0.5)
            .withLocalTemperature()
            .withLocalTemperatureCalibration(-3, 3, 0.1)
            .withSystemMode(['off', 'cool', 'heat'])
            .withRunningState(['cool', 'heat', 'idle']), 
        e.humidity(),
        e.enum('temperature_display_mode', ea.ALL, ['celsius', 'fahrenheit'])
            .withDescription('The temperature format displayed on the thermostat screen'),
        e.binary('child_lock', ea.ALL, 'LOCK', 'UNLOCK')
            .withDescription('Enables/disables physical input on the device'),
        e.enum('programming_operation_mode', ea.ALL, ['setpoint', 'schedule', 'schedule_with_preheat', 'eco'])
            .withDescription('Controls how programming affects the thermostat. Possible values: setpoint (only use specified setpoint), schedule (follow programmed setpoint schedule), schedule_with_preheat (follow programmed setpoint schedule with pre-heating). Changing this value does not clear programmed schedules.'),
        e.power(),
        e.current(),
        e.energy(),
        ], 

    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = [
            'genBasic',
            'genIdentify',
            'hvacThermostat',
            'seMetering',
            'haElectricalMeasurement',
            'genAlarms',
            'msOccupancySensing',
            'genTime',
            'hvacUserInterfaceCfg',
        ];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        // standard ZCL attributes
        await reporting.thermostatTemperature(endpoint, {min: 0, change: 50});
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatKeypadLockMode(endpoint);

        //Metering
        await endpoint.read('haElectricalMeasurement', ['acCurrentDivisor', 'rmsCurrent', 'activePower']);
        await endpoint.read('seMetering', ['currentSummDelivered']);
        await reporting.rmsCurrent(endpoint, {min: 10, change: 10}); // A - z2m displays only the first decimals, so change of 10 (0,01)
        await reporting.activePower(endpoint, {min: 10, change: 15}); // W - Min change of 1,5W
        await reporting.currentSummDelivered(endpoint, {min: 300}); // Report KWH every 5min
        await reporting.readMeteringMultiplierDivisor(endpoint);

        // Trigger initial read
        await endpoint.read('hvacThermostat', ['systemMode', 'runningState', 'occupiedHeatingSetpoint', 'programingOperMode', 'localTemperatureCalibration']);
        await endpoint.read('hvacUserInterfaceCfg', ['tempDisplayMode', 'keypadLockout']);
        

    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;

bskjon, 2025-01-17:
> @bskjon 
> Can you send me the external converter you currently are using? :) 
> 
> 

Will publish it on a gist when I get off work

ohagane, 2025-01-17:
Have been working on a new config that have most of the things. Have not been able to get some of the Regulator addresses to work. 

0x8007  Regulator                                INT8U   0x00
0x801C  Regulation_mode                 ENUM8   0x00
Does not give any usefull information.

Please try this and report back.
Time sync should also work, and syncs every 24 hours.

The items under # Unknown stuff can be activated if needed.

[NamronEdge.txt](https://github.com/user-attachments/files/18456345/NamronEdge.txt)

bskjon, 2025-01-17:
> [NamronEdge.txt](https://github.com/user-attachments/files/18456887/NamronEdge.txt)
> 
> Have made a new config that uses most of the values coming from this device. Some attributes does not give any useful information. Try it and see if there is anything missing.
> 
> Under Exposes there is a few things slashed out, but can be activated if needed.

Funny enough, 0-2/3 is manual on mine, while the rest to 5 is eco mode 🙃 (programming_operation_mode)

magnus-siv, 2025-01-17:
@bskjon I'm trying to reconfigure my thermostat with your code and z2m logs just throws this error. 

```

Show onlyalldebuginfowarningerror

Filter by text

Log levelerrorwarninginfodebugwarn

Logging level

Logs limit1002005001000

 

info 2025-01-17 18:33:24z2m: Configuring 'Termostat'

info 2025-01-17 18:33:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Termostat', payload '{"current":3,"energy":0,"humidity":32,"linkquality":140,"power":757,"state":"ON","voltage":null}'

error 2025-01-17 18:33:26z2m: Failed to configure 'Termostat', attempt 2 (Error: ZCL command 0x385cfbfffec80846/1 haElectricalMeasurement.read(["acVoltageDivisor","acVoltageMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at runNextTicks (node:internal/process/task_queues:69:3) at processImmediate (node:internal/timers:459:9) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:1924:37) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))

```

I've put the NamronEdge.js in the z2m folder and added it under external converters in the configuration.yaml. Then I restarted z2m, but no dice.

Have I done something wrong somewhere?

bskjon, 2025-01-17:
> [@bskjon](https://github.com/bskjon) I'm trying to reconfigure my thermostat with your code and z2m logs just throws this error.
>
> ```
>
>
> Show onlyalldebuginfowarningerror
>
> Filter by text
>
> Log [...]

Well, yes and no.
From i believe z2m 2.x.x, there are no configuration.yaml changes.
Simply in the same data directory, create a folder called: "external_converters"
Put the file there, and proceed to restart or reload z2m

ohagane, 2025-01-17:
> Funny enough, 0-2/3 is manual on mine, while the rest to 5 is eco mode 🙃 (programming_operation_mode)

Yes I see the same issue. Not sure how to get around it. No matter what I try. I cant get both Manual and Program to work correctly. Eco mode always seems to work. ;)

bskjon, 2025-01-17:
> Yes I see the same issue. Not sure how to get around it. No matter what I try. I cant get both Manual and Program to work correctly. Eco mode always seems to work. ;)

I see that you have made some nice work with the time n stuff.
I'll "merge" some of the stuff you made into mine, test it a bit, and then probably add it to the converters if you dont beat me to it

ohagane, 2025-01-18:
> > > > [NamronEdge.txt](https://github.com/user-attachments/files/18456887/NamronEdge.txt)
> > > > Have made a new config that uses most of the values coming from this device. Some attributes does not give any useful information. Try it and see if there is anything missing.
> > > > Under Exposes there is a few things slashed out, but can be activated if needed.
> > > 
> > > 
> > > Funny enough, 0-2/3 is manual on mine, while the rest to 5 is eco mode 🙃 (programming_operation_mode)
> > 
> > 
> > Yes I see the same issue. Not sure how to get around it. No matter what I try. I cant get both Manual and Program to work correctly. Eco mode always seems to work. ;)
> 
> I see that you have made some nice work with the time n stuff. I'll "merge" some of the stuff you made into mine, test it a bit, and then probably add it to the converters if you dont beat me to it
> 
> Edit: One thing I would love to get done, is merging the other modes into one using enums, so it doesnt become a mess in homeassistant
> 
> Edit 2: Strangely enough, setting or changing the current_operating_mode (0x0025) requires BITMAP8 🙃
> 
> Edit 3: system_mode on mine returns "1" which is the default, but not present in the zigbee table for what it corresponds to, so i set mine to "auto"

I had the same idea. To have all the modes under one. But found it hard since holiday and boost have prerequisites for starting. So even if you are able to trigger Boost it will only run for 5 min(default). And Holiday mode need dates to be set first.
But you are probably better at programming, cause I have spent hours looking at different configs, and just tried a bunch of shit ;)

bskjon, 2025-01-19:
> I had the same idea. To have all the modes under one. But found it hard since holiday and boost have prerequisites for starting. So even if you are able to trigger Boost it will only run for 5 [...]

Tried a bunch of shit myself ;P 
Some stuff never triggered on my device to logging 🙃 
This is the stuff I've set towards mine.
https://gist.github.com/bskjon/f12069aa4fec4633527342b44a8eba03
Consulted the manual for fault status, hope the value is correct though.
Dropped the regulator stuff and sensor mode, cuz they did never do the thing consistency..

ohagane, 2025-01-23:
Very nice work, but missing Humidity......

bskjon, 2025-01-23:
> Very nice work, but missing Humidity......

Do you mind going over this and see if I'm missing anything? https://github.com/Koenkk/zigbee-herdsman-converters/pull/8650

zarilion, 2025-01-23:
Git/zigbee2mqtt noob here. 

When it's added to the herdsman converters, will it then be automatically discovered in zigbee2mqtt, and I can just delete the external converters file I added? Or will I still have to install the converter manually?

bskjon, 2025-01-23:
> Git/zigbee2mqtt noob here. 
> 
> When it's added to the herdsman converters, will it then be automatically discovered in zigbee2mqtt, and I can just delete the external converters file I added? Or will I still have to install the converter manually?

If or when this is accepted, yes you don't need the external converter. Ofc you will need to restart z2m to unload the external one.
If there is any functionality missing, comment on the Mr and I might be able to add it.

bskjon, 2025-01-23:
> Very nice work, but missing Humidity......

I see that I forgot it in the PR as well. Will add it

zarilion, 2025-01-23:
Great! Thanks for the awesome work you guys have put in. Working great here, but missing the humidity as the author pointed out.

bskjon, 2025-01-23:
> Great! Thanks for the awesome work you guys have put in. Working great here, but missing the humidity as the author pointed out.

Just re-added it, to the PR, not going to bother with the local converter for now.

monsivar, 2025-01-24:
Used the Namrong_Edge.js file. Am i missing out on something here? Got it working in Z2M, but in HA the thermostat is not working: 

[image]

bskjon, 2025-01-24:
> Used the Namrong_Edge.js file. Am i missing out on something here? Got it working in Z2M, but in HA the thermostat is not working:
> 
> [image]

If you used mine, then yes, i forgot to add the state 🙃

monsivar, 2025-01-24:
> 
> If you used mine, then yes, i forgot to add the state 🙃

Yes, it was yours. How do i add state in te file?

monsivar, 2025-01-24:
> While you wait for a new z2m version, you can add this to the "fromZigbee": fz.thermosta
Almost.... 
First tried fz.thermostat and no change. Then  fz.hvac_user_interface :

`const definition = {
    zigbeeModel: ["4512783"],
    model: "Edge Thermostat",
    vendor: "Namron",
    description: "Namron Zigbee Edge Termostat",
    fromZigbee: [fzLocal.namron_edge_thermostat, fz.hvac_user_interface, fz.namron_hvac_user_interface, fz.metering, fz.thermostat, fz.electrical_measurement],
    toZigbee: [
        tz.thermostat_local_temperature, 
        tzLocal.namron_edge_thermostat, 
        tz.thermostat_occupied_heating_setpoint, 
        tz.thermostat_unoccupied_heating_setpoint, 
        tz.namron_thermostat_child_lock, 
        tz.thermostat_control_sequence_of_operation,
        tz.thermostat_programming_operation_mode,
        tz.thermostat_temperature_display_mode,
    ],`

And got this:

[image]

bskjon, 2025-01-24:
> Almost....
> First tried fz.thermostat and no change. Then  fz.hvac_user_interface :
>
> `const definition = {
>     zigbeeModel: ["4512783"],
>     model: "Edge Thermostat",
>     vendor: "Namron", [...]

Hmm, even without it on mine I do have the incorrect off and heat, but not idle. Try to reconfigure it by clicking the yellow button I z2m.

monsivar, 2025-01-24:
> Hmm, even without it on mine I do have the incorrect off and heat, but not idle. Try to reconfigure it by clicking the yellow button I z2m.

That did not help.... Still the same

bskjon, 2025-01-24:
> > Hmm, even without it on mine I do have the incorrect off and heat, but not idle. Try to reconfigure it by clicking the yellow button I z2m.
> 
> That did not help.... Still the same

Ahh, sorry. I have like 3 or 4 different files, and apparently uploaded the oldes one to gist.
I've re-added humidity and added run mode

monsivar, 2025-01-24:
> Ahh, sorry. I have like 3 or 4 different files, and apparently uploaded the oldes one to gist.
> I've re-added humidity and added run mode

Still not working for me, but thanks for your time late on a friday evening :-)

bskjon, 2025-01-24:
> > Ahh, sorry. I have like 3 or 4 different files, and apparently uploaded the oldes one to gist.
> > I've re-added humidity and added run mode
> 
> Still not working for me, but thanks for your time late on a friday evening :-)

Ugh.. Thanks for pointing it out. I'm afraid this is also now in master 🙃 , will give you a heads up if i think i have fixed it

bskjon, 2025-01-24:
> > Ahh, sorry. I have like 3 or 4 different files, and apparently uploaded the oldes one to gist.
> > I've re-added humidity and added run mode
> 
> Still not working for me, but thanks for your time late on a friday evening :-)

https://gist.github.com/bskjon/545fdb80661c77371d29140623751e83

Try this. Seems like i forgot small stuff in the gist one. master should be fine :)
This one is straight from my hass

monsivar, 2025-01-24:
> Try this. Seems like i forgot small stuff in the gist one. master should be fine :)

Getting closer....

[image]

[image]

bskjon, 2025-01-24:
> > Try this. Seems like i forgot small stuff in the gist one. master should be fine :)
> 
> Getting closer....
> 
> Z2M `{ "active_display_brightness": 76, "anti_frost": "Off", "auto_time": "On", "boost_time_left": 0, "boost_time_set": "Off", "child_lock": "UNLOCK", "current": 0, "current_operating_mode": "Manual", "display_auto_off": "Off", "energy": 0.9, "fault": "No Fault", "holiday_temp_set_f": 50, "humidity": 43, "linkquality": 114, "local_temperature": 23, "pi_cooling_demand": 0, "pi_heating_demand": 0, "power": 0, "regulator_percentage": 0, "running_state": "idle", "sensor_mode": "Floor", "state": "OFF", "summer_winter_switch": "Off", "temperature_display_mode": "celsius", "time_sync": "On", "time_sync_value": 0, "vacation_end_date": "2000.01.01", "vacation_mode": "Off", "vacation_start_date": "2000.01.01", "window_open": "Closed", "window_open_check": "Off", "work_days": "Mon-Fri Sat-Sun", "holiday_temp_set": 500, "local_temperature_calibration": null, "action": "off" }`
> 
> HA
> 
> [image]
> 
> [image]
> 
> [image]

Hmmmm.
Strange.

Could you show me what your modes say in hass?
Also, something is borked here.... now mine wont enter eco.. 🙃

monsivar, 2025-01-24:
> Could you show me what your modes say in hass?

`hvac_modes:
  - "off"
  - cool
  - heat
min_temp: 15
max_temp: 35
target_temp_step: 0.5
current_temperature: 22.8
temperature: null
hvac_action: idle
friendly_name: Termostat bad kjeller
supported_features: 385`

bskjon, 2025-01-24:
> > Could you show me what your modes say in hass?
> 
> `hvac_modes:
> 
> * "off"
> * cool
> * heat
>   min_temp: 15
>   max_temp: 35
>   target_temp_step: 0.5
>   current_temperature: 22.8
>   temperature: null
>   hvac_action: idle
>   friendly_name: Termostat bad kjeller
>   supported_features: 385`

I've updated my gist (the last one i sendt). For now, it works in my hass, as well as eco mode.
If this solves it, i will make changes to the integration one, and open another pr

monsivar, 2025-01-25:
> If this solves it, i will make changes to the integration one, and open another pr

It works for mee to now :-) Thanks for the help!

seime, 2025-02-17:
Support added via https://github.com/Koenkk/zigbee-herdsman-converters/pull/8650, issue should be closed @ohagane ?

closed 2025-02-17 by ohagane