Koenkk/zigbee2mqtt #24049: TS0601 / _TZE204_ya4ft0w4 (Human Presence Sensor) - Invalid value for number *** move_sensitivity / presence_sensitivity: 0 (range 1.0 - 10.0)

fabianosan, 2024-09-21:
### What happened?

Home Assistant is displaying the errors below, informing that the entity values ​​must be within a range.

It turns out that when I set these values ​​to within the range, either in Zigbee2MQTT or in Home Assistant, it starts working again, but after a while it seems to lose these values, starts triggering the error again and I need to go there and set it again to the range between 1 and 10, I have to do this a few times a day.

**Device Info in Zigbee2mqtt:**
[image]
[image]

**Error in Home Assistant:**
[image]

### What did you expect to happen?

I would like the device to work as expected, not lose the values ​​of the move_sensitivity and presence_sensitivity entities

### How to reproduce it (minimal and precise)

Add the device into zigbee2mqtt, export to Home Assistant and after a few hours the errors will start to appear, probably because the values ​​of these entities have been lost.

### Zigbee2MQTT version

1.40.1-dev commit: 7ad51ce

### Adapter firmware version

7.4.3 [GA]

### Adapter

Sonoff Dongle-E with Ember firmware

### Setup

Beelink EQ12 with debian 12 and docker

### Debug log

This is a log from Home Assistant, not from Zigbee2mqtt:

[zb_ha_log.txt](https://github.com/user-attachments/files/17084048/zb_ha_log.txt)

fabianosan, 2024-09-27:
The presence device keeping lost move and presence values and throwing this error!

fear, 2024-09-29:
I have the same issue with 2 tuya devices: 

ZY-M100-24GV3 ZY-M100-24GV2

Koenkk, 2024-10-01:
Fixed! Will be available in todays release.

iamthekings5, 2024-10-07:
> Fixed! Will be available in todays release.

I still have this issue. On normal branch latest version.

fabianosan, 2024-10-07:
> > Fixed! Will be available in todays release.
> 
> I still have this issue. On normal branch latest version.

As device said, at this moment, is only available on dev branch.

You need to install dev release or wait until it is published as stable release.

iamthekings5, 2024-10-10:
> > > Fixed! Will be available in todays release.
> > 
> > I still have this issue. On normal branch latest version.
> 
> As device said, at this moment, is only available on dev branch.
> 
> You need to install dev release or wait until it is published as stable release.

I switched to dev release (edge) and the issue still occurs.
[image]
Periodically at some time every day my settings will reset and logs will get spammed.
[image]

iamthekings5, 2024-10-10:
> > > Fixed! Will be available in todays release.
> > 
> > I still have this issue. On normal branch latest version.
> 
> As device said, at this moment, is only available on dev branch.
> 
> You need to install dev release or wait until it is published as stable release.

I switched to dev (edge) and issue still occurs.

1.40.2-dev commit: 3bb4af2

Koenkk, 2024-10-12:
@iamthekings5 do you have a `_TZE204_ya4ft0w4`?

iamthekings5, 2024-10-12:
> @iamthekings5 do you have a `_TZE204_ya4ft0w4`?

@Koenkk  Mine is a `_TZE204_7gclukjs` recognised as a ZY-M100-24GV2. I saw the model number and assumed it was the same device, do I need to make a separate issue for this?

A side question, may I know why my device shows up as ZY-M100-24GV2 in z2m when it looks more like the picture for ZY-M100-24G (wall mounted rather than ceiling mounted)?

Koenkk, 2024-10-12:
That's a different model indeed, fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

iamthekings5, 2024-10-13:
> That's a different model indeed, fixed it!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

@Koenkk 

I'm on the latest dev branch where the issue was fixed and the log spam has stopped but my settings are still periodically being reset to some value.

The screenshot contains all values being reset, only presence timeout retains its settings.

[image]

fabianosan, 2024-10-13:
Here is the same.

Koenkk, 2024-10-14:
Could you provide the debug log when these values reset? I think it's an issue in the device firmware.

Koenkk, 2024-10-16:
`{"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}]}` here you see the device send presence sensitivity 0, so this confirms it's an issue of the device and not of z2m.

fabianosan, 2024-10-16:
> `{"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}]}` here you see the device send presence sensitivity 0, so this confirms it's an issue of the device and not of z2m.

But it works fine in Tuya App.
Maybe Tuya cloud ignore this values and maintain the current value if zero? 

Now I'm using it on Tuya with LocalTuya and it's working flawless.

iamthekings5, 2024-10-17:
Do all ZY-M100-24GV2 have this issue? I might want to get a replacement.

I have tried using the tuya integration but it wouldn't recognise state: presence as me being there, only state:move. Would localtuya solve this issue?

Any idea what my options are now?

kkossev, 2024-10-19:
The same problem when changing the minimum and the maximum detection distance parameters is confirmed in another hub.
We may need a sniff from a Tuya GW on what is actually sent when the distance parameters are changed.

xino222, 2024-11-02:
hello

I have two ZY-M100-24GV3 sensors and the same thing happens, when I disconnect it from the power supply and reconnect it, it loses all the values and they are set to default. This does not happen with the old 5.8G version.

Anyway, although the values are set to 0 I think it is just visual because it really still retains the sensitivity and distances that you set.

I hope it can be fixed, thank you very much.

KeithL52, 2024-11-02:
> `{"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}]}` here you see the device send presence sensitivity 0, so this confirms it's an issue of the device and not of z2m.

I have
2 x  Tuya ZY-M100-24GV2     _TZE204_7gclukjs
2 x  Tuya ZY-M100-24GV3     _TZE204_ya4ft0w4
they all exhibit the same problem of loosing the settings when the power is turned off then back on.

From your answer above, am I correct in saying that these devices are never going to work correctly with Zigbee2mqtt ?

xino222, 2024-11-02:
> > `{"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}]}` aquí ves que el dispositivo envía sensibilidad de presencia 0, por lo que esto confirma que es un problema del dispositivo y no de z2m.
> 
> Tengo2 x Tuya ZY-M100-24GV2 _TZE204_7gclukjs2 x Tuya ZY-M100-24GV3 _TZE204_ya4ft0w4todos presentan el mismo problema de perder la configuración cuando se apaga y luego se vuelve a encender la alimentación.
> 
> De su respuesta anterior, ¿tengo razón al decir que estos dispositivos nunca funcionarán correctamente con Zigbee2mqtt?

hello

I am working fine with Zigbee2mqtt the only annoying thing is the loss of the visual configuration when there is a power outage or even sometimes it modifies some value of the representation sensor randomly....

xino222, 2024-11-14:
Hello, I would like to know if this problem can be solved in the near future or is it not possible? I have several presence sensors with this problem of losing the configured values ​​and it is annoying.

Mark-Muc, 2025-01-02:
Hello ... also have this problem and also like to know if it is "fixable"
Mark-Muc
In the meanwhile i am experimenting to set the values in the configuration.yaml file to nail it down.

    detection_distance_max: 6.75
    detection_distance_min: 1.5
    move_sensitivity: 0
    presence_sensitivity: 0
    presence_timeout: 300

ESP4Ever, 2025-01-06:
Hi!

Having same issue (find switch & move/presence sensitivity settings loss) with _TZE204_ya4ft0w4 sensors but it seems that it doesn't affect their operation.

stuartm, 2025-01-11:
For whatever it is worth, this doesn't seem to be specific to ZB2MQTT, as I'm experiencing the exact same issue with ZHA. Whatever value I set will revert back to an earlier value (e.g. zero) between a few seconds and a couple of minutes later. By directly writing the attribute to different values via the ZHA toolkit I can ultimately get it to 'stick' at a value close enough to the one I wanted.

So this does, to me at least, seem like a firmware bug.

xino222, 2025-02-02:
good afternoon

I just want to inform you that after the last FIX the sensor still does not save the changes.   :(

thanks for your efforts.

jjfs127, 2025-02-18:
Same issue here with three of my _TZE204_ya4ft0w4 flashed with firmware from here to reduce spamming the zigbee network

https://github.com/Koenkk/zigbee2mqtt/issues/19045

xino222, 2025-02-21:
> El mismo problema aquí con tres de mis _TZE204_ya4ft0w4 parpadeados con firmware desde aquí para reducir el spam en la red zigbee
> 
> [#19045](https://github.com/Koenkk/zigbee2mqtt/issues/19045)
> 
> se utilizó este firmware: https://github.com/Andrik45719/ZY-M100/blob/main/ZY-M100-24GV3_TZE204_ya4ft0w4_increased_report_interval_TD_0_disable.bin
> 
> move_sensitivity / presence_sensitivity: y nunca guardó y volvió a 0, lo que guarda los registros de HA
> 
> [@Koenkk](https://github.com/Koenkk) lamento molestarte, pero supongo que este problema sigue ocurriendo para mí y para los demás, ¿alguna solución para esto?

I hope they can fix it because it is very annoying....

have tried several times without success ... I do not understand why it is so difficult .. since the previous version of this sensor, the 5.8G does not have this problem.

arnevanhove, 2025-02-21:
Does anybody know if the MOES ZSS-QY-HP has the same issues (spamming network)? Seems like a good alternative. I know there are a lot of great mmWave sensors developed by the community or less know brands but they all use a usb cable and are very difficult to hide somewhere. I need a ceiling mounted one. 

https://www.expressalarm.cz/documents/zss-qy-hp-ms/cs/v/manual%20dd15-240415.pdf?srsltid=AfmBOorKDId8U6KbsSFenKrRotSrMJdu68YcXLJkCOWfq3QiH8_X9bQU

I now have a _TZE204_ya4ft0w4. Works "great" but it spams my zigbee network. Only noticed this when it was allready installed. The battery versions work great btw. At least in my experience. So, if anybody has tried the MOES ZSS-QY-HP. Let me know!

jjfs127, 2025-02-22:
> Does anybody know if the MOES ZSS-QY-HP has the same issues (spamming network)? Seems like a good alternative. I know there are a lot of great mmWave sensors developed by the community or less know [...]

Check my post, there is a custom firmware flash to fix the zigbee spamming. But this is not the problem reported here, rather values not being saved thus HA spamming the logs with number invalid errors.

arnevanhove, 2025-02-22:
> Check my post, there is a custom firmware flash to fix the zigbee spamming. But this is not the problem reported here, rather values not being saved thus HA spamming the logs with number invalid [...]

Sorry if it was not entirely related! So the custom firmware, which should fix the spamming issue, has issues? I have _TZE204_ya4ft0w4 , just like you. The problem is that I'm not able to flash firmware since due to my visual impairment, soldering is very difficult :( .

jjfs127, 2025-02-26:
@Koenkk you think you can open this issue?

arnevanhove, 2025-02-26:
My device _TZE204_ya4ft0w4
 was not flashed with the firmware that fixes the constant distance reporting. But I do also see these in my HA logs:
Invalid value for number.badkamer_plafond_mmwave_move_sensitivity: 0 (range 1.0 - 10.0)
Invalid value for number.badkamer_plafond_mmwave_presence_sensitivity: 0 (range 1.0 - 10.0)

The "find switch" option (whatever that may be, regulary switches back to on. So i'm guessing that the same will happen if you change these values? (not sure).

musapinar, 2025-03-01:
Same as https://github.com/Koenkk/zigbee2mqtt/issues/25692

arnevanhove, 2025-03-05:
Follow up on my post from 26/02. My _TZE204_ya4ft0w4 ceiling mounted mmwave (230v) seems to have "lost" settings. There has not been any power outages in my house.

> **First occurred: March 1, 2025** at 17:47:44 (659065 occurrences)
> Last logged: 12:17:41
> 
> Invalid value for number.badkamer_plafond_mmwave_move_sensitivity: 0 (range 1.0 - 10.0)
> Invalid value for number.badkamer_plafond_mmwave_presence_sensitivity: 0 (range 1.0 - 10.0)

I'm also testing a _TZE200_kb5noeto battery operated version and that seems to work flawlessly. Maybe this is a hardware issue?

xino222, 2025-03-11:
hi,

since the last or penultimate update of zigbee2mqtt it seems that this alert was modified so that if the presence sensor is NOT configured between 1 and 10 these messages jump continuously in the log and every second they appear so I have millions of messages continuously indicating to configure the sensor between value 1 and 10. This is a problem because the sensor sets it to 0 automatically by the problem that we are all dealing with in this thread.

I think it is better to modify it again so that it does NOT pop up this alert continuously as it can worsen the performance therefore spam...

log:

2025-03-11 11:45:59.833 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:45:59.834 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_move_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:45:59.935 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:45:59.936 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_move_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:45:59.975 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c1384e3d5d7076_move_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:45:59.976 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c1384e3d5d7076_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:45:59.991 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:45:59.992 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_move_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:46:00.049 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:46:00.050 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_move_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:46:00.098 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:46:00.103 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_move_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:46:00.152 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-11 11:46:00.152 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c13824c58f12f0_move_sensitivity: 0 (range 1.0 - 10.0)

Regards

sincze, 2025-03-11:
[image]
Let's see how long these values are staying in the current settings. Seem to reset over time causing:

`Invalid value for number.XXXXXX_presence_sensitivity: 0 (range 1.0 - 10.0)`

In my  _TZE200_ya4ft0w4 sensor.

montoyra, 2025-03-12:
Hi,

I have the same issue with all my devices _TZE204_ya4ft0w4, _TZE200_ya4ft0w4.
My two centims to mitigate the spam reporting is use the z2m debounce capability:

debounce: 2
debounce ignore: presence

jjfs127, 2025-03-12:
> z2m debounce

So  the following will mitigate these errors:
```

       debounce: 2
       debounce_ignore:
            - presence
            - move
```

tomav, 2025-03-15:
Same error here with TS0601 / _TZE200_ya4ft0w4

```
2025-03-15 18:25:55.148 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.dressing_presence_presence_sensitivity: 0 (range 1.0 - 10.0)
2025-03-15 18:25:55.155 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.dressing_presence_presence_sensitivity: 0 (range 1.0 - 10.0)
```

Mark-Muc, 2025-03-16:
Hello . I did a small Node-Red node that if 
`move_sensitivity: 0` or
`presence_sensitivity: 0 `
it set this values to da def. value in the node!!

[image]

Code:`
[{"id":"e3872866c54d4a39","type":"group","z":"f63a41d5.4029","name":"Praesenzmelder <> 0 set to ...","style":{"label":true,"color":"#0070c0","fill":"#ffffbf"},"nodes":["c565fe02b93f6236","43411cdfcfe3edb9","00f467989d87d0c1","aa9c96c3b14baa75","653d3a0dbd29a574","b0e811ca73a5af4f","05c9c7cba2ab55cb"],"x":54,"y":3819,"w":892,"h":162},{"id":"c565fe02b93f6236","type":"zigbee2mqtt-out","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -1- (ZY-M100-24GV3)","device_id":"0xa4c138909ad2d176","command":"{}","commandType":"json","payload":"payload","payloadType":"msg","optionsValue":"","optionsType":"nothing","x":770,"y":3900,"wires":[]},{"id":"43411cdfcfe3edb9","type":"inject","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"0","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{\"move_sensitivity\":0,\"presence_sensitivity\":0}","payloadType":"json","x":150,"y":3860,"wires":[["00f467989d87d0c1","c565fe02b93f6236"]]},{"id":"00f467989d87d0c1","type":"zigbee2mqtt-out","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -2- (ZY-M100-24GV3)","device_id":"0xa4c138dfeb5f3c7d","command":"{}","commandType":"json","payload":"payload","payloadType":"msg","optionsValue":"","optionsType":"nothing","x":770,"y":3940,"wires":[]},{"id":"aa9c96c3b14baa75","type":"function","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"sensitiv 2","func":"if (msg.payload.move_sensitivity == 0 || msg.payload.presence_sensitivity == 0 )\n  {\n    return {payload: { move_sensitivity: 2, presence_sensitivity: 2 }};\n  }\nreturn;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":480,"y":3900,"wires":[["c565fe02b93f6236"]],"inputLabels":["\"0\"??"],"outputLabels":["1"]},{"id":"653d3a0dbd29a574","type":"zigbee2mqtt-in","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -1- (ZY-M100-24GV3)","device_id":"0xa4c138909ad2d176","state":"0","outputAtStartup":true,"filterChanges":false,"enableMultiple":false,"x":230,"y":3900,"wires":[["aa9c96c3b14baa75"]]},{"id":"b0e811ca73a5af4f","type":"zigbee2mqtt-in","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -2- (ZY-M100-24GV3)","device_id":"0xa4c138dfeb5f3c7d","state":"0","outputAtStartup":true,"filterChanges":false,"enableMultiple":false,"x":230,"y":3940,"wires":[["05c9c7cba2ab55cb"]]},{"id":"05c9c7cba2ab55cb","type":"function","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"sensitiv 8","func":"if (msg.payload.move_sensitivity == 0 || msg.payload.presence_sensitivity == 0 )\n  {\n    return {payload: { move_sensitivity: 8, presence_sensitivity: 8 }};\n  }\nreturn;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":480,"y":3940,"wires":[["00f467989d87d0c1"]],"inputLabels":["\"0\"??"],"outputLabels":["1"]},{"id":"67c9aa0135085d8d","type":"zigbee2mqtt-server","name":"Zigbee2MQTT","host":"localhost","mqtt_port":"1883","mqtt_username":"pi","mqtt_password":"Markusk1","mqtt_qos":"0","tls":"","usetls":false,"base_topic":"zigbee2mqtt"}] 
`
if anyone is interested!
Mark-Muc

musapinar, 2025-03-16:
> Hello . I did a small Node-Red node that if `move_sensitivity: 0` or `presence_sensitivity: 0 ` the node set this values to da def. value in the node!!
> 
> [image]
> 
> Code:`[{"id":"e3872866c54d4a39","type":"group","z":"f63a41d5.4029","name":"Praesenzmelder <> 0 set to ...","style":{"label":true,"color":"#0070c0","fill":"#ffffbf"},"nodes":["c565fe02b93f6236","43411cdfcfe3edb9","00f467989d87d0c1","aa9c96c3b14baa75","653d3a0dbd29a574","b0e811ca73a5af4f","05c9c7cba2ab55cb"],"x":54,"y":3819,"w":892,"h":162},{"id":"c565fe02b93f6236","type":"zigbee2mqtt-out","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -1- (ZY-M100-24GV3)","device_id":"0xa4c138909ad2d176","command":"{}","commandType":"json","payload":"payload","payloadType":"msg","optionsValue":"","optionsType":"nothing","x":770,"y":3900,"wires":[]},{"id":"43411cdfcfe3edb9","type":"inject","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"0","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{\"move_sensitivity\":0,\"presence_sensitivity\":0}","payloadType":"json","x":150,"y":3860,"wires":[["00f467989d87d0c1","c565fe02b93f6236"]]},{"id":"00f467989d87d0c1","type":"zigbee2mqtt-out","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -2- (ZY-M100-24GV3)","device_id":"0xa4c138dfeb5f3c7d","command":"{}","commandType":"json","payload":"payload","payloadType":"msg","optionsValue":"","optionsType":"nothing","x":770,"y":3940,"wires":[]},{"id":"aa9c96c3b14baa75","type":"function","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"sensitiv 2","func":"if (msg.payload.move_sensitivity == 0 || msg.payload.presence_sensitivity == 0 )\n {\n return {payload: { move_sensitivity: 2, presence_sensitivity: 2 }};\n }\nreturn;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":480,"y":3900,"wires":[["c565fe02b93f6236"]],"inputLabels":["\"0\"??"],"outputLabels":["1"]},{"id":"653d3a0dbd29a574","type":"zigbee2mqtt-in","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -1- (ZY-M100-24GV3)","device_id":"0xa4c138909ad2d176","state":"0","outputAtStartup":true,"filterChanges":false,"enableMultiple":false,"x":230,"y":3900,"wires":[["aa9c96c3b14baa75"]]},{"id":"b0e811ca73a5af4f","type":"zigbee2mqtt-in","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"","server":"67c9aa0135085d8d","friendly_name":"Praesenzmelder -2- (ZY-M100-24GV3)","device_id":"0xa4c138dfeb5f3c7d","state":"0","outputAtStartup":true,"filterChanges":false,"enableMultiple":false,"x":230,"y":3940,"wires":[["05c9c7cba2ab55cb"]]},{"id":"05c9c7cba2ab55cb","type":"function","z":"f63a41d5.4029","g":"e3872866c54d4a39","name":"sensitiv 8","func":"if (msg.payload.move_sensitivity == 0 || msg.payload.presence_sensitivity == 0 )\n {\n return {payload: { move_sensitivity: 8, presence_sensitivity: 8 }};\n }\nreturn;","outputs":1,"timeout":0,"noerr":0,"initialize":"","finalize":"","libs":[],"x":480,"y":3940,"wires":[["00f467989d87d0c1"]],"inputLabels":["\"0\"??"],"outputLabels":["1"]},{"id":"67c9aa0135085d8d","type":"zigbee2mqtt-server","name":"Zigbee2MQTT","host":"localhost","mqtt_port":"1883","mqtt_username":"pi","mqtt_password":"Markusk1","mqtt_qos":"0","tls":"","usetls":false,"base_topic":"zigbee2mqtt"}] ` if anyone is interested! Mark-Muc

I had a similar workaround via an automation with triggers binded to move sensitvity + presence  sensitivity + distance switch but it did not work. Here is why : https://github.com/Koenkk/zigbee2mqtt/issues/25692#issuecomment-2682824327

Since how long have you been running this Node-Red node ?

Mark-Muc, 2025-03-17:
it run noch since 3 day's ... a lot of restarts by HA and or z2m .. works every time so far
Mark-Muc

arnevanhove, 2025-03-26:
So is there any way to fix this issue? I assume that "losing settings" is a hardware problemen? But my _TZE204_ya4ft0w4
 keeps working fine. Should the logging be adjusted so it doenst report the error? Or should the values be changed so it "includes" the 0 and not the 1.0 to 10 for example? 

I have limited knowledge on how this all work, but I do want to help if I can.

xino222, 2025-04-06:
> Entonces, ¿hay alguna manera de solucionar este problema? Supongo que "perder la configuración" es un problema de hardware? Dar mi _TZE204_ya4ft0w4sigue funcionando bien. ¿debería ajustarse el registro para que no informe el error? ¿O deberían cambiarse los valores que espera el sistema para que "incluya" el 0 y no el 1,0 al 10, por ejemplo?
> 
> Tengo conocimientos limitados sobre cómo funciona todo esto, pero quiero ayudar si puedo.

that would be ideal before that did not happen and was not reporting thousands of logs that the sensor should be configured between 1 and 10 but of course as it is set to 0 continuously ........

it would be nice to leave it as it was before because at least it was not saturated with logs....

Molier, 2025-07-28:
yes ofcourse, should have shared it from the start 
```yaml
description: Auto-fix Tuya presence sensor invalid sensitivity values
triggers:
  - trigger: mqtt
    topic: zigbee2mqtt/presence sensor
    value_template: "{{ value_json.move_sensitivity | default(11) }}"
    payload: "0"
    id: move_sensitivity_zero
  - trigger: mqtt
    topic: zigbee2mqtt/presence sensor
    value_template: "{{ value_json.presence_sensitivity | default(11) }}"
    payload: "0"
    id: presence_sensitivity_zero
conditions: []
actions:
  - data:
      topic: zigbee2mqtt/presence sensor/set
      payload: |-
        {
          "move_sensitivity": 5,
          "presence_sensitivity": 5
        }
    action: mqtt.publish
  - condition: template
    value_template: >-
      {{ now().timestamp() - state_attr('automation.presence_sensor_clamper',
      'last_triggered').timestamp() > 10 }}
  - data:
      message: >-
        Tuya presence sensor spamming invalid sensitivity values (0). 
        Auto-correcting to 5. This is a known firmware bug.
      level: warning
    action: system_log.write
mode: restart
max: 3

```

closed 2025-10-06 by github-actions[bot]