Koenkk/zigbee2mqtt #24051: Nous A1Z smart plug: OTA page not reporting firmware version and build date

Ricc68, 2024-09-21:
### What happened?

As title, the Nous A1Z smart plug is not reporting the fw version and build date in the OTA page:

[image]

In the log files I see the information so it must be something higher level:

[2024-09-21 16:42:35] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lavatrice', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":7.68,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"restore","state":"ON","update":**{"installed_version":192,"latest_version":192,"state":"idle"}**,"update_available":null,"voltage":225}

### What did you expect to happen?

That the firmware version and build date is displayed in z2m OTA frontend.

### How to reproduce it (minimal and precise)

Configure a Nous A1Z smart plug and check the OTA page.

### Zigbee2MQTT version

1.4.1

### Adapter firmware version

7.4.3

### Adapter

Sonoff ZBDongle-E

### Setup

qemu HAOS on x86 NAS and 

### Debug log

This is the log, I have enabled the debug messages then pressed the check for updates on the Nous A1Z then disabled the debug:

[log.log.gz](https://github.com/user-attachments/files/17084494/log.log.gz)

Dremor, 2024-11-02:
z2m 1.41.0, still not fixed.

Ricc68, 2025-01-02:
@koenkk this issue is still there. I'm waiting for the new implementation of firmware update to see, report back and eventually close.

AlexanderLeporiday, 2025-02-20:
Just to chime in on this, my TS011F based Nous D2Z report a version of 69 and latest version 69 (no update needed) but nothing displays in the OTA columns for version or build.
[image]
[image]

Device that does show it:
[image]

Ricc68, 2025-04-13:
@Koenkk and @Nerivec also in z2m 2.2.1 the issue is there: still no firmware versions shown in OTA page for A1Z power plugs.

Nerivec, 2025-04-13:
`installed_version` and `latest_version` are OTA-specific values, they are not `firmware build date` or `firmware version`, which are both _optional_ attributes of the `basic` cluster. Although _most_ devices do implement them, I assume this one does not, hence the values are not displayed. You can confirm this by looking at the `database.db` line for the device, and see if you find a `swBuildId` and `dateCode` in the line (which correspond to `firmware version` and  `firmware build date` respectively). It should also show an error in the logs for both values if you re-interview the device (likely `UNSUPPORTED_ATTRIBUTE`).

Ricc68, 2025-04-14:
@Nerivec yes, I confirm that swBuildId and dateCode are not reported by this device.

The only reported values are:

"update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },

I assume that firmware update will work once a new firmware is available.

As this is confusing for many users, including me, I can try to code something in the UI to fill in the blanks. My idea is to use installed_version as fw version and N/A as build date. Will see if I can tag the entry in the page (with a short message for example) that the firmware is missing the standard attributes.

This way users can be reassured about firmware version and that updates are still working if available and maybe stop opening issues that are device issues and not z2m.

Do you agree with this course?

Ricc68, 2025-04-14:
Yes, I agree: I think it doesn't make much sense to use non OTA related variables in the OTA page.

The swBuildId and dateCode are already exposed in the Info page of the device so there's no need to duplicate.

If you agree I can review the OTA UI to use installed_version and latest_version variables.

Nerivec, 2025-04-14:
Sure. You might want to wait until https://github.com/nurikk/zigbee2mqtt-frontend/pull/2449 is merged so you don't conflict with it.

Ricc68, 2025-04-22:
I've started the implementation (and let's put aside the steep learning curve for the frontend code 😭).
For the file version conversion to human readable format I will follow the ZCL spec "11.4.2.7 File Version" ignoring any vendor-specific encoding to avoid the mess of customizing the conversions: 8 bit app release, 8 bit app build, 8 bit stack rel, 8 bit stack build, to hex and BCD-ish conversion (A.B.C.D).
Btw, I know that real comparison for updates is over the 32 bit integer and I could stay with it, but I want to have a proper conversion that tells something rather than the raw number.

Ricc68, 2025-04-25:
@Nerivec now I'm satisfied with the result, I will make the PR for the frontend.
@Koenkk I have changed the UI of the OTA page of the frontend to show the real installed and available file versions of the devices firmware (see comments from me and Nerivec above):

[image]

This will remove the confusion of using the firmware reported "version" and build date which in general are not even necessarily related with the real file version defined by the ZigBee protocol (meaning running firmware file version and ota update file version).
In addition the "version" and build date are not always reported by the devices because they are manufacturer dependent so this change will remove the confusion of devices not showing firmware version and build date in the OTA page. With this PR we can close all those many issues of "device not reporting firmware version in OTA page" which were caused by the false belief that not finding version and release date in the OTA page meant to the users that there must be a bug: in fact it's not a bug, but on one side a manufacturer choice and on the other side the OTA page was reporting values not directly related to OTA update files (manufacturer specific version reported by firmware can be different from update file version as defined by the ZigBee protocol).

Ricc68, 2025-05-03:
Fixed by mentioned PR.

closed 2025-05-03 by Ricc68

Soulfly999, 2025-09-02:
Hey, i'm confused about it. Is it fixed now. I have still this problem with the Nous A1Z. It shows Firmware ID N/A

[image]

Nerivec, 2025-09-02:
The firmware ID and date are **optional** parameters, devices are not required to report it.

Nerivec, 2025-09-03:
Yes, the firmware ID (and date code) is only a "pretty display" that manufacturers can set for the firmware version. It actually takes no part in OTA at all, it's purely cosmetic.
_Since this is configured in-firmware, it could be shown with one, and then the manufacturer could have forgotten/abandoned in a new update, and it would not show anymore._