Koenkk/zigbee2mqtt #24065: [New device support]: CFMT02ZB Schneider electric thermostat

huco95, 2024-09-22:
### Link

https://www.se.com/es/es/product/CFMT02ZB/termostato-conectado-wiser/

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0xdc8e95fffeabd464","nwkAddr":43363,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"CFMT02ZB","epList":[1,2,4,5,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,513,516,2821,65027,65302,65315],"outClusterList":[3,10,25,1026,1030],"clusters":{"genBasic":{"attributes":{"modelId":"CFMT02ZB","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":8}},"seMetering":{"attributes":{}},"hvacThermostat":{"attributes":{"occupiedHeatingSetpoint":1400,"localTemp":2296,"systemMode":4}},"hvacUserInterfaceCfg":{"attributes":{"keypadLockout":0,"tempDisplayMode":0}},"schneiderSpecificPilotMode":{"attributes":{"pilotMode":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2311}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffeb0cf9e","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}},"4":{"profId":260,"epId":4,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffeb0cf9e","endpointID":1}],"configuredReportings":[{"cluster":1030,"attrId":0,"minRepIntval":10,"maxRepIntval":60,"repChange":0}],"meta":{}},"5":{"profId":260,"epId":5,"devId":13,"inClusterList":[0,3,4,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"seMetering":{"attributes":{"divisor":1000,"multiplier":1,"instantaneousDemand":0,"currentSummDelivered":[0,0]}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffeb0cf9e","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":1024,"minRepIntval":10,"maxRepIntval":65000,"repChange":5000},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":[0,100]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1727027305519}

### Zigbee2MQTT version

1.40.1-1

### Comments

I have tried the following externar definition from a similar schneider electric device but I was not able to make everything work. The running state is not working, you can set the temperature but I can not know when it is running or not.

### External definition

```shell
const { } = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CFMT02ZB'],
    model: 'CFMT02ZB',
    vendor: 'Schneider Electric',
    description: 'Smart thermostat',
    fromZigbee: [
        fz.stelpro_thermostat,
        fz.metering,
        fz.schneider_pilot_mode,
        fz.wiser_device_info,
        fz.hvac_user_interface
    ],
    toZigbee: [
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_system_mode,
        tz.thermostat_running_state,
        tz.thermostat_local_temperature,
        tz.thermostat_control_sequence_of_operation,
        tz.schneider_pilot_mode,
        tz.schneider_thermostat_keypad_lockout,
        tz.thermostat_temperature_display_mode,
    ],
    exposes: [
        e.binary('keypad_lockout', ea.STATE_SET, 'lock1', 'unlock').withDescription('Enables/disables physical input on the device'),
        e.enum('schneider_pilot_mode', ea.ALL, ['contactor', 'pilot']).withDescription('Controls piloting mode'),
        e
            .enum('temperature_display_mode', ea.ALL, ['celsius', 'fahrenheit'])
            .withDescription('The temperature format displayed on the thermostat screen'),
        e
            .climate()
            .withSetpoint('occupied_heating_setpoint', 4, 30, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'heat'])
            .withRunningState(['idle', 'heat'])
            .withPiHeatingDemand(),
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat']);
        await reporting.thermostatPIHeatingDemand(endpoint1);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['seMetering']);
        await endpoint1.read('hvacUserInterfaceCfg', ['keypadLockout', 'tempDisplayMode']);
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The current temperature, Occupied heating setpoint, Temperature display mode... is working but 'Running state' is not working, if you try to update it it gives the following error: 
`z2m: Publish 'get' 'running_state' to 'Termostato salón' failed: 'Error: ZCL command 0xdc8e95fffeabd464/1 hvacThermostat.read(["runningState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

huco95, 2024-09-25:
I have manage to update the running state by adding this reporting:
[image]

But I don't know how to add this to the external definition.

luixal, 2024-10-08:
Hi!

Have you managed to add support for this?

Thanks!

huco95, 2024-10-15:
> Hi!
> 
> Have you managed to add support for this?
> 
> Thanks!

It is working for me using the external definition in first post and adding the piHeatingDemand reporting. But it would be great to have official support.

closed 2024-12-22 by github-actions[bot]