Koenkk/zigbee2mqtt #24070: BSD29_1 (_TZ3000_okaz9tjs) No longer reporting voltage, current, and power. (But still Energy)

AquaXV, 2024-09-22:
### What happened?

After updating to 1.40.1 I found that the smart plugs no longer report voltage, current, and power. But they do still report energy.

Zigbee Model
    TS011F
Zigbee Manufacturer
    _TZ3000_okaz9tjs
Description
    Smart plug (with power monitoring by polling)
Manufacturer
    [Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
Model
    [BSD29_1](https://www.zigbee2mqtt.io/devices/BSD29_1.html#tuya-bsd29_1)
Firmware build date
    20210625
Firmware version
    500+TZSKT31BS102

I looked at other issues and tried the testing ext.js mentioned in #20028 (but error "cannot find module zigbee-herdsman-converters/lib/extend), as well as making sure all fields in "Settings (Specific)" have values as mentioned in another issue I can't find back.

### What did you expect to happen?

The polling to report values

### How to reproduce it (minimal and precise)

Have a plug. Install 1.40.1.

### Zigbee2MQTT version

latest-dev

### Adapter firmware version

20240710

### Adapter

SONOFF ITead Zigbee 3.0 Dongle (CC2652P)

### Setup

In Docker with MQTT, and Home Assistant Integration

### Debug log

No errors.
Upon inspecting logs it seems z2m sends mqtt topics with updated energy values, but the zigbeeherdsman messages are the same regardless of the energy actually changed or not in the topic. 
I don't see topics with updated power, current, or voltage. Nor any zh debug logs

AquaXV, 2024-09-23:
Update; I updated z2m to latest-dev around 10 minutes prior to this post, currently Energy is reporting Null. Voltage, Power, and Current is correct now.

Koenkk, 2024-09-24:
Energy should be fixed now 

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

AquaXV, 2024-09-25:
Yep, all fixed! Thanks :D

closed 2024-09-25 by Koenkk