Koenkk/zigbee2mqtt #24078: Innr Smart Plug (SP240) not publishing at all, updates values only after manually using the "Update button" per exposure item.

njordan77, 2024-09-23:
### What happened?

I'm using Z2M in its latest version v1.40.1, have 40+ zigbee items, also smart plugs of different brands. With Innr smart plugs it all works normal, except the fact that it simply does not update values, e.g. the PWER value. the device is in standby, however the Exposure Page and all other related still show 200W....when pressong the update button for POWER, it updates immediately to 5W.

The same is true for any other exourse item in the list. Its strange..... all other smart plugs just update by themselves, at least power update very x minutes......

### What did you expect to happen?

It should update like with all other smatplugs installed

### How to reproduce it (minimal and precise)

yes, always the same. switched all arround, but the effect stays the same.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20220219

### Adapter

ZZH!

### Setup

Z2M plain, with Openhab cooperation

### Debug log

...

SchoolGuy, 2024-09-30:
I can confirm this report. Configured two SP 240 and they show an extremely delayed update cycle. The Aquara Plugs that are already in the network are updating at a much higher rate.

One of the plugs has an LQI of 170 and another one of 145. As such I wouldn't suspect a bad link quality for this issue.

I don't know if the issue is unrelated, but since I added the Innr Smart Plugs, my network map in the dashboard doesn't seem to load anymore. Before loading it was a matter of ~20 seconds.

SchoolGuy, 2024-10-03:
It appears that my fix was to reduce the maximum reporting time of the plug down to 60 seconds. This change was recommended by the repository owner (not tagging to reduce the noise) in the original issue of the device support.

Koenkk, 2024-10-04:
@SchoolGuy can you try (on z2m 1.40.2) to change the `change` value instead of the `max` time (set max to `3600`)

Treiblesschorle, 2024-10-25:
Having the same problem, unfortunately changing "min rep change" does not work for me, the value never updates. Can someone confirm that this works for them? I'm setting it to 0.1, not sure if the floating point number could be a problem (I didn't try a 1 yet). However, setting "max rep interval" to 60 seconds as suggested works. Would still be nice if "min rep change" would work as well.

closed 2025-01-01 by github-actions[bot]

Tockra, 2025-03-25:
I've the same problem. See : https://github.com/Koenkk/zigbee2mqtt/discussions/26855