Koenkk/zigbee2mqtt #24079: [New device support]: SE CCTFR6001

enthalpie91, 2024-09-24:
### Link

https://www.se.com/fr/fr/product/CCTFR6001/wiser-actionneur-de-chaudi%C3%A8re-1-canal-220v/

### Database entry

id: 0x0c4314fffe989321

### Zigbee2MQTT version

1.40.1

### Comments

Hello I want to use Z2M and if t have all item ( thermostatiques valves .. I can't because i don't be aible to use the CCTFR6001

### External definition

```shell
const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CCTFR6001'],
    model: 'CCTFR6001',
    vendor: 'Schneider Electric',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"11":11}}), onOff({"powerOnBehavior":false,"endpointNames":["1","11"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I don't know

enthalpie91, 2024-09-24:
I got and error syntaxe in the configuration.yaml file
[image]

enthalpie91, 2024-09-24:
may be it is not important .. but it will cold in my appartement if i don't get answer

enthalpie91, 2024-09-24:
I should said the external definition, given by console_ dev is only representative of the ON/OFF mode used actually
but the CCTFR6001 is able to work with the OT protocole, so I think the external definition file is not complete
Best and i hope i will get answer soon

senna1992, 2024-09-27:
Please investigate according to this guideline =)

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

you can help make it be supported this way!

enthalpie91, 2024-09-27:
@senna1992 
if because i did that i ask for support !!
device is see in Z2M but not usable !

enthalpie91, 2024-09-27:
HI
1°) we are the 27/09 and the heating will be start soon
2°) i am not a specialist , so i am waiting to get this device supported  officially as the other SE devices like thermostatic vanne [Wv704r0a0902)
3°) as i said the value return by dev_console are certainly not complete because the CCTFR6001 can run in two mode ON/OFF and opentherm
4°) it is unbelievable to have all stuff from SE and no boilier command stuff, 
Regards

senna1992, 2024-09-27:
all of the devices here are integrated by someone who has invested time to make it work. This is a free and open project.
If noone used that device with z2m until now, thats why its not supported.
This project is open and depends on users to create the converters for the devices they own, if not supported.

Please stay humble and participate by learning to take those steps or just get the manufactures solution and use that.

enthalpie91, 2024-09-27:
Hi know that ! when i can i help , i did as soon as possible
i just tell you i don't have knowledge to do 
best

enthalpie91, 2024-10-01:
Hello. this will not be necessary. the all wiser stuffs are in thi projet
https://github.com/asantaga/wiserHomeAssistantPlatform

closed 2024-10-01 by enthalpie91