Koenkk/zigbee2mqtt #24081: Unknown key

Lassejoe, 2024-09-24:
### What happened?

After flashing new firmware on my Sonoff zigbee plus dongle I wanted to add a new device.
I have already paired 3 of these device before to my network.

When it goes to pair mode I receive an error:
```
[2024-09-24 13:16:06] warning: 	zhc:lumi:trv: Unknown key 252 = 0
```

The device I'm trying to add:
https://www.zigbee2mqtt.io/devices/SRTS-A01.html#aqara-srts-a01

### Zigbee2MQTT version

 1.40.1 commit: 403d3c0

### Adapter firmware version

20240710

### Adapter

sonoff zigbee 3.0 usb dongle plus

### Setup

Docker - RPI5

### Debug log

[log.log](https://github.com/user-attachments/files/17113904/log.log)

Koenkk, 2024-09-25:
Not sure what this value means but I've changed the log level to debug

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Lassejoe, 2024-09-26:
@Koenkk thanks for the input.
I found out the issue, why the device couldn't be paired.
It was not remove properly from the network, once done I was able to re-pair it.

I'm sorry the inconveniences this might have caused.

closed 2024-09-26 by Lassejoe