Koenkk/zigbee2mqtt #24085: [New device support]: Tuya PIR Motion Sensor ZM-35ZH-Q TS0601  _TZE200_gjldowol

dr-apple, 2024-09-24:
### Link

https://de.aliexpress.com/item/1005005769341971.html?spm=a2g0o.order_list.order_list_main.10.788d5c5fECwDX5&gatewayAdapt=glo2deu

### Database entry

{"id":58,"type":"EndDevice","ieeeAddr":"0xa4c13845da857e68","nwkAddr":47598,"manufId":4417,"manufName":"_TZE200_gjldowol","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":12,"inClusterList":[0,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1727184411211}

### Zigbee2MQTT version

1.40.1

### Comments

It dosn't won't to work....

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_gjldowol',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

all,, the device have nothing

tanelkoth, 2024-09-26:
I have the same problem. In the documentation it states that the model number is 'ZM-35H-Q'. Searching through the source code I found a motion sensor that has the same model TS0202. After some cut and paste work with this model and the new fingerprint the device shows up in the Zigbee configuration. However, it seems that nothing works. 
This is what the configuration currently looks like:

_zigbeeModel: ['TS0601'],
		fingerprint: tuya.fingerprint('TS0601', ['_TZE200_gjldowol']),
        model: 'ZM-35H-Q',
        vendor: 'TuYa',
        description: 'Motion sensor',
        fromZigbee: [fromZigbee_1.default.ias_occupancy_alarm_1, fromZigbee_1.default.battery, fromZigbee_1.default.ignore_basic_report, fromZigbee_1.default.ZM35HQ_attr, legacy.fromZigbee.ZM35HQ_battery],
        toZigbee: [toZigbee_1.default.ZM35HQ_attr],
        extend: [(0, modernExtend_1.quirkCheckinInterval)(15000)],
        exposes: [e.occupancy(), e.battery_low(), e.battery(),
            e.enum('sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),
            e.enum('keep_time', ea.ALL, [30, 60, 120]).withDescription('PIR keep time in seconds'),
        ],
        configure: tuya.configureMagicPacket,
        whiteLabel: [
            tuya.whitelabel('Aubess', '40ZH-O', 'Motion sensor', ['_TZ3040_msl6wxk9']),
        ],_

Can you help in determining what is going wrong here?
Kind regards,
erwin

gp-pg, 2024-10-28:
I have same problem too.

HarsiTomiii, 2024-11-01:
same issue for me, leaving a comment for getting notified.

i don't know enough to make the definition myself, so i asked AI models, at one point it got connected and recognized, but nothing was exposed so it was useless.

vantus1987, 2024-11-02:
Good afternoon! Am I correct in understanding that there is currently no fully-functional support for this device?

tanelkoth, 2024-11-02:
Good afternoon too. Yes, I think that you are correct. Currently this device is not working!
Regards,
erwin

vantus1987, 2024-11-02:
Look, through the tuya gateway I found the starting points
{"1":"PIR state","4":"Battery level","9":"PIR Sensitivity","12":"Illuminance value","101":"Interval Time","102":"Pir Delay"}

vantus1987, 2024-11-02:
Now I have made an external converter

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_gjldowol',
        },
    ],
    model: 'TZE200_gjldowol',
    vendor: 'Tuya',
    description: 'Luminance motion sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.occupancy(),
            e.illuminance().withUnit('lx'),
            e.battery(),
            e
                .enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
                .withDescription('PIR sensor sensitivity (refresh and update only while active)'),
            e
                .enum('keep_time', ea.STATE_SET, ['10', '30', '60', '120'])
                .withDescription('PIR keep time in seconds (refresh and update only while active)'),
            e
                .numeric('illuminance_interval', ea.STATE_SET)
                .withValueMin(1)
                .withValueMax(720)
                .withValueStep(1)
                .withUnit('minutes')
                .withDescription('Brightness acquisition interval (refresh and update only while active)'),
        ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'occupancy', tuya.valueConverter.trueFalse0],
            [4, 'battery', tuya.valueConverter.raw],
            [9, 'sensitivity', tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
            [101, 'keep_time', tuya.valueConverterBasic.lookup({'10': tuya.enum(0), '30': tuya.enum(1), '60': tuya.enum(2), '120': tuya.enum(3)})],
            [12, 'illuminance', tuya.valueConverter.raw],
            [102, 'illuminance_interval', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

vantus1987, 2024-11-02:
So, it happened and it turned out :rolleyes: it's just the first time I'm doing this
Dedicated to all who do not mind the time for all sorts of nonsense))
The 2 in 1 motion + illumination sensor, which has the IH012-RT01 model written on the box, and the ZM-35ZH-Q instruction, but in fact is not supported in z2m and is defined as TZE200 gjldowol, sketched an external converter, everything seems to work take it away
[TZE200_gjldowol.txt](https://github.com/user-attachments/files/17608352/TZE200_gjldowol.txt)

dr-apple, 2024-11-03:
You are the best! It works perfectly!!! Thank you very much

HarsiTomiii, 2024-11-03:
> So, it happened and it turned out, it's just the first time I'm doing this Dedicated to all who do not mind the time for all sorts of nonsense)) The 2 in 1 motion + illumination sensor, which has the IH012-RT01 model written on the box, and the ZM-35ZH-Q instruction, but in fact is not supported in z2m and is defined as TZE200 gjldowol, sketched an external converter, everything seems to work take it away [TZE200_gjldowol.txt](https://github.com/user-attachments/files/17608352/TZE200_gjldowol.txt)

Awesome! thank you @vantus1987 !

nnov4k, 2024-12-27:
Guys, how to set default values?
log says for every values:
error: 	z2m: Publish **'set' 'illuminance_interval' to 'PIR@motion sensor' failed:** 'Error: ZCL command 0xa4c138b4af2f658a/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":101,"datatype":2,"data":[0,0,0,15]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'

Programmer86, 2025-01-11:
> So, it happened and it turned out, it's just the first time I'm doing this Dedicated to all who do not mind the time for all sorts of nonsense)) The 2 in 1 motion + illumination sensor, which has [...]

Thanks, this worked for me. Can you tell me how soon this will be merged into the main branch?

zoomtronicOR, 2025-01-19:
There is no way I could make this PIR sensor becomes functional, I did update of the coordinator firmware, created .js file in zigbee2mqtt folder in HA, an it does not work, device is still unsupported.

`Zigbee2MQTT
version [2.0.0]
Coordinator type EmberZNet
Coordinator revision 7.4.5 [GA]
Coordinator IEEE Address 0x84b4dbfffec9c30d
Frontend version 0.9.4
zigbee-herdsman-converters version 21.12.0
zigbee-herdsman version 3.2.1

.js file

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
// Since a lot of Tuya devices use the same modelID, but use different datapoints
// it's necessary to provide a fingerprint instead of a zigbeeModel
fingerprint: [
{
// The model ID from: Device with modelID 'TS0601' is not supported
// You may need to add \u0000 at the end of the name in some cases
modelID: 'TS0601',
// The manufacturer name from: Device with modelID 'TS0601' is not supported.
manufacturerName: '_TZE200_gjldowol',
},
],
model: 'TZE200_gjldowol',
vendor: 'Tuya',
description: 'Luminance motion sensor',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
configure: tuya.configureMagicPacket,
exposes: [
e.occupancy(),
e.illuminance().withUnit('lx'),
e.battery(),
e
.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
.withDescription('PIR sensor sensitivity (refresh and update only while active)'),
e
.numeric('keep_time', ea.STATE_SET)
.withValueMin(5)
.withValueMax(3600)
.withValueStep(1)
.withUnit('seconds')
.withDescription('PIR keep time in seconds (refresh and update only while active)'),
e
.numeric('illuminance_interval', ea.STATE_SET)
.withValueMin(1)
.withValueMax(720)
.withValueStep(1)
.withUnit('minutes')
.withDescription('Brightness acquisition interval (refresh and update only while active)'),
],
meta: {
// All datapoints go in here
tuyaDatapoints: [
[1, 'occupancy', tuya.valueConverter.trueFalse0],
[4, 'battery', tuya.valueConverter.raw],
[9, 'sensitivity', tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
[12, 'illuminance', tuya.valueConverter.raw],
[101, 'illuminance_interval', tuya.valueConverter.raw],
[102, 'keep_time', tuya.valueConverter.raw],
],
},
extend: [
// A preferred new way of extending functionality.
],
};

module.exports = definition;`

and here is database entry

{"id":8,"type":"EndDevice","ieeeAddr":"0xa4c13839d0ffc28c","nwkAddr":63927,"manufId":4417,"manufName":"_TZE200_gjldowol","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":12,"inClusterList":[0,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_gjldowol","powerSource":3,"zclVersion":3,"appVersion":80,"stackVersion":1,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736027608520}`

zoomtronicOR, 2025-01-26:
I made it, but on whole internet there is no clear explanation how and where to put external converter, just vague explanations.
You need to create **external_converters** directory under **zigbee2mqtt** directory, and there You need to put your .js files!
And no one mentioned this small info , so small that if You didn't did this, nothing will work, no matter which external converter you define.  Thanks for Nothing Chat GPR shitty as is knows more than all github.

barrypearseburke, 2025-03-07:
> I made it, but on whole internet there is no clear explanation how and where to put external converter, just vague explanations. You need to create **external_converters** directory under [...]

You deserve a medal . TY

rolfvreijdenberger, 2025-03-08:
can confirm this works.
* in your specific zigbee2mqtt folder (I have multiple since I have multiple instances) create the folder 'external_converters'.
* in that directory, create a javascript file <whatever-you-want-to-call-your-file>.js 
* put the contents from the txt file from @vantus1987 in the js file
* restart zigbee2mqtt
* the device should work (albeit without a nice picture to show for it in the GUI)
* enjoy a good 4 euro device from Ali :)

closed 2025-05-15 by github-actions[bot]