Koenkk/zigbee2mqtt #24089: TRADFRI Routers frequently go offline/unavailable

Ar-eh, 2024-09-24:
### What happened?

Routers, especially but not exclusively TRADFRI, frequently go unavailable or offline, and remain offline until the power to the router is cycled. 

### What did you expect to happen?

Devices to remain online

### Zigbee2MQTT version

1.39.1 (also 1.40.x)

### Adapter firmware version

0x26530900

### Adapter

Conbee III

### Setup

Docker container on Synology DS718

### Debug log

[2024-09-24 10:45:26] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 103
[2024-09-24 10:45:26] debug: 	zh:deconz:frameparser: device state: 100010
[2024-09-24 10:45:26] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-24 10:45:33] debug: 	zh:controller:endpoint: Error: ZCL command 0xd0cf5efffe090f71/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (84))
[2024-09-24 10:45:33] warning: 	z2m: Failed to ping 'Bulb 24 (Office)' (attempt 1/2, ZCL command 0xd0cf5efffe090f71/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (84)))
[2024-09-24 10:45:34] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2024-09-24 10:45:34] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-24 10:45:34] debug: 	zh:deconz:driver: query aps data indication
[2024-09-24 10:45:34] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 104
[2024-09-24 10:45:34] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 104 srcAddr: 0x505d destAddr: 0x0 profile id: 0x104 cluster id: 0x400 lqi: 195
[2024-09-24 10:45:34] debug: 	zh:deconz:frameparser: response payload: 8,158,10,0,0,33,241,90
[2024-09-24 10:45:34] debug: 	zh:controller: Received payload: clusterID=1024, address=20573, groupID=null, endpoint=2, destinationEndpoint=1, wasBroadcast=false, linkQuality=195, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":158,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":33,"attrData":23281}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-09-24 10:45:34] debug: 	zh:controller:endpoint: ZCL command 0x00178801086449f7/2 msIlluminanceMeasurement.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":158,"writeUndiv":false})
[2024-09-24 10:45:34] debug: 	zh:deconz: no response expected (158)
[2024-09-24 10:45:34] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1

…

[2024-09-24 10:45:47] debug: 	zh:deconz: no response expected (93)
[2024-09-24 10:45:47] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-24 10:45:47] debug: 	z2m: Received Zigbee message from 'Philips Sensor (Office)', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 2 with groupID null
[2024-09-24 10:45:47] debug: 	zh:controller:endpoint: Error: ZCL command 0xd0cf5efffe090f71/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (85))
[2024-09-24 10:45:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Sensor (Office)', payload '{"battery":100,"illuminance":26567,"illuminance_lux":454,"led_indication":true,"linkquality":214,"motion_sensitivity":"max","occupancy":true,"occupancy_timeout":0,"temperature":21.06}'
[2024-09-24 10:45:47] warning: 	z2m: Failed to ping 'Bulb 24 (Office)' (attempt 2/2, ZCL command 0xd0cf5efffe090f71/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (85)))
[2024-09-24 10:45:47] debug: 	z2m: Active device 'Bulb 24 (Office)' was last seen '63.86' minutes ago.
[2024-09-24 10:45:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bulb 24 (Office)/availability', payload 'offline'
[2024-09-24 10:45:47] debug: 	zh:deconz:driver: DATA_REQUEST - destAddr: 0xc60c EP:2 SeqNr. 125 request id: 123
[2024-09-24 10:45:47] debug: 	zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 123 status: 0
[2024-09-24 10:45:47] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1

Ar-eh, 2024-10-20:
It appears I was able to solve this problem by adding "adapter_delay: 200" to the Advanced section of the configuration file, after finding [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/9554#issuecomment-1866133354) in a thread about a similar problem impacting the Conbee II. It would appear that the same mitigation is required for Conbee III.

fsedarkalex, 2024-11-11:
I have a similar problem, but for me I need to re-pair the routers. At leas I already did this. I will observe it.

closed 2025-01-19 by github-actions[bot]