Koenkk/zigbee2mqtt #24090: Exception when adding Presence sesnsor

mohamed-easy, 2024-09-24:
### What happened?

Using a Sonoof USB 3.0 P Zigbee Controller i am trying to add a Tuya based presence sensor. When the presence detector is activated it triggers multiple errors in Zigbee2MQTT.
Exception while calling fromZigbee converter: Value '10' is not allowed, expected one of [object Object],[object Object],[object Object],[object Object]}
Exception while calling fromZigbee converter: Value '540' is not allowed, expected one of [object Object],[object Object],[object Object],[object Object]}

### Zigbee2MQTT version

 1.40.1-1

### Adapter firmware version

NA

### Adapter

NA

### Setup

Home Assistant

### Debug log

[logs.txt](https://github.com/user-attachments/files/17118951/logs.txt)

mohamed-easy, 2024-09-25:
model: 'ZG-205ZL'
modelID: 'TS0225',

mohamed-easy, 2024-09-28:
@Koenkk is there any solution for that?

Koenkk, 2024-09-30:
We have to figure out what value 10 means, this is the mode, the current values are:

```
{arm: tuya.enum(0), off: tuya.enum(1), alarm: tuya.enum(2), doorbell: tuya.enum(3)}
```

Do you have any clue what mode it is in when sending this value?

mohamed-easy, 2024-10-01:
I didn't, however i changed to ember, and i update the firmware, and the sensor works now.
i think it will be a mystery.
anyway thanks for your time