Koenkk/zigbee2mqtt #24091: Gledopto GL-C-006P detected as GL-C-009P

mcsrobert, 2024-09-24:
### What happened?

When I'm pairing my GL-C-006P, it shows up as GL-C-009P.

Interestingly enough, this was working correctly ±2 years ago. I'm running into this now, because I'm rebuilding my ZigBee network from scratch.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Setup

k3s on a rpi4

mcsrobert, 2024-09-27:
```
{"id":6,"type":"Router","ieeeAddr":"0xa4c1383229b5a38e","nwkAddr":37552,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"modelId":"GL-C-009P"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20220422","swBuildId":"10051203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1727418232455}
```

Koenkk, 2024-09-28:
The `modelId` of this device is `GL-C-009P` (this comes from the device), I don't see a way to fix this besides stopping z2m, editing the `modelId` manually to `GL-C-003P` and start z2m.

mcsrobert, 2024-09-28:
I was able to change it to the `GL-C-003P`, but as I pointed out before, this is a model for a RGB device, which [this is not](https://www.gledopto.eu/media/image/product/5553/lg/zigbee-2polige-cct-steuerung-dimmer.jpg)...

Koenkk, 2024-09-30:
Fixed that!

mcsrobert, 2024-10-03:
Hoi Koen,

I've tested this on `latest-dev` and interestingly enough the device is reporting as a `GL-C-006P` again. Not sure why! Still not the `GL-C-003P`, but given how similar the devices are, I am able to control it that way.

I then went ahead and edited the DB entry again and it is now showing correctly as `GL-C-003P` as CCT. Thanks!

One thing, reading the docs, I noticed `GL-C-003P` is also referenced as RGB [here](https://www.zigbee2mqtt.io/devices/GL-C-008P.html#gl-c-001p). Just want to check that changing the `GL-C-003P` to CCT won't break the `GL-C-001P`.

dalexloe, 2024-10-04:
Hi Koenkk,
after I updated to 1.40.2 commit: e06848db yesterday, all my 3 5in1 GLEDOPTO mini controllers that I set to GL-C-003P (RGB/blue light) are now recognized as CCT controllers and I can no longer use them. They can only be controlled in brightness.
Here is my controller (same as the 5in1 mini):
https://www.zigbee2mqtt.io/devices/GL-C-008P.html
After the last Z2M update, it is incorrectly identified as this one:
https://www.zigbee2mqtt.io/devices/GL-C-003P.html#gledopto-gl-c-003p

These changes need to be reworked so that I can use my controllers as before

Forgive my English, I had to use a translator because I only speak German.

Koenkk, 2024-10-05:
@dalexloe the `GL-C-003P` is only a CCT device update the modelId in your database.db to `GL-C-008P` to fix it (see https://github.com/Koenkk/zigbee2mqtt/issues/24091#issuecomment-2380643828)

Koenkk, 2024-10-06:
Try changing it to `GL-S-003Z`

tonyazzolino, 2024-10-07:
Will Z2M no longer be able to automatically recognize the RGB mode of the GL-C-001P 5-in-1 controller, which supposedly identifies as a GL-C-003P when in RGB mode? Did GLEDOPTO use the same model ID for two different devices? I've got a few of these that were working fine in RGB mode, but now must be manually set to something else because the definition for the GL-C-003P was changed to a CCT only controller.

Koenkk, 2024-10-07:
@tonyazzolino could you provide the data/database.db entry of your device?

a005, 2024-10-17:
I have the same 3in1 controller in RGB mode. Here is database record:
`{"id":40,"type":"Router","ieeeAddr":"0xa4c138ac6f2b5ab1","nwkAddr":46586,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-003P","epList":[11],"endpoints":{"11":{"profId":260,"epId":11,"devId":258,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-003P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240821","swBuildId":"20751203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":7,"colorTempPhysicalMin":158,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240821","swBuildId":"20751203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1729147848423}`

I have to use it with a custom converter that revert back changes in the main codebase.

calianradu, 2024-10-22:
Same issue here, after Z2M update RGB mode not available. Manually edited modelId from GL-C-003P to GL-C-008P to fix

`{"id":29,"type":"Router","ieeeAddr":"0xa4c138ef2bd68ad9","nwkAddr":934,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-008P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":258,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-008P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20230721","swBuildId":"20351203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":7,"colorTempPhysicalMin":158,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20230721","swBuildId":"20351203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1729627561781}
`

helgek, 2024-11-02:
This also fixed it for me. Running Z2M as addon, had to shut down Z2M first to make edits to database.db become effective. Hopefully there will be a better fix sometime soon.

> Same issue here, after Z2M update RGB mode not available. Manually edited modelId from GL-C-003P to GL-C-008P to fix [...]

SatAmnesiac, 2024-11-02:
@Koenkk what prompted the change in Z2M for recognising the GL-C-003P model as a new and different device? 

If it is a new device then there will be NO existing Z2M installations with one. But plenty of people with now broken GL-C-001P devices.

There those of us with broken GL-C-001P installations out there  now-  Plus there may be lots of frustrated Z2M users who can't use Gledopto controllers on sale now eg GL-C-001P, GL-C-002P, GL-C-103P, GL-C-201P & GL-C-301P for RGB strips in Z2M or do the 10xP devices show a different controller for RGB?

If I have to mess around with my Z2M to make a previously working controller continue to work will I have to make the database change for every future Z2M update? (as you can imagine I am not confident/happy messing around in databases I don't know - are there some simple straightforward instructions somewhere for people like me?)

Is there another unique identifier that's different between an RGB 003P and a ColourTemp 003P that Z2M can detect??

Is it possible to add the original characteristics of the RGB controller into Z2M so that rather than map the RGB-003 to another not quite the same we can change the model id to something unique that is just RGB?

Thanks

SatAmnesiac, 2024-11-03:
Also the OP noted that his Colour Temp 003P was detected as an 006P initially and it worked that way - so why not keep that behaviour - the RGB 003Ps get detected as RGB and  Colour/Dim 003Ps get detected as working 006Ps. If documented it means no-one messes with databases and existing users aren't (as I am doing now as a work around) left to back out recent Z2M updated in order to get the house automation back to something that works for my wife and kids. 

Appreciate this is a Gledopto inspired mess, but, if there is a work-around that leaves everyone with a working environment those of us who want to keep using Z2M AND not spend a fortune retrofitting controllers that do work with Z2M that seems like a sensible solution to me.

Koenkk, 2024-11-05:
@SatAmnesiac could you provide the data/database.db entry of this device?

SatAmnesiac, 2024-11-09:
> @SatAmnesiac could you provide the data/database.db entry of this device?

If you tell me how to I can- currently when I configure the GL-C-001P as an RGB controller it identifies as a GL-C-003P (which I understand it should do) and Z2M 1.41.0 identifies it as a CCT/DIM controller. 

I contacted GledOpto on Reddit and I got this reply
_Gledopto Official
For GL-C-003P 2 in 1 CCT/Dim controller, The CCT displayed on HA is GL-C-006P, The Dim displayed on HA is GL-C-009P._

So it seems that they don't expect a GL-3-003P to identify as such but as either an 06 or 09

Hope this helps.

Thanks

Koenkk, 2024-11-10:
You can find the `data/database.db` file next to the z2m `configuration.yaml` (you can open it with a text editor)

pjxturner, 2024-11-11:
Thanks for the workaround.

The frustrating part of this is that last year, I renovated my kitchen and dining areas and used Gledopto 5-in-1 and 2-in-1 controllers for all light strips.

The 5-in-1 controllers (GL-C-001P on the box) I use for coloured strips, accurately report as GL-C-008P when connected to RGBCCT strips, GL-C-007P when connected to RGBW strips and GL-003P when connected to the RGB strips in my bar.

In testing, these 5-in-1 controllers, also correctly reported as GL-C-006P when connected to a white CCT strip and GL-C-009P when used as a dimmer on a mono-coloured/white strip.

I also have two new 5-in-1 controllers, GL-C-201P and GL-C002P, with the wago-type connecters, and these work and report in exactly the same way.

**I also have 5 2-in-1 controllers and yes, they're called GL-C-003P on the unit and instructions, but these report as GL-C-009P as expected when used as dimmers.  And although I don't use this mode, changing the option to control a CCT strip, has them report to Zigbee2MQTT as GL-C-006P.**

With the change that was made, I lost the ability to control my RGB strips (GL-C-003P) as originally designed.  Changing them in the database to show as, and be controlled as, RGBCCT strips (GL-C-008P) is clunky, as I get CCT controls in Home Assistant that are obviously superfluous and ineffective for an RGB strip.

I don't understand why you chose to remove RGB functionality just so you could match a model number printed on a device (GL-C-003P) that actually accurately reports that correct and different modelId's (GL-C-009P & GL-C-006P) depending on which mode is selected.

And, most importantly, thank you for everything you do.

Aogu181, 2024-11-12:
> > @dalexloe the `GL-C-003P` is only a CCT device update the modelId in your database.db to `GL-C-008P` to fix it (see [#24091 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/24091#issuecomment-2380643828))`GL-C-003P` 只是一个 CCT 设备，将 database.db中的 modelId 更新到 `GL-C-008P` 以修复它（参见 [#24091 （评论）](https://github.com/Koenkk/zigbee2mqtt/issues/24091#issuecomment-2380643828)）
> 
> GL-C-003P is a RGB device,not CCT. I am an employee of this company.GL-C-003P 是 RGB 设备，不是 CCT。我是这家公司的员工。

We have a customer response that the RGB mode under GL-C-003P is mistakenly identified as CCT mode.

Koenkk, 2024-11-12:
Since you are a Gledopto employee I guess you know better, but it is still strange that only CCT and DIM is listed on the device itself:

[image]

pjxturner, 2024-11-13:
The confusion here, is that we're talking about two different devices, and that even though the original device is called the GL-C-003P, it can never output a `GL-C-003P` 'signal' to z2m because it paradoxically doesn't support RGB mode. (Terrible product naming by Gledopto)

The OP is referencing the following:
**Model GL-C-003P - Gledopto ZIGBEE 2 Wires CT/DIM 2in1 LED Controller Pro**
_Modes/Outputs (click reset button to switch)_
`GL-C-009P` - Dimmer Mode (Indicator light on)
`GL-C-006P` - CCT Mode (Indicator light off)

So it can never tell z2m it's in `GL-C-003P` mode, because it doesn't support it.  The only way to do this is to manually override the `data/database` to change the name that one of these modes is reporting, and create a reference to this mode in `src/devices.gledopto.ts`.  In this case, the OP arbitrarily decided he wanted the `GL-C-006P` mode to be represented as `GL-C-003P` and `src/devices.gledopto.ts` was changed accordingly.

**Unfortunately, in doing this, it broke a different device**. (Actually, a number of different devices).

**Model GL-C-001P - Gledopto 5 in 1 Smart LED Controller**
_Modes/Outputs (click reset button to switch)_
`GL-C-008P` - RGBCCT Mode (White light on)
`GL-C-007P` - RGBW Mode (Yellow light on)
`GL-C-003P` - RGB Mode (Blue light on)
`GL-C-006P` - CCT Mode (Green light on)
`GL-C-009P` - Dimmer Mode (Red light on)

So now when you set a 5in1 controller to RGB mode, and it sends it's `GL-C-003P` message to the z2m database, z2m now thinks it's a CCT device because of the change made to `src/devices.gledopto.ts` the other day.

To show that this isn't just a theoretical discussion, here's a part of my setup:

[image]

You can see that the **Model GL-C-003P 2in1** is attached to an LED sign, and acting as a dimmer `GL-C-009P`, as shown by the indicator light being on.
The **Model GL-C-001P 5in1** is attached to an RGB strip, and acting as an RGB controller `GL-C-003P`, as shown by the light being blue.

My preferred outcome is that the change to `src/devices.gledopto.ts` is reversed so that those of us that use the RGB setting in a Gledopto multi controller have a consistent usable mode that relates to the output to z2m, and not what's printed on the device and packaging, whether it's a 2-in-1, 5-in-1, mini versions, the new 3-in-1 that supports RGBCCT/RGBW/RGB, or whatever else they put out in the future.

At this point at least they are being consistent with the message to z2m about which mode is being used, no matter which device is used.

The OP will have to accept that his **Model GL-C-003P 2in1** will not output a `GL-C-003P` signal and that if he connects to a CCT strip and selects CCT mode, the device will output a `GL-C-006P` signal and z2m will work accordingly, as it always has, and that if he want to connect it to a mono strip or LED device and use it as a dimmer, he selects dimmer mode, and it outputs a `GL-C-009P` signal, and once again z2m works as it also has. No `data/database` editing required.

Just trying to help, and keep my setup running as before. Thanks.

pjxturner, 2024-11-13:
The confusion here, is that we're talking about two different devices, and that even though the original device is called the GL-C-003P, it can never output a `GL-C-003P` 'signal' to z2m because it paradoxically doesn't support RGB mode. (Terrible product naming by Gledopto)

The OP is referencing the following:
**Model GL-C-003P - Gledopto ZIGBEE 2 Wires CT/DIM 2in1 LED Controller Pro**
_Modes/Outputs (click reset button to switch)_
`GL-C-009P` - Dimmer Mode (Indicator light on)
`GL-C-006P` - CCT Mode (Indicator light off)

So it can never tell z2m it's in `GL-C-003P` mode, because it doesn't support it.  The only way to do this is to manually override the `data/database` to change the name that one of these modes is reporting, and create a reference to this mode in `src/devices.gledopto.ts`.  In this case, the OP arbitrarily decided he wanted the `GL-C-006P` mode to be represented as `GL-C-003P` and `src/devices.gledopto.ts` was changed accordingly.

**Unfortunately, in doing this, it broke a different device**. (Actually, a number of different devices).

**Model GL-C-001P - Gledopto 5 in 1 Smart LED Controller**
_Modes/Outputs (click Option button to switch)_
`GL-C-008P` - RGBCCT Mode (White light on)
`GL-C-007P` - RGBW Mode (Yellow light on)
`GL-C-003P` - RGB Mode (Blue light on)
`GL-C-006P` - CCT Mode (Green light on)
`GL-C-009P` - Dimmer Mode (Red light on)

So now when you set a 5in1 controller to RGB mode, and it sends it's `GL-C-003P` message to the z2m database, z2m now thinks it's a CCT device because of the change made to `src/devices.gledopto.ts` the other day.

To show that this isn't just a theoretical discussion, here's a part of my setup:

[image]

You can see that the **Model GL-C-003P 2in1** is attached to an LED sign, and acting as a dimmer `GL-C-009P`, as shown by the indicator light being on.
The **Model GL-C-001P 5in1** is attached to an RGB strip, and acting as an RGB controller `GL-C-003P`, as shown by the light being blue.

My preferred outcome is that the change to `src/devices.gledopto.ts` is reversed so that those of us that use the RGB setting in a Gledopto multi controller have a consistent usable mode that relates to the output to z2m, and not what's printed on the device and packaging, whether it's a 2-in-1, 5-in-1, mini versions, the new 3-in-1 that supports RGBCCT/RGBW/RGB, or whatever else they put out in the future.

At this point at least they are being consistent with the message to z2m about which mode is being used, no matter which device is used.

The OP will have to accept that his **Model GL-C-003P 2in1** will not output a `GL-C-003P` signal and that if he connects to a CCT strip and selects CCT mode, the device will output a `GL-C-006P` signal and z2m will work accordingly, as it always has, and that if he want to connect it to a mono strip or LED device and use it as a dimmer, he selects dimmer mode, and it outputs a `GL-C-009P` signal, and once again z2m works as it also has. No `data/database` editing required.

Just trying to help, and keep my setup running as before. Thanks.

Aogu181, 2024-11-13:
> > Since you are a Gledopto employee I guess you know better, but it is still strange that only CCT and DIM is listed on the device itself:
> > <img alt="Screenshot 2024-11-12 at 21 16 06" width="686" src="https://private-user-images.githubusercontent.com/2892853/385451458-49e4c664-fdee-4f67-9b48-c6ab92a06244.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NTgzNzMsIm5iZiI6MTczMTQ1ODA3MywicGF0aCI6Ii8yODkyODUzLzM4NTQ1MTQ1OC00OWU0YzY2NC1mZGVlLTRmNjctOWI0OC1jNmFiOTJhMDYyNDQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTNUMDAzNDMzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjUyZmRiNGU1NDg3NGNiMjY4Zjc3NzM4ZGEwZmIyYTA3M2E4NjhjYWViYzUwZjUwNDIzM2Q2Yjg4YWY3NzQxMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ONM1RMXx0ZiUNBAp3tCurMn0RSIu6ouUO31Kgyb2kis">
> 
> At the beginning, our company only made a lighting controller with one mode, as I said GL-C-008P - RGBCCT Mode (White light on) GL-C-007P - RGBW Mode (Yellow light on) GL-C-006P - CCT Mode (Green light on) GL-C-009P - Dimmer Mode (Red light on) But there was no separate RGB mode. But as the business increased, we combined multiple modes into one controller (for example, there are two-in-one controllers and five-in-one controllers) The two-in-one controller is as shown in your picture. We named it GL-C-003P, but it has nothing to do with RGB. The RGB mode only appears on other multi-in-one controllers, and the model number is also GL-C-003P. So I am very sorry for the misunderstanding (but now the two-in-one has been discontinued). The most important point is that our company stipulates according to the different modes of the equipment, rather than just looking at the model on the product shell . The model numbers on the all-in-one controller are just for us to distinguish, and the actual model numbers displayed on the z2m should correspond to the following: GL-C-008P - RGBCCT Mode (White light on) GL-C-007P - RGBW Mode (Yellow light on) GL-C-003P - RGB Mode (Blue light on) GL-C-006P - CCT Mode (Green light on) GL-C-009P - Dimmer Mode (Red light on)

So you don't need to look at the model of the product's casing, but its working mode. The product models he sent to z2m were only sent according to different work mode. @Koenkk @pjxturner

Aogu181, 2024-11-13:
Although it says GL-C-003P, its working mode is only CCT/Dimmer.
So on z2m, CCT corresponds to GL-C-006P, and Dimmer corresponds to GL-C-009P.
Instead of GL-C-003P written on the shell corresponding to RGB.

SatAmnesiac, 2024-11-13:
@Koenkk 
I agree with @pjxturner 
> **My preferred outcome is that the change to `src/devices.gledopto.ts` is reversed** so that those of us that use the RGB setting in a Gledopto multi controller have a consistent usable mode that [...]

Gledopto produce, as @Aogu181 says, devices that consistently identify regardless of what's printed on the case of the product in the following manner:-
GL-C-008P - RGBCCT Mode 
GL-C-007P - RGBW Mode 
GL-C-003P - RGB Mode
GL-C-006P - CCT Mode 
GL-C-009P - Dimmer Mode 

In the example of controller with GL-C-001 printed on it never reports as a GL-C-001, just as one of the above to tell Z2M which mode it is in.

Similarly the GL-C-003 labled device never reports as such, it just reports the device number for the mode it is in.

Please revert the change that was made and all the Gledopto devices will work fine in Z2M, they just won't necessarily match the name in the database with the name printed on the device, and that's the way it's always been.

Finally as @pjxturner has said too - your work is really really appreciated.

Thanks again.

Koenkk, 2024-11-13:
Thanks for all the explanations, I've merged https://github.com/Koenkk/zigbee-herdsman-converters/pull/8305 so assuming this can be closed now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-11-13 by Koenkk

mcsrobert, 2024-11-14:
Hi all, I want to clarify that the original issue I had was that for some reason on a new install I had my device identified as a GL-C-009P. Not GL-C-006P.

I confirmed that previously it identified as a GL-C-006P and this worked fine.

I never requested any changes to be  made, and I was merely trying to understand why the device identified differently. I think a lot of the whole confusion came from the fact that none of this weird behavior was documented clearly anywhere.