Koenkk/zigbee2mqtt #24096: Avoid icons into configuration.yaml

Danieldiazi, 2024-09-25:
### What happened?

Into configuration.yaml I've images of each icon device

` friendly_name: zb2
    description: room1
    icon: >-
      data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAgAAAAIACAYAAAD0eNT6AAAABGdBTUEAALGPC/xhBQAAACBjSFJNAAB6JgAAgIQAAPoAAACA6AAAdTAAAOpgAAA6mAAAF3CculE8AAAABmJLR0QA/wD/AP+gvaeTAAAACXBIWXMAAAsSAAALEgHS3X78AAABIHpUWHRSYXcgcHJvZmlsZSB0eXBlIHhtcAAAKJF1UktyxSAM23OKHoHIxibHSR+w60yXPX5lkr6keS2e/GwjSyLp6+MzvcUSSJKHV9m8WvNszYqrLcjxbQ/rLlGTBthiasNgRbY9/+weAHI6YZh8jy2laisZWcWGcyOyGLrkeUF63pAjSAEEN9kKVJPabf5eDA7VlZFl48zhc6E7m9DnCMeQRdYIjCRZwAR4bzsIn+IrYUnbK1oMiPLJ5c5I1UpyMWFindJWutDJ8GigL+BsOhEMacQvoGt4TRdRAWR/yZqc+mk33+mfNQo+9oS0ZZo4Xqk/BfwM2DScWa0csg8F0unUI52o///`

I don't know how I did in order to this images are into this file, but it's a problem, because when I try to edit it with a editor, It consumes a lot of CPU and freezes my system 10 seconds.

### What did you expect to happen?

I wish to know how to disable it and delete this images into my configuration.yaml.

### How to reproduce it (minimal and precise)

I don't know how I activate this option.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Docker container

sjorge, 2024-09-26:
They can be added with the frontend, it's probably what you did to get them there:
[image]

It would be better to have a way to just provide a 'overrides' directory or something for the frontend. With a lot of device it indeed becomes impossible to easily edit the configuration.yaml

Danieldiazi, 2024-09-26:
> They can be added with the frontend, it's probably what you did to get them there: <img alt="Screenshot 2024-09-26 at 14 32 30" width="468" src="https://private-user-images.githubusercontent.com/379665/371143786-a23edee0-91ab-437e-b532-886e5bf0ebbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczNTcyNjAsIm5iZiI6MTcyNzM1Njk2MCwicGF0aCI6Ii8zNzk2NjUvMzcxMTQzNzg2LWEyM2VkZWUwLTkxYWItNDM3ZS1iNTMyLTg4NmU1YmYwZWJiYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyNlQxMzIyNDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMmRjNjJlZTk2NzYxZDA0OThlY2VmOTlmN2ZmNDg5NWE2MWJjNjBlOGExMTNkNjVhYzc1MWVjZjljNDk2Njg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.HYOZo_6duBiED9dP0jZPHHhNrFtpStGPd0pMmn2Irl8">
> 
> It would be better to have a way to just provide a 'overrides' directory or something for the frontend. With a lot of device it indeed becomes impossible to easily edit the configuration.yaml
> 
> It should be OK to stop z2m and manually remove them all ones. The frontend should then go back to fetching the online images.

Maybe I've clicked there but without know what was the purpose of that button :(

How can I disable it?

If I delete image icon entries into configuration.yaml is enough?

sjorge, 2024-09-26:
If you so it while z2m is stopped and don't push that button again afterwards, that should be enough yes.

Danieldiazi, 2024-09-26:
> They can be added with the frontend, it's probably what you did to get them there: <img alt="Screenshot 2024-09-26 at 14 32 30" width="468" [...]

Maybe I've clicked there but without know what was the purpose of that button :(

How can I disable it?

If I delete image icon entries into configuration.yaml is enough?

> If you so it while z2m is stopped and don't push that button again afterwards, that should be enough yes.

yes! It works!!
Thanks!

closed 2024-09-26 by Danieldiazi