Koenkk/zigbee2mqtt #24097: MGM210PA32JIA Zigbee module not working with Zigbee2MQTT

mohanblaze, 2024-09-25:
### What happened?

I’m using an MGM210PA32JIA Zigbee module with Zigbee2MQTT, which is malfunctioning. The module fails to join the network or communicate with other Zigbee devices.

### What did you expect to happen?

The MGM210 module should be able to connect with the Zigbee2mqtt and communicate as per the Zigbee protocol.

### How to reproduce it (minimal and precise)

1) Set up Zigbee2MQTT with MGM210 module on the latest version.
2) Attempt to connect zigbee2mqtt using the MGM210.
3) Observe failure in logs.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

Gecko-SDK-3.2(NCP -6.10.3)

### Adapter

MGM210PA32JIA

### Setup

Linux b1edge 4.9.56 #697 SMP PREEMPT Sat May 18 10:51:11 IST 2024 aarch64 aarch64 aarch64 GNU/Linux

### Debug log

[2024-09-25 14:59:35] info:     z2m: Logging to console, file (filename: log.log)
[2024-09-25 14:59:35] info:     z2m: Starting Zigbee2MQTT version 1.40.1 (commit #403d3c0)
[2024-09-25 14:59:35] info:     z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-25 14:59:38] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyS2","baudRate":57600,"rtscts":false,"autoOpen":false}
[2024-09-25 14:59:38] info:     zh:zstack:znp: Serialport opened
[2024-09-25 14:59:38] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-09-25 14:59:39] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-09-25 14:59:58] error:    z2m: Error while starting zigbee-herdsman
[2024-09-25 14:59:58] error:    z2m: Failed to start zigbee
[2024-09-25 14:59:58] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-25 14:59:58] error:    z2m: Exiting...
[2024-09-25 14:59:58] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:123:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:138:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:64:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:140:27)
    at start (/opt/zigbee2mqtt/index.js:154:5)

mohanblaze, 2024-09-25:
Hi. I made a little progress in that by choosing the adapter with ember and here is the log that I got.

root@se-704a0efe60d8:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.40.1 start
> node index.js

Starting Zigbee2MQTT without watchdog.
[2024-09-25 19:28:16] info:     z2m: Logging to console, file (filename: log.log)
[2024-09-25 19:28:16] debug:    z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
[2024-09-25 19:28:16] info:     z2m: Starting Zigbee2MQTT version 1.40.1 (commit #403d3c0)
[2024-09-25 19:28:16] info:     z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-25 19:28:16] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\'
[2024-09-25 19:28:17] info:     zh:ember: Using default stack config.
[2024-09-25 19:28:17] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"chann'
[2024-09-25 19:28:17] info:     zh:ember: ======== Ember Adapter Starting ========
[2024-09-25 19:28:17] info:     zh:ember:ezsp: ======== EZSP starting ========
[2024-09-25 19:28:17] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-09-25 19:28:17] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-09-25 19:28:17] debug:    zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyS2","baudRate":57600,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-09-25 19:28:17] info:     zh:ember:uart:ash: Serial port opened
[2024-09-25 19:28:17] info:     zh:ember:uart:ash: ======== ASH starting ========
[2024-09-25 19:28:17] debug:    zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-09-25 19:28:17] debug:    zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-09-25 19:28:18] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-09-25 19:28:19] debug:    zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2024-09-25 19:28:19] info:     zh:ember:uart:ash: ======== ASH connected ========
[2024-09-25 19:28:19] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=0)
[2024-09-25 19:28:19] info:     zh:ember:uart:ash: ======== ASH started ========
[2024-09-25 19:28:19] info:     zh:ember:ezsp: ======== EZSP started ========
[2024-09-25 19:28:19] debug:    zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=0 Len=4]
[2024-09-25 19:28:19] debug:    zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0](ackRx=0)
[2024-09-25 19:28:19] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-25 19:28:19] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
[2024-09-25 19:28:19] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=0](frmRx=0) Added to rxQueue
[2024-09-25 19:28:19] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=1)
[2024-09-25 19:28:19] debug:    zh:ember:ezsp: <=== [FRAME: ID=0:"VERSION" Seq=0 Len=7]
[2024-09-25 19:28:19] error:    z2m: Error while starting zigbee-herdsman
[2024-09-25 19:28:19] error:    z2m: Failed to start zigbee
[2024-09-25 19:28:19] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-25 19:28:19] error:    z2m: Exiting...
[2024-09-25 19:28:19] error:    z2m: Error: Adapter EZSP protocol version (8) is not supported by Host [13-14].
    at EmberAdapter.emberVersion (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1428:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at runNextTicks (node:internal/process/task_queues:64:3)
    at processImmediate (node:internal/timers:454:9)
    at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:705:9)
    at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1708:24)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:138:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:64:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:140:27)
    at start (/opt/zigbee2mqtt/index.js:154:5)

----------------------------------------
Below is the configuration file:
root@se-704a0efe60d8:/opt/zigbee2mqtt# cat data/configuration.yaml 
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

logging:
  level: debug

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyS2
  baudrate: 57600
  rtscts: false
  flow_control: xon/xoff
  adapter: ember

advanced:
  log_level: debug

root@se-704a0efe60d8:/opt/zigbee2mqtt#

Help us in that to move further. Thank you.

mohanblaze, 2024-09-25:
@Koenkk Please support.

marrem, 2024-12-10:
Same thing here:

```
zigbee2mqtt    | [2024-12-10 23:13:42] error: 	z2m: Error: Adapter EZSP protocol version (8) is not supported by Host [13-14].
zigbee2mqtt    |     at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1426:19)
zigbee2mqtt    |     at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:688:9)
zigbee2mqtt    |     at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
zigbee2mqtt    |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
zigbee2mqtt    |     at Zigbee.start (/app/lib/zigbee.ts:69:27)
zigbee2mqtt    |     at Controller.start (/app/lib/controller.ts:161:27)
zigbee2mqtt    |     at start (/app/index.js:154:5)
```

First of all the "Adapter" is the usb zigbee adapter I plugin to the usb port? Please confirm.

Secondly what is meant by "Host" in this errormessage?

Then I think I should somehow get a firmware on the adapter that has a EZSP protocol version supported by "Host". Which EZSP protocol versions are supported by Host? EZSP protocol version 13, 14??? Which firmware has that version of the EZSP protocol?

Or alternatively I could make "Host" support the adapter's version of EZSP protocol. But what exactly is "Host" and how do I make Host support EZSP version 8?

Would by nice if error messages would be a little less cryptic.

closed 2025-02-17 by github-actions[bot]