Koenkk/zigbee2mqtt #24099: [Feature request]: Add default reporting setting per device in configuration.yaml

sdaclin, 2024-09-25:
### Is your feature request related to a problem? Please describe

I've a set of SNZB-02P and I'm not pleased by the default configuration when they are joining.
So I've to change it for each new device joining.

### Describe the solution you'd like

A new Yaml section is added in `configuration.yaml` :
```yaml
device_models:
    'SNZB-02':
        temperature:
            'min_rep_interval': 3600,
            'max_rep_interval': 7200,
            'min_rep_change': 10
```

### Describe alternatives you've considered

I don't zee any other alternative :/

### Additional context

Thanks a lot for reading me.
I searched in the backlog for "default" but didn't manage to find such kind of feature request.
Thanks for your wonderfull work.

sjorge, 2024-09-26:
And bindings would be nice too! I had a similar need ages ago, I filed https://github.com/Koenkk/zigbee2mqtt/issues/13025 because this a more generic way to push 'per-device-type/model/...' settings.

One could then easily write a extension to do this. I guess technically one could write one anyway and use a separate config tree.

For now I just have a bash script that calls mosquitto_pub to setup the reporting/binds how I like them and run it each time I pair a certain device (well it's more of a if I pair a temperature sensor or a motion sensor, as I configure them all at the same interval)

sdaclin, 2024-09-26:
My installation is brand new and I've plenty to do but this kind of script is definitely a good idea ! 
For now I'm not fully convinced that all my sensors are configured the same way and it's haunting me 👻

sjorge, 2024-09-26:
Just for reference here is the (more complex one) that I use to configure all my lights, I have a mix of Hue and Trådfri

```
#!/usr/bin/bash

## Bulbs
## <type=c,ws,cws> <brand=g,h> <friendly_name>
# cws h light/bathroom/mood/bulb
# cws h light/office/mood/bulb
# cws h light/livingroom/mood/bulb1
# cws h light/livingroom/mood/bulb2
# ws g light/balcony/main/bulb
# cws h light/bedroom/main/bulb1
# cws h light/bedroom/main/bulb2
# cws h light/bedroom/main/bulb3
# cws h light/bedroom/main/bulb4
# ws h light/entrance/main/bulb
# ws h light/kitchen/counter/bulb1
# ws h light/kitchen/counter/bulb2
# ws h light/kitchen/counter/bulb3
# ws h light/kitchen/table/bulb1
# ws h light/kitchen/table/bulb2
# ws h light/kitchen/table/bulb3
# ws h light/kitchen/table/bulb4
# ws h light/livingroom/main/bulb1
# ws h light/livingroom/main/bulb2
# ws h light/livingroom/main/bulb3
# ws h light/livingroom/main/bulb4
# cws h light/hallway/main/bulb
# ws g light/pantry/main/bulb1
# ws g light/pantry/main/bulb2
# ws g light/bathroom/main/panel1
# ws g light/bathroom/main/panel2
# ws g light/office/main/panel1
# ws g light/office/main/panel2
# ws g light/office/main/panel3
# ws g light/office/main/panel4

## loop and setup
grep -E '^# (cws|ws|c) ' $0 | \
while read bulb; do
        btype=$(awk '{ print $2 }' <<< $bulb)
        bbrand=$(awk '{ print $3 }' <<< $bulb)
        bname=$(awk '{ print $4 }' <<< $bulb)

        echo -ne "[....] Setting up ${bname} as type ${btype} ..."
        # NOTE: power on and color change when off behavior
        mosquitto_pub -t zigbee2mqtt/${bname}/set/power_on_behavior -m 'off'

        # NOTE: state and level reporting
        mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                -m '{"id":"'${bname}'","cluster":"genOnOff","attribute":"onOff","minimum_report_interval":0,"maximum_report_interval":10800,"reportable_change":0}'
        mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                -m '{"id":"'${bname}'","cluster":"genLevelCtrl","attribute":"currentLevel","minimum_report_interval":5,"maximum_report_interval":10800,"reportable_change":1}'

        sleep 1

        # NOTE: color reporting
        case $btype in
                cws|ws|c)
                        if [ "$bbrand" != "h" ]; then
                                mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                                        -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"colorMode","minimum_report_interval":5,"maximum_report_interval":10800,"reportable_change":1}'
                        fi
                ;;
        esac
        case $btype in
                cws|ws)
                        mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                                -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"colorTemperature","minimum_report_interval":5,"maximum_report_interval":10800,"reportable_change":1}'
                        mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                                -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentX","minimum_report_interval":5,"maximum_report_interval":10800,"reportable_change":1}'
                        mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                                -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentY","minimum_report_interval":5,"maximum_report_interval":10800,"reportable_change":1}'
                ;;
        esac
        case $btype in
                cws|c)
                        mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                                -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentHue","minimum_report_interval":5,"maximum_report_interval":10800,"reportable_change":1}'
                        mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting \
                                -m '{"id":"'${bname}'","cluster":"lightingColorCtrl","attribute":"currentSaturation","minimum_report_interval":5,"maximum_report_interval":10800,"reportable_change":1}'
                ;;
        esac
        sleep 1

        # NOTE: color options
        case $btype in
                cws|ws|c)
                        mosquitto_pub -t zigbee2mqtt/${bname}/set/color_options -m '{"execute_if_off": true}'
                ;;
        esac
        # XXX: seems to not be support by hue, new tradfri stuff
        #mosquitto_pub -t zigbee2mqtt/${bname}/set/level_options -m '{"execute_if_off": true}'
        sleep 5
        echo -e "\r[DONE"
done
```

sdaclin, 2024-09-27:
Thanks a lot Jorge I'll manage to extract the logic out of it.

closed 2025-02-19 by github-actions[bot]