Koenkk/zigbee2mqtt #24100: Devices disappearing during a join process

major328, 2024-09-25:
### What happened?

Hi,
Since I'm on ember driver I have an issue. During a join a process one or more other previously well working battery powered devices dieappering from the network, or stop working. I've no idea why or what could be the problem. It happened with temperature sensors, motion sensors or door/window sensors. Sometime the device or devices deleted from the devices page, sometime only no more data from the device.

### What did you expect to happen?

Fix the problem

### How to reproduce it (minimal and precise)

just I try to add a new device to the network and mostly it happens

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

ZBDongle-E

### Setup

Home assistant on a proxmox

Nerivec, 2024-09-25:
Can you provide a debug log of the issue happening?

major328, 2024-09-25:
Currently I run z2m with default "info" log settings, not debug settings, but there are few error messages maybe it helps
1: this happened when I tried to rejoin a moiton sensor previously working for months just lost when I added a temperature sensor. This happenes 3 times this is the attempt3:
"[2024-09-19 18:16:48] error: 	z2m: Failed to configure 'Mozgásérzékelő mosogató', attempt 3 (Error: ZCL command 0x00124b002a65273f/1 genPowerCfg.read(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":1,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":113})
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:101:31)
    at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:480:42)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4982:14)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:738:26)
    at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22)
    at UartAsh.emit (node:events:517:28)
    at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45)
    at processImmediate (node:internal/timers:476:21))"

Last time it happened on 22th of September I lost a temperature sensor during repairing another motion sensor, but there is absolutly no log entry for that day. The last directory under the log named "2024-09-21.19-13-18" contains 3 files log.log,log1.log and log2.log but the oldest entry in log2.log from 2024-09-23 21:17:47 , so absolutly no log information from 22th of September.
 Meanwhile I'm full with netrok/route error messages after switching to ember , but in another thread I got message this is normal and not worry about this.

Nerivec, 2024-09-25:
Network/route errors are common enough, but too many is indicative of a problem (unstable network, bad router, that sort of things). These were previously not reported with `ezsp` driver, so, you would not have seen them before (even if they were there).
If you have long enough logs (`info` level or below, _note: you can also carefully combine them if they are split_), try using https://nerivec.github.io/z2m-ember-helper/ to see if you can spot a common source of problems.

If battery devices are failing configure like in your last log, make sure you wake them up before and during the process (by pushing the button at 2-3 seconds intervals until it's done).

Since your issue appears to be around joining, just to confirm, you can try running the `Check for EUI64 mismatch` repair with `ember-zli`:
https://github.com/Nerivec/ember-zli/wiki/Stack#check-for-eui64-mismatch
_It won't do anything if your adapter is not affected._

major328, 2024-09-26:
Thanks for your assist. I've checked the z2m-ember-helper and got some
points to check for example low number of routers (total 19 + coordinator
for 66 end devices) which is not enough , "packer retry count is high",
"stale neighbors count is high" , so I have some points to check . Thanks
again.

Meanwhile during weekend, I switch the log level and try to reconnect the
lost temperature sensor and If I lose another device again I will upload
the debug log, maybe it will contain some useful entry.

Br,
Akos

Nerivec ***@***.***> ezt írta (időpont: 2024. szept. 25.,
Sze, 17:04):

major328, 2024-09-29:
Yesterday I've rejointed the sensor and also adder 2 new without any issue, meantime I've upgraded the firmware of dongle to 7.4.4- Now significantly less route error message, much more LQI-s and no devices disappeared /stoped working during the join process. Maybe this was a firmware issue.

Akos

Nerivec, 2024-09-29:
Sounds like things are a lot better. _Flashing a firmware, even the same one again, can sometimes be beneficial in itself._
You probably should look into adding a few good routers to level things out a bit (you can also use a dedicated router, the Dongle-E can be flashed with a router firmware, if you have a spare one, or can get another).

I'll let you close this once you're satisfied that the issue has indeed been resolved.

major328, 2024-10-03:
> ### What happened?
> Hi, Since I'm on ember driver I have an issue. During a join process one or more previously well working battery powered devices disappering from the network, or stop working. I've no idea why or what could be the problem. It happened with temperature sensors, motion sensors or door/window sensors. Sometime the device or devices deleted from the devices page, sometime only no more data from the device.
> 
> ### What did you expect to happen?
> Fix the problem
> 
> ### How to reproduce it (minimal and precise)
> just I try to add a new device to the network and mostly it happens
> 
> ### Zigbee2MQTT version
> 1.40.1
> 
> ### Adapter firmware version
> 7.4.3 [GA]
> 
> ### Adapter
> ZBDongle-E
> 
> ### Setup
> Home assistant on a proxmox
> 
> ### Debug log
> _No response_

major328, 2024-10-03:
> Sounds like things are a lot better. _Flashing a firmware, even the same one again, can sometimes be beneficial in itself._ You probably should look into adding a few good routers to level things [...]

Hi,

It happened again , now during a reboot, not a join. Yesterday morning I've updated the Z2M with 88 perfectly running devices (log_1), during the day I've restarted the HA, 88 devices came back (log_2), and during the even I've updated the HA to 2024.10 but after restart only 87 devices are back (log_3)
[log_3.log](https://github.com/user-attachments/files/17240140/log_3.log)
[log_2.log](https://github.com/user-attachments/files/17240142/log_2.log)
[log_1.log](https://github.com/user-attachments/files/17240144/log_1.log)
.

Nerivec, 2024-10-03:
```
[2024-10-02 19:13:39] warning: 	z2m: Device 'Hőmérő pince' left the network
[2024-10-02 19:13:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hőmérő pince","ieee_address":"0x00158d0009cbc861"},"type":"device_leave"}'
```

Seems the device leaves the network on its own.
I see wild fluctuations in LQI for that one (from below 50 to nearly 200). Seems it either has troubles finding a good router, or it is affected by temporary (but regular) interference, or it is forced to change router a lot due to it going offline.
Do you have any router that is sometimes offline?
Any chance the area around that device is closer to a possible source of interference? WiFi, Bluetooth, Thread, etc...

major328, 2024-10-03:
Hi,

All of the routers are powered all time.
I've 2 (one Nous A1Z smartplug, and one Sonoff ZBMINI) routers within 1
meter, and nearly I've a solar inverter with Wifi and in 4 meters distance
a Wifi access point, but Zigbee on channel 11 Wifi on channel 1 with 20MHz
bandwidth.
So this could happen because device is too close to a solar inverter with
wifi?

Nerivec ***@***.***> ezt írta (időpont: 2024. okt. 3., Cs,
13:01):

Nerivec, 2024-10-03:
You should try changing channels, either WiFi at 11 and keep ZigBee at 11, or ZigBee at 20 or 25 and keep WiFi at 1. It should improve the stability of the ZigBee network a lot, then we can see what problems remain after that.

[image]

Note: `ember` supports changing channel, so, you should only have a minimal amount of devices to re-pair (if any).
https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#changing-the-zigbee-channel

major328, 2024-10-04:
> You should try changing channels, either WiFi at 11 and keep ZigBee at 11, or ZigBee at 20 or 25 and keep WiFi at 1. It should improve the stability of the ZigBee network a lot, then we can see what problems remain after that.
> 
> ![zigbee-wifi-channels](https://private-user-images.githubusercontent.com/62446222/373264127-a06a02e7-9269-4912-9ab5-0e53db4c01ef.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgwMjg4NTQsIm5iZiI6MTcyODAyODU1NCwicGF0aCI6Ii82MjQ0NjIyMi8zNzMyNjQxMjctYTA2YTAyZTctOTI2OS00OTEyLTlhYjUtMGU1M2RiNGMwMWVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA0VDA3NTU1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ2YTZkNDNkZjFmOTBmNjNhNGEyNTkxZWFmMTdlYWJmMTk0YjVjOWU2YmJiNjY5NzA3ZWYzZTI1ZGNiNmYxZDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.3DLP12PYejTyMvbEKb-fvrEDiwIP1G1vzjM8VEvb3RI)
> 
> Note: `ember` supports changing channel, so, you should only have a minimal amount of devices to re-pair (if any). https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#changing-the-zigbee-channel

Thanks. Yesterday I've lost again another temperature sensor which is located in the base floor  (the previous is installed in the basement), where I've another WIfi AP which operates on Wifi channel 11 (2462MHz) far away from zigbee channel 11. I did nothing around Homeassistant or zigbee or Wifi that time
[log1.log](https://github.com/user-attachments/files/17255140/log1.log)

closed 2024-12-20 by github-actions[bot]