Koenkk/zigbee2mqtt #24103: "Configuration is not consistent with adapter state/backup" while everything is consistent

t-b-c-22, 2024-09-25:
### What happened?

Zigbee2MQTT stopped working on 21/9 with the following error:
"Configuration is not consistent with adapter state/backup!"
I have tried adjusting the configuration to match the PAN ID, Extended PAN ID, and Network Key. Despite everything being configured correctly, the error persists. 

Logs show that the PAN ID, Extended PAN ID, and Network Key match both in the configuration and the backup. The add-on refuses to start and asks to re-commission the network, which would involve re-pairing all devices.

Also, the coordinator SoC temperature is running very high (around 98°C), while others run between 50-60°C. I contacted SMLight in order to ask them about this and they told me that the temperature sensor is not very precise. I didn't check the temperature before, so I'm not sure if the temperature value was always that.

```
[2024-09-25 10:08:44] error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2024-09-25 10:08:44] error: 	zh:adapter:zstack:manager: - PAN ID: configured=19191, adapter=19191
[2024-09-25 10:08:44] error: 	zh:adapter:zstack:manager: - Extended PAN ID: configured=f4d8f68f758ef596, adapter=f4d8f68f758ef596
[2024-09-25 10:08:44] error: 	zh:adapter:zstack:manager: - Network Key: configured=5b174336f1801bd84919b267abcb54e2, adapter=5b174336f1801bd84919b267abcb54e2
[2024-09-25 10:08:44] error: 	zh:adapter:zstack:manager: - Channel List: configured=11, adapter=11
[2024-09-25 10:08:44] error: 	zh:adapter:zstack:manager: Please update configuration to prevent further issues.
```

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20221226

### Adapter

SMLight SLZB-06 (CC2652P)

### Setup

Add-on on HASS OS on a x86 machine

### Debug log

[log.log](https://github.com/user-attachments/files/17134773/log.log)

sebastian-bugajny, 2024-10-13:
Hit the same issue 2 days ago on ZigStart UZG-01 (CC2652P7) and cannot fix it.

connorbenton, 2024-10-28:
Same issue here with SONOFF ZBDongle-P (CC2652P) on firmware 20210708

Micha70, 2024-12-07:
I had the same issue today...
`error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!`

Yesterday I have added a Tuya smoke detector.... Not sure if there is any relation.

Using Tasmota for Z2Mqtt...

Micha70, 2024-12-07:
Here a complete log of the error message...

```[08:03:17] INFO: Preparing to start...
[08:03:18] INFO: Socat not enabled
[08:03:18] INFO: Enabled Zigbee2MQTT watchdog with value '1,5,15,30,60'
[08:03:18] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT with watchdog (60000,300000,900000,1800000,3600000).
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: - PAN ID: configured=16200, adapter=16200
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: - Extended PAN ID: configured=cccccccc27da3f48, adapter=cccccccc27da3f48
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: - Network Key: configured=a1401554483fda27a1401554483fda27, adapter=a1401554483fda27a1401554483fda27
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: - Channel List: configured=11, adapter=11
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: Please update configuration to prevent further issues.
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
[2024-12-07 08:03:28] error: 	zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
[2024-12-07 08:03:28] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-07 08:03:28] error: 	z2m: Failed to start zigbee
[2024-12-07 08:03:28] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-07 08:03:28] error: 	z2m: Exiting...
[2024-12-07 08:03:28] error: 	z2m: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:251:35)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:62:26)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
```

Is there any way to recover, without building up the network again?

Koenkk, 2024-12-10:
@Micha70 could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Micha70, 2024-12-10:
I have solved it in the meanwhile. By deleting coordinator_backup.json, reinstallation zigbee2mqtt and repairing of the devices. Works now since Sunday. If problem reappears I will provide a debug log.

closed 2025-02-17 by github-actions[bot]

felixschndr, 2025-05-10:
Same issue here. After a restart I am hit with this. The values of all attributes are the same between the config and the state

felixschndr, 2025-05-10:
FYI: I opened this issue for the underlying software: https://github.com/Koenkk/zigbee-herdsman/issues/1403