Koenkk/zigbee2mqtt #24115: [New device support]: Philips Hue Solo Lightstrip 3 Meter

GerdRuetten, 2024-09-26:
### Link

https://www.philips-hue.com/de-de/products/all-products/product-page/solo-lightstrip

### Database entry

{"id":49,"type":"Router","ieeeAddr":"0x001788010e4b6429","nwkAddr":18640,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCL009","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCL009","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240628","swBuildId":"1.116.5"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":34516,"currentY":27086,"enhancedCurrentHue":7169,"currentSaturation":225,"colorTemperature":500}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240628","swBuildId":"1.116.5","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1727383553037}

### Zigbee2MQTT version

1.40.1-dev commit: 15d845b

### Comments

The external definition is from the 3 Meter model.
Model No. is: 929003816902

This lightstrip is also available in 5 meters and 10 meters.

Model No. 5 meters: 929003817002
Model No. 10 meters: 929003817102

### External definition

```shell
const {philipsLight} = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['LCL009'],
    model: 'LCL009',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philipsLight({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Works perfect with the external definition!
Exposes:
- Switch (state)
- Brightness
- Color temp
- Color temp startup
- Color (X/Y)
- Color (HS)
- Power-on behavior
- Effect
- Linkquality

Koenkk, 2024-09-27:
Great, could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/lib/philips.ts)?

ben-tsk, 2024-10-28:
Any news on this one? I just got mine today

ben-tsk, 2024-10-29:
Ah i see
Thank you very much 😊

> > Any news on this one? I just got mine today
> 
> Yes i do a PR for the 5 Mtr. Modell

ben-tsk, 2024-11-01:
I just installed the new release 1.41.0 which should include support for the 5 mtr. version. But the device is still unsupported and not recogognized.

abort, 2024-11-10:
> I just installed the new release 1.41.0 which should include support for the 5 mtr. version. But the device is still unsupported and not recogognized.
> 
> z2m: Device '0x001788010e336cdb' with Zigbee model 'LCL009' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Can confirm this... still seems to be controllable though

ben-tsk, 2024-12-01:
I just installed version [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0)

Still an unsupported device...

GerdRuetten, 2024-12-22:
Use the latest 1.42.0-Dev version from today.
I repair the support for this device and with this z2m version it's supported

[image]

If you don't want to use the Dev-Version from z2m, wait for the next stable.

Best regards
Gerd

closed 2024-12-22 by GerdRuetten

ben-tsk, 2024-12-22:
Thank you very much! 😊

Best regards
Ben