Koenkk/zigbee2mqtt #24116: zigbee2mqtt always restarting

XsniferX, 2024-09-26:
### What happened?

since couple days i have Z2M that always restarting

----------------------
[2024-09-26 23:53:23] debug: 	zh:controller:endpoint: Error: Bind 0x00158d0005d243e2/1 seMetering from '0x00124b002a2f0fa6/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lixee.ts:1803:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17)
[2024-09-26 23:53:23] error: 	z2m: Failed to configure 'garage/gtl/tableau_elec/linky', attempt 1 (Error: Bind 0x00158d0005d243e2/1 seMetering from '0x00124b002a2f0fa6/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lixee.ts:1803:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))

### Zigbee2MQTT version

1.40.1-dev commit: 15d845b

### Adapter firmware version

20230507

### Adapter

Sonoff_ZBDongle-P

### Setup

docker on unraid

### Debug log

[log.log](https://github.com/user-attachments/files/17156463/log.log)

nco1512, 2024-10-01:
got the same problem

nco1512, 2024-10-01:
> me too, downgrade back to 1.39.0 solved the problem

could you share, how you downgraded?

peterbe1, 2024-10-01:
Make sure to backup your configuration before experimenting. What I did is 
1. renamed z2m installation directory, 
2. reinstalled 1.39.0 
3. copied back configuration file and database. 

``
mv /opt/zigbee2mqtt /opt/zigbee2mqtt2

sudo mkdir /opt/zigbee2mqtt
sudo chown -R ${USER}: /opt/zigbee2mqtt

git clone --depth 1 -b 1.39.0 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt

cd /opt/zigbee2mqtt
npm ci

npm run build

cp /opt/zigbee2mqtt2/data/configuration.yaml /opt/zigbee2mqtt/data/configuration.yaml

cp /opt/zigbee2mqtt2/data/database.db /opt/zigbee2mqtt/data/database.db

closed 2024-12-20 by github-actions[bot]