Koenkk/zigbee2mqtt #24117: Sinope TH1123ZB not reporting Current Voltage Energy and others correctly

geniusled, 2024-09-27:
### What happened?

Multiple values are not fetched correctly from the thermostats, they stay NULL and dont get periodically pushed to z2m. as result, in Homeassistant values stay UNKNOWN. theres also Errors when trying to setup reporting for these values
- Power
- current
- energy

### What did you expect to happen?

receive values for:
- Power
- current
- energy

correctly and being able to setup reporting for periodic updated values

### How to reproduce it (minimal and precise)

1. Pair thermostat
2. receive error prompts:
3. notice in Homeassistant and z2m device states tab, most values are empty or unkown until manually set
4. current, power, energy, voltage always stay at unknown/null unless some fiddling is done.

### Zigbee2MQTT version

1.40.1-1

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

Add-on in Homeassistant (Rpi 4 with HAOS)

### Debug log

error after pairing (and reconfiguring also)
> z2m: Failed to configure 'thermostat', attempt 7 (Error: ZCL command 0x500b9140000105a6/1 seMetering.read(["multiplier","divisor"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:443:29) at Object.readMeteringMultiplierDivisor (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:37:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/sinope.ts:722:13) at Configure.configure (/app/lib/extension/configure.ts:125:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:56:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

output in states after pairing
> {
"linkquality": 126,
"local_temperature": 24.37,
"occupied_heating_setpoint": 7,
"pi_heating_demand": 0,
"power": null,
"running_state": "idle",
"voltage": 2368,
"backlight_auto_dim": null,
"current": null,
"enable_outdoor_temperature": null,
"energy": null,
"keypad_lockout": null,
"main_cycle_output": null,
"outdoor_temperature_timeout": null,
"second_display_mode": null,
"temperature_display_mode": null,
"thermostat_occupancy": null,
"thermostat_outdoor_temperature": null,
"time_format": null
}

after trying to add reporting

> error 2024-09-27 01:18:08z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0x500x9140000105a6/1 haElectricalMeasurement from '0x00124b0030ca972a/242' failed (SRSP - ZDO - bindReq after 6000ms)'
info 2024-09-27 01:18:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{},"error":"Bind 0x500b9140000105a6/1 haElectricalMeasurement from '0x00124b0030ca972a/242' failed (SRSP - ZDO - bindReq after 6000ms)","status":"error","transaction":"81cgs-1"}'

jzielke84, 2024-10-03:
I've also noticed that for devices like TS011F.

geniusled, 2024-10-04:
Update:
i was able to update the firmware of the thermostats from 0.3.9 to 0.6.9
this has resolved the issue and it is now possible to setup reporting of Power/Voltage/Current

Note for anyone with sinope devices, firmware updates require to have the Sinope GT130 proprietary hub. the updates are automatically randomly pushed (took 32 hours until first unit updated itself)

closed 2024-10-04 by geniusled