Koenkk/zigbee2mqtt #24119: Issues with Green Power devices past 1.36.1

DennisJohnsen, 2024-09-27:
### What happened?

Hello

So i'm in a bit of a pickle atm. I'm currently running 1.36.1, but there is a fix in 1.40.1 for something else i would like. I digress.

But all versions past 1.36.1 (I've tried 1.40.1 and 1.40.0 and 1.38.0) causes over half of my Green Powered devices to stop responding with any action. They show up in the UI just fine, but no action is sent to MQTT when clicking them, but it seems to some of them. It seems random which ones work and dosn't work, nothing with range to the zigbee adapter. So i'm at a bit of a loss.

The devices in question: https://www.zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z

As soon as i revert back to 1.36.1 they all just work as expected again when Zigbee2Mqtt is running again.

I tried to interact with two switches in the log, one that works and one that doesn't. Not sure if makes a difference or not.
The one that works: 0x00000000017199a1
The one that doesn't work: 0x000000000171db5f

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

0x26690700

### Adapter

Raspbee 2

### Setup

Raspberry PI x86 via Docker

### Debug log

[log.log](https://github.com/user-attachments/files/17161352/log.log)

chris-1243, 2024-09-27:
You should check #23814. Try the latest-dev. Your problem should be fixed

DennisJohnsen, 2024-09-29:
Hmm, the same issue with `:latest-dev`. Only a few of my switches work. At least it's consistent the same switches that works/fails with these different versions than `1.36.1`

`:latest-dev` seems to be restarting though. Heres just a quick look at the logs without debug enabled:
```
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2024-09-29 10:04:12] error: 	z2m: Cannot find '0x00178801093e0030/11' of group 'Stue'
[2024-09-29 10:04:12] error: 	z2m: Cannot find endpoint '11' of device '0x00178801093e0030'
Error: waiting for response TIMEOUT
    at DeconzAdapter.checkReceivedDataPayload (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:718:28)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:77:18)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2024-09-29 10:06:11] error: 	z2m: Cannot find '0x00178801093e0030/11' of group 'Stue'
[2024-09-29 10:06:11] error: 	z2m: Cannot find endpoint '11' of device '0x00178801093e0030'
Error: waiting for response TIMEOUT
    at DeconzAdapter.checkReceivedDataPayload (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:718:28)
    at EventEmitter.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:70:18)
    at EventEmitter.emit (node:events:517:28)
    at parseReadReceivedDataResponse (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:361:27)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:566:32)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:515:53)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:63:25)
    at Driver.emit (node:events:517:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:881:14)
    at Parser.emit (node:events:517:28)
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2024-09-29 10:07:36] error: 	z2m: Cannot find '0x00178801093e0030/11' of group 'Stue'
[2024-09-29 10:07:36] error: 	z2m: Cannot find endpoint '11' of device '0x00178801093e0030'
```

I will try to add `legacy_triggers: true` to my config, and set logging info to debug and post a full log.

DennisJohnsen, 2024-09-29:
Still no dice. The same switches still fail/work on `:latest-dev`

I attached the log for this restart, where i interacted with both a working switch and a failing one. 
Same devices as mentioned above

> The one that works: 0x00000000017199a1
The one that doesn't work: 0x000000000171db5f

[log from `:latest-dev`](https://github.com/user-attachments/files/17178544/log.log)

DennisJohnsen, 2024-10-04:
I just tried `1.40.2` today, same issue.

DennisJohnsen, 2024-10-28:
Same issue as of 28-10-2024 with `latest-dev`.

chris-1243, 2024-10-28:
Could you try to factory reset one device not working and try to pair it.

Carefully read this 

https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices

Pair your device **without** using the `permit join all`. Choose a compatible router and use `permit join via router_x`

DennisJohnsen, 2024-11-14:
So i tried to remove a switch.

Put it into pairing mode as specified [here](https://www.zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z).
And not using permit all, but directly to coordinator, with direct line of sight. And nothing.

chris-1243, 2024-11-15:
> If i remove the battery from the switch. Hold Top left button to join my zigbee channel 15 network, and then confirm as per the documentation - it pairs just fine. With its random friendly name. But it's stuck in "Offline" and nothing is exposed when i click the buttons.

Which ZGP device do you try to pair ? Would you mind to try once but this time, do not use the coordinator but a router close to your device. This [list](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices) is not exhaustive.

DennisJohnsen, 2024-11-15:
I dont have any devices that is on that list though. So i tried with the cordinator, which works just fine on 1.36.1 - so i dont see why i have to specifically use one of those devices to pair to. Even pairing it just fine on 1.41.1, but nothing on exposed until i just roll software back to 1.36.1.

But i have these Ikea and Philips hue devices. Maybe you can help me indicate if one of those are considered a proper Green Power router:
**IKEA**

- [LED1537R6/LED1739R5](https://www.zigbee2mqtt.io/devices/LED1537R6_LED1739R5.html#ikea-led1537r6%252Fled1739r5)
- [LED1837R5](https://www.zigbee2mqtt.io/devices/LED1837R5.html#ikea-led1837r5)
- [LED1923R5/LED1925G6](https://www.zigbee2mqtt.io/devices/LED1923R5_LED1925G6.html#ikea-led1923r5%252Fled1925g6)
- [LED2104R3](https://www.zigbee2mqtt.io/devices/LED2104R3.html#ikea-led2104r3)

**Philips Hue**
- [LWG004](https://www.zigbee2mqtt.io/devices/LWG004.html#philips-lwg004)
- [929001953101](https://www.zigbee2mqtt.io/devices/929001953101.html#philips-929001953101)
- [9290024688](https://www.zigbee2mqtt.io/devices/9290024688.html#philips-9290024688)
- [929002471601](https://www.zigbee2mqtt.io/devices/929002471601.html#philips-929002471601)
- [8718699688820](https://www.zigbee2mqtt.io/devices/8718699688820.html#philips-8718699688820)
- [8719514279131](https://www.zigbee2mqtt.io/devices/8719514279131.html#philips-8719514279131)
- [1746630P7](https://www.zigbee2mqtt.io/devices/1746630P7.html#philips-1746630p7)
- [1746230V7](https://www.zigbee2mqtt.io/devices/1746230V7.html#philips-1746230v7)

chris-1243, 2024-11-17:
From version 1.36 to 1.41, the dev might have done some changes. `ember` driver had several changes and a huge improvement to pair ZGP devices through specific routers for example.

The list provided is not exhaustive at all. It is based on users who checked the compatibility. I added some based on what I have. What would be awsome is if you could try yours and check the output and maybe extend this list. Do not forget a lot is done by the dev and the community. It is a way to improve the software we are using.
By experience, `HUE` device are able to translate easily ZGP frames. There is one issue with them. Depending on the model and firmware, the data might be transmitted by `unicast` (much better) or `broadcast`. #22897
Recent IKEA devices seem to implement well ZGP translation as `unicast`.
You should avoid to pair your device as `broadcast` as your network might become slow and there is a limit. Only 7-8 messages/second are sent. More messages will be queued and you will see the latency. Honestly and when you pair your device via the coordinator, I do not know how it is done (unicast/broadcast). The only way to know is to [sniff](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) your traffic.

Would you mind specify which ZGP devices is causing issues? We may then check the converters or deeper in the code to find the errors. A debug log while pairing a device not working could help as well. It might be outside of my knowledge but at least we would have data to work with.

DennisJohnsen, 2024-11-17:
**I would love to help expand on that list with the devices i have - but i would need some a step by step guide** to check the output of the devices so i can confirm if they use `broadcast` or `unicast` for green power. Would also make me able to see if i can pair one of my failing devices directly to a `unicast` device instead and see if that helps.

The only greenpower devices i have is [this switch](https://github.com/Koenkk/zigbee2mqtt/issues/24119#issuecomment-2481091486). 

It's always the same switches that works/fail when moving from `1.36.1` to something more recent.

Luckily i havn't had any "slow" devices, even though i have a pretty big Zigbee network.
```
Total 116
Router: 54
End devices: 45
GreenPower: 17
```
I have, because of just reading that most Phillips Hue is required for greenpower devices, scattered Phillips Hue bulbs strategically around the house. So they might just be `unicast` since i havn't experienced issues with performance expect when some updates came along that broke something.

I will get on the log file while paring (to the coordinator) a switch that doesn't work on `1.41.0` as soon as possible.

DennisJohnsen, 2024-12-05:
This is odd.. I don't have that many other buttons than my greenpowered devices - so i didn't initial notice this issue.

But on `1.42.0` i experienced the same with [this non Green Power device](https://www.zigbee2mqtt.io/devices/TS0043.html#tuya-ts0043)

It was also shown as `online` but no `action` was being registered by the device. As soon as i jumped back to `1.36.1` it started again. So this might not be a Green Power only issue?

closed 2025-02-12 by github-actions[bot]