Koenkk/zigbee2mqtt #24120: Tuya ZG-227Z _TZE200_a8sdabtg reports battery always 100%

mleone87, 2024-09-27:
### What happened?

I have 5 devices of this kind, all of them always report 100% battery. 3 out of 5 stopped working silently because the battery went to 2600mV or similar.

### What did you expect to happen?

Battery should report at least an estimation

### How to reproduce it (minimal and precise)

Just use this device and see

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20230507

### Adapter

Zdongle-E

### Setup

Docker

pmpbpl, 2024-10-16:
Same here.
Firmware: 1.40.2
Hardware: zStack3x0 20230507

mleone87, 2024-10-17:
`"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}`

the battery report seems skewed

Koenkk, 2024-10-18:
The number needs to be divided by 2, so 200 = 100% battery

mleone87, 2024-12-19:
PR submitted

closed 2024-12-20 by Koenkk

nephrotranz, 2025-01-05:
Unfortunately I would like to reopen this thread.

I also own three of these devices and they all showed a battery level of 100% since 23rd of January 2024 until the update to Z2M 2.0.0.
After the update the battery levels of all three devices suddenly went down to 50% and - although I read the release note -  I found this a bit suspicious.

Because of that I put a brand new battery into one of the devices and waited 12 hours now but the battery level is still at 50%. At the same time, because I was curious, i did put the old battery from this device into a Hue dimmer v1 and this one went up from 34% to 100% within a few hours.

**So battery level really is still at 100% after nearly one year.**

In the database.db my devices also show a voltage value which is 29 or 30 for all three batteries. As 3 V is the voltage of a CR2450 battery at 100% that also shows that they are measuring correctly.

Additionally I put one of the thermometers into our freezer last August and the battery level went down to 10% when it was at approx. -19°C for a fews days, so it was definitely working.

I can only imagine that @mleone87 had some faulty devices or something like that.