Koenkk/zigbee2mqtt #24133: [Feature request]: Dissable ROUTING capability or make JOIN with Coordinator stick.

ortofan, 2024-09-29:
### Is your feature request related to a problem? Please describe

Zigbee network fails during power outage. Despite most nodes being battery powered and IT infrastructure UPS protected. 
There are many use cases my is described below in "Additional Conext". We got to fix this somehow. 

### Describe the solution you'd like

Make the Join options stick. Like almost everyone my understanding was that if and when you use Join with a specific node it will stick. The ideal solution would be exactly that. When selecting a specific node to join with It should be permanent. Otherwise not even sure what the purpose of this options is, since afte a while the MESH will force routes as it sees fit. 

### Describe alternatives you've considered

Turn off Routing capabilits on specific nodes.  This may only be possible if every single manufacture would adhear to this in their firmware. (Zigbee alliance are you listening?). 

### Additional context

You have frequent power outages (with eather lately this is becoming more common). During a power outage your IT infrastructure is protected by a UPS including PoE switches that power Zigbee PoE routers, cameras etc..So everything works for a few hours os as long as your batteries last (mine is 8 hours). However, your entire carefully desinged Zigbee network goes down since Zigbee "smart" MESH is basically forcing routes with AC connected devices. So, you can not use such system as a security system for example, but there may be even more drastic functionality loss (unable to open a door etc..) due to this.

Koenkk, 2024-09-29:
This is not supported in the Zigbee protocol, so not something that can be implemented by z2m

ortofan, 2024-09-29:
I was hoping that someone from the zigbee alliance reads this, or perhaps you may have some contacts or provide some pointers on how to get this implmeneted, how to push the key players to understand how limiting this is. I even tried some dedicated routers (coordinators in router mode), nope. Plugs and Bulbs are still prefered even if the signal qiality is far worse. This is a massive limitation in Zigbee and I can not fathom why this is being forced in such fashion.

ortofan, 2024-09-29:
This is the best I could come up with https://www.reddit.com/r/ZigBee/comments/1fs3xs3/zigbee_achilles_heel/

closed 2024-12-20 by github-actions[bot]