Koenkk/zigbee2mqtt #24137: SLZB-06P7 Fatel Error Ember

Blackduke77, 2024-09-29:
### What happened?

I have been have stability issues recently so thought I would try and use ember and see if that improves things. I am using a SLZB-06P7 and when I switch config to ember, I receive a HOST_FATEL_ERROR in the logs and the service does not start.

[2024-09-29 12:39:47] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-29 12:39:47] info: 	z2m: Loaded external converter 'DoubleSocket50AU3.5.js'
[2024-09-29 12:39:47] debug: 	z2m: Loaded state from file /app/data/state.json
[2024-09-29 12:39:47] info: 	z2m: Starting Zigbee2MQTT version 1.40.1 (commit #403d3c0)
[2024-09-29 12:39:47] info: 	z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-29 12:39:47] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":62533,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":false,\"path\":\"tcp://192.168.40.31:6638\",\"adapter\":\"ember\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false,\"transmitPower\":20}}"'
[2024-09-29 12:39:47] info: 	zh:ember: Using default stack config.
[2024-09-29 12:39:47] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":62533,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"tcp://192.168.40.31:6638","adapter":"ember"},"adapter":{"disableLED":false,"concurrent":null,"delay":null,"transmitPower":20},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
[2024-09-29 12:39:47] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-09-29 12:39:47] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-09-29 12:39:47] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: Opening TCP socket with 192.168.40.31:6638
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: Socket connected
[2024-09-29 12:39:47] info: 	zh:ember:uart:ash: Socket ready
[2024-09-29 12:39:47] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-09-29 12:39:47] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2024-09-29 12:39:48] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2024-09-29 12:39:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2024-09-29 12:39:50] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-09-29 12:39:50] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2024-09-29 12:39:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2024-09-29 12:39:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2024-09-29 12:39:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2024-09-29 12:39:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2024-09-29 12:39:52] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[... 306 lines cut ...]
[2024-09-29 12:39:58] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2024-09-29 12:39:58] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2024-09-29 12:39:58] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2024-09-29 12:39:59] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2024-09-29 12:40:00] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2024-09-29 12:40:00] error: 	z2m: Error while starting zigbee-herdsman
[2024-09-29 12:40:00] error: 	z2m: Failed to start zigbee
[2024-09-29 12:40:00] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-29 12:40:00] error: 	z2m: Exiting...
[2024-09-29 12:40:00] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:701:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1708:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:138:29)
    at Zigbee.start (/app/lib/zigbee.ts:64:27)
    at Controller.start (/app/lib/controller.ts:140:27)
    at start (/app/index.js:154:5)

Koenkk, 2024-09-29:
06P7 is a zstack adapter (Texas instruments) you cannot is use the ember (Sillicon labs) driver with this

Blackduke77, 2024-09-29:
ahh thanks, which one should i switch to?

Blackduke77, 2024-09-29:
Looks like the SLZB-06M is the one I should be getting, if I understand correctly the ember driver is the future supported driver

indexample, 2024-09-30:
> ahh thanks, which one should i switch to?

adapter: zstack

closed 2024-12-20 by github-actions[bot]