Koenkk/zigbee2mqtt #24140: CK-BL702-AL-01(7009_Z102LG03-1 tuya smart bulbs goes offline

ankokdog, 2024-09-29:
### What happened?

Z2M v 1.40.1 and new zstack firmware for sonoff usb dongle p. after that 4 same bulbs goes offline after about 10 minutes. when i switch off - on works for some minutes and then the same thing happens. and i get error

z2m: Failed to ping '0x7cb94c7889840000' (attempt 1/1, ZCL command 0x7cb94c7889840000/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 14881 - 1 - 19 - 0 - 1 after 10000ms))

the bulbs are these

Device type
Router
Zigbee Model
CK-BL702-AL-01(7009_Z102LG03-1)
Zigbee Manufacturer
eWeLink
Description
Zigbee LED bulb
Support status
Supported

IEEE Address
0x7cb94c788be30000
Network address
0x8509 / 34057
Firmware build date
20230105
Firmware version
1.2.3
Manufacturer
[Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
Model
[CK-BL702-AL-01](https://www.zigbee2mqtt.io/devices/CK-BL702-AL-01.html#tuya-ck-bl702-al-01)

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

sonoff usb3 dongle p cc2652p

### Setup

pc with i5 8th gen 16gb ram and m2 ssd 512gb

mremedi2023, 2024-12-03:
Same problem here

moekli, 2024-12-22:
With z2m version 1.42.0-2 i can confirm the problem with gu10 spots an e14 bulbs.
Anyone an idea?

evil-life, 2025-01-24:
Same here... I already upgraded to z2m 2

walberjunior, 2025-03-29:
Same here with 2 lamps, but using the latest version of Z2M

CK-BL702-AL-01(7009_Z102LG03-1)
```
Zigbee2MQTT version
2.1.3 commit: unknown
Coordinator type
zStack3x0
Coordinator revision
20240710
Coordinator IEEE Address
0x00124b00258db1f9
Frontend version
0.9.4
zigbee-herdsman-converters version
23.2.0
zigbee-herdsman version
3.2.7 
```

janaxhell, 2025-05-12:
Same here with even newer version, 2 bulbs that keep going offline for no reason. 

```
Zigbee2MQTT version
2.3.0 commit: unknown
Coordinator type
ZStack3x0
Coordinator revision
20240710
Coordinator IEEE Address
0x00124b002f8aabb4
Frontend version
0.9.10
zigbee-herdsman-converters version
23.36.0
zigbee-herdsman version
4.0.0
```

closed 2025-09-09 by github-actions[bot]