Koenkk/zigbee2mqtt #24141: [Wrong device]: Water leak detector detected as SNZB-03

robygi, 2024-09-29:
### Link

https://it.aliexpress.com/item/1005006966713885.html?spm=a2g0o.order_list.order_list_main.29.42823696auD73k&gatewayAdapt=glo2ita

### Model

SQ510A

### Description

Zigbee Water leakage detector

### Vendor

Ewelink

### Picture (link)

https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector_1.html

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xa4c138f219c054ff","nwkAddr":7582,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"manufacturerName":"eWeLink"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x8c65a3fffeb0972b","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":27,"batteryPercentageRemaining":182}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffeb0972b","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffeb0972b","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-268385843},"lastSeen":1727610099649,"checkinInterval":3600}

### Notes

The device is paired without problems but it is detected as a movement sensor SONOFF SNZB-03. The sensor is functioning, signaling Clear when no water is sensed and Occupied when water is sensed.
Is there anything I could do to make it recognized as it should be?

KarlAzytzeen, 2024-10-11:
Same here. 

> 
const {battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SNZB-03'],
    model: 'SNZB-03',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
> 

Should be identified as this: [https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html](url)
or this [](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector_1.html)
mine is the second one butr at logic and software level must be the same.

Koenkk, 2024-10-11:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

KarlAzytzeen, 2024-10-12:
Thansk for your fast support.
Sorry, didnt work. 
Here are images about how it is just in case i did something wrong. 
I fully reset the HA server. It is running HAOS bare metal. Sonnof dongle on ember driver.
If you need any more info to adress this issue, please, let me know.

[image]

[image]

[image]

Koenkk, 2024-10-12:
@KarlAzytzeen could you provide the data/database.db entry of your device?

robygi, 2024-10-12:
@Koenkk  I tried what you suggest but also for me nothing changed. The description is empty, so the external converter seems didn't has effect.

KarlAzytzeen, 2024-10-12:
> @KarlAzytzeen could you provide the data/database.db entry of your device?

Thanks for your support again.
I think this is what you are asking for. 
If not, please provede some aditional instructions on how to provide the info you need/ if its need to anonimize any info.

database.db
`{"id":36,"type":"EndDevice","ieeeAddr":"0xa4c13886a6029afb","nwkAddr":50608,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SNZB-03","manufacturerName":"eWeLink","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x84b4dbfffec79f78","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec79f78","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec79f78","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-268385843},"lastSeen":1728747452691,"checkinInterval":3600}`

Didnt see any data.db file.

TY

robygi, 2024-10-13:
@Koenkk  after testing the external converter as you suggested I noted that the sensor wasn't changing anymore its status in the z2m dashboard. 
However it was sending mqtt messages on each status change.
I removed the file ext_converter.js and the modifications to configuration.yaml, than I restarted the z2m container.
I deleted the sensor from z2m and I repaired it again.
The sensor send out mqtt messages but doesn't update the z2m dashboard.
What is wrong now?

KarlAzytzeen, 2024-10-14:
Here is a link of the sensor. Just in case it can help.

[aliexpress.com/item/1005006966713885.html?spm=a2g0o.order_list.order_list_main.219.21da194dbzi0pa&gatewayAdapt=glo2esp](url)

Koenkk, 2024-10-14:
Could you provide a screenshot of the device -> about page with the ext converter?

KarlAzytzeen, 2024-10-15:
> Could you provide a screenshot of the device -> about page with the ext converter?

Of course.
[image]

also the state of the configuration file at this moment.
[image]

I did try modfing indentations etc but z2m refuses to start if it is not write down as you told me.

Koenkk, 2024-10-16:
Found the issue, updated https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9

KarlAzytzeen, 2024-10-16:
> Found the issue, updated https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9

Seems to be working with little caveats. 

Compared to this
https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector_1.html
it is missing the "tamper" detection. Anyway dont know how it is implmented as i didnt see any way for the device to detect manipulation. Also "battery low" is mising too.
Zigbee model continues as SNZB-03.
Also the device image is missing.

buuuuut....
[image]
[image]

seems to be working correctly.

Thanks for your patience and hope its easy to intregrate this and deploy it.

robygi, 2024-10-18:
I confirm what @KarlAzytzeen said. The converter is functioning also on my side with the same caveats. @Koenkk thanks a lot for what you are doing.
However I have two more issues:
1) the device doesn't report in real time water_leak status in z2m dashboard but it publishes in real time on mqtt 
2) mqtt info published by the device contain the new `water_leak` attribute but also `occupancy,` `tamper` and `battery_low`
Iissue 1) could be not related to the external converter but to my installation (even though I can't figure what it could be, any advice is appreciated)

Koenkk, 2024-10-19:
Updated https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9, this adds `tamper`. I see the `battery_low` in the exposes so it seems to be there?

Picture will be fixed once this is merged.

KarlAzytzeen, 2024-10-23:
> Updated https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9, this adds `tamper`. I see the `battery_low` in the exposes so it seems to be there?
> 
> Picture will be fixed once this is merged.

And Waht about battery as a numeric value?
TY

KarlAzytzeen, 2024-10-25:
Nope, only 4 values are exposed, as you can see. 
If any other affected can comment on how they have it...?
[image]

robygi, 2024-10-25:
I have five values exposed. The battery is a percentage value but I think is correct.
[image]

Koenkk, 2024-10-25:
I've added the device, @KarlAzytzeen please check if everything is there with the next release.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `SQ510A.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

KarlAzytzeen, 2024-10-26:
> I've added the device, @KarlAzytzeen please check if everything is there with the next release.
>
> Could you also submit a picture for the docs? This can be done by:
>
>     1. Create a fork by clicking [...]

Done.

beckynet, 2024-11-01:
After Update to 1.41.0 my motion sensor SNZB-03 is now detected as a Leak Detector.
In version 1.40 and before it is detected as a Motion detector

[SNZB-03.json](https://github.com/user-attachments/files/17600545/SNZB-03.json)
[image]
[image]

kennethpangky, 2024-11-02:
This must be why my kitchen lights decided to stay on for the past few days.

melnibon, 2024-11-02:
Hello everyone,

I am sorry to say that this patch has broken true Sonoff SNZB-03 motion detectors.... I have four of those detectors, and now they are all detected as eWeLink Water leak sensor.

Could you check how to distinguish between these two kind of devices, and bring my motion detectors back ?

Thank you by advance,
Pierre

Koenkk, 2024-11-02:
@beckynet could you provide the data/databse.db entry of a real SNZB-03?

Koenkk, 2024-11-02:
Seems the issue is not fixable as the device shares the same fingerprint as the real SNZB-03.

soif, 2024-11-03:
Hi guys.

Same issue here!

Since 1.41, my motion sensor (that was seens as  SNZB-03, even if it looks more like a very common AE round sensor, as shown in the [photo](https://github.com/Koenkk/zigbee2mqtt/issues/24141#issuecomment-2451966226) of @beckynet ) has now changed to this water leak sensor, and thus misses the occupancy sensor.

If the fingerprint is the same, you have to choose which device is rendered in Z2M and I guess that they are certainly way more people with motion sensors (ie the SNZB03) than people with a water leak sensor. 

For this reason, I vote to reverse this commit 🍺

Koenkk, 2024-11-03:
Already reverted it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

soif, 2024-11-03:
Really impressed by your speed. 😲 

You guy simply rock ! 👍

szwarx, 2024-11-04:
Hey, I have the same problem, my motion sensor is recognized as a water leak, what should I do to make it work?

KarlAzytzeen, 2024-11-05:
Is there any way can later manually force to be recognized as the water leak sensor instead the "default" motion sensor?
Maybe some kid of option to switch or something like that? 
TY

soif, 2024-11-05:
@szwarx 
> Hey, I have the same problem, my motion sensor is recognized as a water leak, what should I do to make it work?

Simply wait until @Koenkk merge his dev branch into the master, and release it. 

_Then updating Z2M to the latest version will fix your/our issue 😎_

Koenkk, 2024-11-06:
Next release is 1 December

edudele, 2024-11-11:
> Next release is 1 December

Back to koenkk/zigbee2mqtt:1.40.2

kennethpangky, 2024-11-13:
How often does such things happen eh? That 2 different devices have the same identifier? Any way to mitigate in future?

For me it's merely an inconvenience as they're backups to working setups, but I can see how this might be an issue for others. And having to wait for nearly a month or be forced to switch to the dev branch? Not sure about the procedure to jump between branches but should be a headache haha.

melnibon, 2024-11-13:
Hello everyone,

Would it be possible to expose all the sensors of both types of devices, as it was only one unique kind of device? Those who have the motion detector would see motion sensor as active, and those who have a water leak detector would see related sensors. And the other sensors, not available, would simply be not available, and we can choose to hide them in HA.

Sorry if my idea does not help.

Best regards,

eknirk, 2024-11-30:
@melnibon your idea would not work. The detection as a water leak sensor is causing havoc when used with homebridge! Water leak alarms cannot be silenced or disabled in HomeKit.

scuppasteve, 2024-12-02:
The production update came out, these sensors now show properly as a motion sensor. But, they also report some states i do not think they have like water leak.

{
    "battery": 97,
    "battery_low": false,
    "last_seen": "2024-12-02T09:47:13-08:00",
    "linkquality": 0,
    "occupancy": true,
    "tamper": false,
    "water_leak": true,
    "voltage": null
}

Qwietse, 2024-12-02:
How do I update the zigbee2mqtt thing so that the sensor now shows up correct

eknirk, 2024-12-02:
They are now detected as Occupancy sensors in Home Assistant and Homebridge. Shouldn't it be a Motion sensor (not an Occupancy sensor)?

davidrpfarinha, 2024-12-04:
Folks, I don't know why, but on this old version, it looks like that I have both working as expected.
[image]
[image]

Axlfoley2, 2024-12-19:
I have the same Problem here but in the other Way. My Leak Sensor is now shown as the Motion Sensor.
I got the  Zigbee2MQTT v1.42.0-2 - 0.0 B Version.

OleksandrNastashenko, 2024-12-19:
> I have the same Problem here but in the other Way. My Leak Sensor is now shown as the Motion Sensor. I got the Zigbee2MQTT v1.42.0-2 - 0.0 B Version.

me too

davidrpfarinha, 2025-01-13:
Hey folks, what is the current state of this issue under the version 2.0.0?

DavidAntonin, 2025-01-19:
@davidrpfarinha Under the version 2.0.0 my water leak detector appears as motion sensor. Is this solvable?

tomtpcz, 2025-01-24:
I have the same Problem. Z2M version 2.0.0. Water leak detector Tuya TS0207 appears asi motion sensor eWeLink SNZB-03

Yves-Be, 2025-02-05:
On 2.0.0, you should place the external converter from https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9 in the external_converters folder of your data folder. (Create the folder if it does not exist, same level as configuration.yaml)

madmi1, 2025-02-14:
> On 2.0.0, you should place the external converter from https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9 in the external_converters folder of your data folder. (Create the folder if it does not exist, same level as configuration.yaml)

thanks a lot, it works

[image]
[image]

davidrpfarinha, 2025-03-13:
> On 2.0.0, you should place the external converter from https://gist.github.com/Koenkk/e88e795bdd59bfc7da73005d13ff51c9 in the external_converters folder of your data folder. (Create the folder if it does not exist, same level as configuration.yaml)

This worked for me! Thanks for the tip. 
Is this a final solution or temp workaround?

smhc, 2025-03-26:
This problem appears to be back, presumably after the fix was reverted: https://github.com/Koenkk/zigbee-herdsman-converters/commit/8cbca454f1922dada30d651f79177b8055c7c097

If I use the ext.js from here it seems to work ok.

tiimjcb, 2025-05-18:
i confirm, the problem is back.

mech78, 2025-07-05:
Same for a flood sensor connected to Homey. Also detected as SNZB-03 while connecting as generic Zigbee device, not even using zigbee2mqtt..

Mirarkitty, 2025-08-02:
Happily occupancy is reported instead of water leak, so it's possible to use that value instead, just wrong label.

djborsi, 2025-09-05:
I have the same issue.
Yesterday shipment from China arrived with flood sensor SQ510AA and it is recognized as "SNZB-03_eWeLink".

Any news for this issue? Is it in focus to be solved soon?

dbmaxpayne, 2025-09-30:
Same here. I haven't tried the custom converter, though.
I will wait for an upstream fix for now.

xicoxicao, 2025-11-08:
Hey, add me to the list. And I can't use the converter because I have a SNZB-03 motion sensor so to fix one I have to loose the other. A this is today after the updated to the new 2.6.3-1 version. I'm stuck

davidrpfarinha, 2025-11-10:
> Hey, add me to the list. And I can't use the converter because I have a SNZB-03 motion sensor so to fix one I have to loose the other. A this is today after the updated to the new 2.6.3-1 version. I'm stuck

Actually, I have both sensors, and they still work properly, but running 2.6.3

[image]

[image]