Koenkk/zigbee2mqtt #24148: Aqara TVOC (VOCKQJK11LM) - Unable to reconfigure

Munsio, 2024-09-29:
### What happened?

When trying to reconfigure the Device I get the following error:
```
z2m: Failed to configure 'Office - TVOC', attempt 4 (Error: Bind 0x54ef441000106b9b/1 genPowerCfg from '0x84b4dbfffebc4762/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4160,"groupId":0,"sequence":54}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:101:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:480:42) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4982:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:738:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21))
```

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Home Assistant Add-On

piwi3910, 2024-10-06:
I can confirm, I went from the home-assistent yellow build-in adaptor on ember, to the new SLZB-06 on zstack
And all my battery powered Aqara devices get detected but all give this error during configuration, that includes my airsensors, Motion sensors and Doorsensors.
Some of them after many retries do success,

closed 2024-12-20 by github-actions[bot]

xelemorf, 2025-01-07:
I just bumped into the same issue but with zstack on SLZB-06 coordinator , FW 20240710 (with baudrate 115200 and on 20240315 (with baudrate 460800) while using Z2M v1.42.0-2 and also with the latest stable v2.0.0 (also tried Edge v2.0.0).

```
[2025-01-07 17:57:10] error: 	z2m: Failed to configure 'bedroom-tvoc', attempt 1 (Error: Bind 0x54ef4410007953de/1 genPowerCfg from '0x00124b002b481b65/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:403:47
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZdo (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:311:16)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:30)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:3483:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13))
```

piwi3910, 2025-01-08:
I fixed it by running the firmware with sdk6.2 and not sdk7.0 on the coordinator.

Internal Use - Confidential

[EXTERNAL EMAIL]

xelemorf, 2025-01-08:
The firmware I downgraded to is based on SDK 6.10, all of the older firmwares are still SDK 6.10 based, I don't see SDK 6.2 based ones available on SLZB-06 FW page.

The one I currently have is the below one:
```
Dev firmware
Revision: 20240315
Release notes
Public Patched Coordinator release for CC2652P chips [15-Mar-2024]:
- For source, please read: https://github.com/Andrik45719/Z-Stack-firmware;
- SDK 6.10 based;
- Patched version of Public Coordinator release for CC2652P chips [15-Mar-2024] - baudrate changed to 460kbps from 115kbps;
- IMPORTANT!!! Baudrate: 460800;
- May help if 'spamming' devices are in Zigbee network;
Terms of use - by downloading and installing this firmware, you agree to the aforementioned terms
```.

xelemorf, 2025-01-08:
Downgraded SLZB06 core to the latest stable 2.3.6 (previously was using the latest Dev) and Zigbee FW to the oldest available 20221226, noticed some "improvement" as the Air Quality showed an "excellent" value but the rest of the sensors were "Unavailable" still, leading to essentially the same situation, the TVOC sensor no longer checked in with button press.
Until there is a solution I have switched this sensor to an E1 hub and connected through Homekit, so at least I can use it.