Koenkk/zigbee2mqtt #24149: [New device support]: TS0601 manufacturer _TZE284_nlrfgpny solar siren

peroanjo, 2024-09-29:
### Link

https://es.aliexpress.com/item/1005007022197993.html

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0xa4c138e5cdd4c8e4","nwkAddr":62092,"manufId":4417,"manufName":"_TZE284_nlrfgpny","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"�S�.f�S�.\u0012","65506":56,"65508":0,"appVersion":77,"stackVersion":0,"dateCode":"","modelId":"TS0601","manufacturerName":"_TZE284_nlrfgpny","powerSource":3,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1727637030467}

### Zigbee2MQTT version

 1.40.1

### Comments

Hello everyone,

It seems like this device is already supported as NAS-AB06B2 in the official z2m release. However _TZE284_nlrfgpny manufacturer is not registered.

Also, in the vendor page, there are comments of people saying it works with z2m. So it would seem like the seller recently modified the manufacturer id and is not recognized by z2m anymore.

Any help will be kindly appreciated!

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_nlrfgpny',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I tried using an external definition I found in #19827. However I couldn't get it to work...

averter, 2024-10-03:
Hello, indeed I am experiencing the same issue with a siren from the same vendor. Thanks for mentioning issue #19077 as it might be relevant to some extent.

averter, 2024-10-05:
Recently updated to 
Zigbee2MQTT version
    [1.40.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2) commit: [e06848db](https://github.com/Koenkk/zigbee2mqtt/commit/e06848db)

Coordinator type
    zStack3x0

Coordinator revision
    20230507

Coordinator IEEE Address
    0x00124b002c3a6118

Frontend version
    0.7.4

zigbee-herdsman-converters version
    20.21.0

zigbee-herdsman version
    2.1.3

and when I try to interview there is a new log message:

2024-10-05 07:33:07zh:controller:device: Device '0xa4c1382f18ffcfc2' is only compliant to revision '21' of the ZigBee specification (current revision: 23).

Any thoughts?

peroanjo, 2024-10-06:
For me I can interview the siren just fine and I won’t see any errors. Are you getting this error consistently? I saw the issue with the new manufacturer has also been mentioned in https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-10803048 #17325

averter, 2024-10-07:
> For me I can interview the siren just fine and I won’t see any errors. Are you getting this error consistently? I saw the issue with the new manufacturer has also been mentioned in [#17325 (comment)](https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-10803048) #17325

Based on the comment in #17325 could it be as simple as adding `_TZE284_nlrfgpny` to [this line](https://github.com/Koenkk/zigbee-herdsman-converters/blob/0fcc1eb52847c299d32fd80387131d9ea75bec1f/src/devices/neo.ts#L47)? As in

`        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_t1blo2bj','_TZE284_nlrfgpny', '_TZE204_t1blo2bj', '_TZE204_q76rtoa9']),`

peroanjo, 2024-10-07:
I think so, yes. But I haven’t been able to test it yet, as I am using the HassOS addon and don’t have access to the z2m internal files. Are you able to modify it?

averter, 2024-10-07:
Bingo!
[image]

I didn't even had to re-interview; simply had to restart zigbee2mqtt. 
Specifically, I'm using docker on a raspberry pi and had to change the above file, which is located in `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/neo.js`
I'll create a PR as soon as possible. Thanks for directing me to the comment which provided the solution :-)

averter, 2024-10-07:
Can't test those atm as I am not at home. I'd also note that the battery hasn't shown up yet, which is also known as a long-standing issue-

gregoriusus, 2024-10-11:
Just for info. For me everything is working. Battery is shown, changing alarm mode is working. Melody haven't tested yet. Maybe it also depends on manufacturer.
[image]

averter, 2024-10-11:
@gregoriusus Interesting...just to confirm, your manufacturer is `_TZE284_nlrfgpny`?

gregoriusus, 2024-10-11:
Hmm, no, I have:
[image]

averter, 2024-10-12:
@gregoriusus ok then it's not applicable to our case since it's not the same manufacturer.
@peroanjo I've now fully tested the siren and the alarm melody and alarm mode work perfectly. The only thing which does not work is the Battery SoC.

averter, 2024-10-12:
There seems to be an active development on the battery issue though. See https://github.com/Koenkk/zigbee2mqtt/issues/19827

peroanjo, 2024-10-12:
@averter that’s good news! I have switched to the dev branch and the siren got recognized perfectly, however I haven’t been able to test it to be working correctly as my siren ran out of battery and I don’t have physical access to it for now.
Thank you for your insight, I think we can then close this issue as the battery problem seems to be a general issue with this model.

closed 2024-12-19 by github-actions[bot]

ioanvladcristian, 2025-02-07:
Now working with Zig2MQTT.
Tested with this device: 
https://www.aliexpress.com/item/1005007022197993.html?spm=a2g0o.order_list.order_list_main.76.5ff81802HxyZ7R

averter, 2025-04-09:
> Now working with Zig2MQTT. Tested with this device: https://www.aliexpress.com/item/1005007022197993.html?spm=a2g0o.order_list.order_list_main.76.5ff81802HxyZ7R

But without the battery I presume? :-)

ioanvladcristian, 2025-04-09:
> > Now working with Zig2MQTT. Tested with this device: https://www.aliexpress.com/item/1005007022197993.html?spm=a2g0o.order_list.order_list_main.76.5ff81802HxyZ7R
> 
> But without the battery I presume? :-)

Correct!

averter, 2026-01-17:
Wow. I can't believe it but the battery percentage HAS FINALLY SHOWN UP! My manufaturer is the same as before i.e. `_TZE284_nlrfgpny` which was unreported as working as the sirens from @gregoriusus  and others, which were of different manufacturers.

[image]

[image]

I can confirm that their strategy is the way to go, i.e., keep the siren out of direct sunlight, and very importantly as well, keep constantly activating the lights to drain the battery as much as possible. I ended up doing this last step manually, but as said, it'd be useful to be able to do this with a script in an automated way. Hope this helps others.

tomiisp, 2026-02-02:
I have exact same Zigbee model/Model and still battery reported as Null% not matter what I try to do (reconfig, repair, discharge (it was about 2 months active in closed box) ) and still no battery level... 

> Wow. I can't believe it but the battery percentage HAS FINALLY SHOWN UP! My manufaturer is the same as before i.e. `_TZE284_nlrfgpny` which was unreported as working as the sirens from [@gregoriusus](https://github.com/gregoriusus) and others, which were of different manufacturers.

> I can confirm that their strategy is the way to go, i.e., keep the siren out of direct sunlight, and very importantly as well, keep constantly activating the lights to drain the battery as much as [...]

averter, 2026-02-03:
> I have exact same Zigbee model/Model and still battery reported as Null% not matter what I try to do (reconfig, repair, discharge (it was about 2 months active in closed box) ) and still no battery [...]

Sorry to hear that. Note that leaving the siren out of direct sunlight is not enough to make it discharge. The siren is incredibly  energy-efficient, and so unless you manually discharge it multiple times when out of sunlight it won't reach the 98% threshold. I spent a lot of time turning its lights on until the 98% showed up (I had lost hope to be honest). Good luck