Koenkk/zigbee2mqtt #24151: Zemismart Energy Meter SDM01 stopped working after updating Tuya firmware

rp2019433, 2024-09-29:
### What happened?

I have two Zemismart Energy Meter SDM01 integrated with Home Assistant and Z2M. Both were working properly, but after updating one of them with the most recent Tuya firmware (1.0.6), all the values were messed up. It looks like value mapping was changing in this new firmware version.

Voltage, current, power and all other values are messed up. Before updating firmware, everything was working as expected.

[image]

### What did you expect to happen?

All the values should be present correctly.

### How to reproduce it (minimal and precise)

Pairing SDM01 device with Tuya Zigbee hub and updating its firmware to the most recent (1.0.6).

### Zigbee2MQTT version

1.40.1-1

### Adapter firmware version

7.4.3 [GA]

### Adapter

Sonoff Dongle-E

### Setup

Add-on on Home Assistant

### Debug log

Info 2024-09-29 20:28:03z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Medidor Energia 3F', payload '{"ac_frequency":0,"current_a":2.2,"current_b":null,"current_c":0.02,"energy":0.6,"energy_a":22,"energy_b":1.28,"energy_c":null,"energy_produced_a":6.73,"energy_produced_b":null,"energy_produced_c":0.86,"last_seen":"2024-09-29T20:28:03-03:00","linkquality":188,"power":null,"power_a":675,"power_b":null,"power_c":0,"power_factor":null,"power_factor_a":20,"power_factor_b":0,"power_factor_c":null,"produced_energy":null,"voltage_a":1.5,"voltage_b":12.8,"voltage_c":8.6}'

EcoHouseThailand, 2024-10-06:
My first post here so apologies if I am not following the posting rules.

I have the same problem. I installed a Zemismart Energy Meter SDM01 yesterday adding it to a Zemismart M1 Zigbee hub then Home Assistant. At first it worked fine, then I noticed that an update for the SDM01 was available. After installing it showed a fixed power consumption and no history 

Before: 
[image]
[image]

After:
[image]
[image]

EcoHouseThailand, 2024-10-07:
I tested the connection between the SDM01 and the Zigbee hub:

[image]

EcoHouseThailand, 2024-10-07:
To fix this I removed the device from the Tuya App and did a factory reset by holding down the reset button. 

I added the device back to the Tuya app and I renamed the device to a different name to before

I removed the cards I had used in HA and restarted HA then added back the cards searching using the new device name

[image]

rp2019433, 2024-10-08:
Hi. 

The problem is related to SDM01 device and probably due to change in value mapping after software update.

> To fix this I removed the device from the Tuya App and did a factory reset by holding down the reset button.
> 
> I added the device back to the Tuya app and I renamed the device to a different name to before
> 
> I removed the cards I had used in HA and restarted HA then added back the cards searching using the new device name
> 
> ![IMG_1885](https://private-user-images.githubusercontent.com/170913215/374006877-85f163fe-da6b-4a56-b06f-6b9298e42634.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0MDAwNjEsIm5iZiI6MTcyODM5OTc2MSwicGF0aCI6Ii8xNzA5MTMyMTUvMzc0MDA2ODc3LTg1ZjE2M2ZlLWRhNmItNGE1Ni1iMDZmLTZiOTI5OGU0MjYzNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwOFQxNTAyNDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTViNjJiM2RlOTk0MGE5YTcwMGEwNzFlYjQzZWE5YTE1MGRmYjgwOWFiZjdiM2M3YTFlNGQ4ZmMzNDc1MTVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.G8JQ8ll4RZFVcxhfUyqndEdMWXkatwnj_Fd4AIYv3rs)

rp2019433, 2024-10-09:
For those who are experiencing problems like this, especially after updating SDM01 to Tuya version 1.0.6, use the new external converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_ugekduaj'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ugekduaj'},],
    model: 'SDM01',
    vendor: 'TuYa',
    description: 'Smart energy monitor for 3P+N system',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes:  [tuya.exposes.voltageWithPhase('A'), tuya.exposes.voltageWithPhase('B'), tuya.exposes.voltageWithPhase('C'),
        tuya.exposes.powerWithPhase('A'), tuya.exposes.powerWithPhase('B'), tuya.exposes.powerWithPhase('C'),
        tuya.exposes.currentWithPhase('A'), tuya.exposes.currentWithPhase('B'), tuya.exposes.currentWithPhase('C'),
        e.energy().withDescription('Total forward active energy'),
        e.produced_energy().withDescription('Total reverse active energy'),
        e.power_factor().withUnit('%').withDescription('Total power factor'), e.power().withDescription('Total active power'),
        e.ac_frequency(),
        tuya.exposes.energyWithPhase('A'), tuya.exposes.energyWithPhase('B'), tuya.exposes.energyWithPhase('C'),
        tuya.exposes.energyProducedWithPhase('A'), tuya.exposes.energyProducedWithPhase('B'), tuya.exposes.energyProducedWithPhase('C'),
        tuya.exposes.powerFactorWithPhase('A'), tuya.exposes.powerFactorWithPhase('B'), tuya.exposes.powerFactorWithPhase('C')
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [23, 'produced_energy', tuya.valueConverter.divideBy100],
            [29, 'power', tuya.valueConverter.raw],
            [32, 'ac_frequency', tuya.valueConverter.divideBy100],
            [50, 'power_factor', tuya.valueConverter.raw],
            [103, 'voltage_A', tuya.valueConverter.divideBy10],
            [104, 'current_A', tuya.valueConverter.divideBy1000],
            [105, 'power_A', tuya.valueConverter.raw],
            [108, 'power_factor_A', tuya.valueConverter.raw],
            [109, 'energy_A', tuya.valueConverter.divideBy100],
            [110, 'energy_produced_A', tuya.valueConverter.divideBy100],
            [112, 'voltage_B', tuya.valueConverter.divideBy10],
            [113, 'current_B', tuya.valueConverter.divideBy1000],
            [114, 'power_B', tuya.valueConverter.raw],
            [117, 'power_factor_B', tuya.valueConverter.raw],
            [118, 'energy_B', tuya.valueConverter.divideBy100],
            [119, 'energy_produced_B', tuya.valueConverter.divideBy100],
            [121, 'voltage_C', tuya.valueConverter.divideBy10],
            [122, 'current_C', tuya.valueConverter.divideBy1000],
            [123, 'power_C', tuya.valueConverter.raw],
            [126, 'power_factor_C', tuya.valueConverter.raw],
            [127, 'energy_C', tuya.valueConverter.divideBy100],
            [128, 'energy_produced_C', tuya.valueConverter.divideBy100],
        ],
    },
};

module.exports = definition;
```

closed 2024-10-09 by rp2019433

Tsury, 2024-10-15:
Did you notice whether the latest firmware tones down the network spam? I didn't update yet and it constantly spams the network with messages, sometimes 3-5 times in one second. Currently using the debounce to at least not process all of it, but it's still clogging the network and I have no clue how to handle it. @rp2019433 @EcoHouseThailand

rp2019433, 2024-11-17:
> Did you notice whether the latest firmware tones down the network spam? I didn't update yet and it constantly spams the network with messages, sometimes 3-5 times in one second. Currently using the [...]

Hi. I'm using Debounce 3 and this setting is working well with my Zigbee network. Are you experiencing any performance problem?

I'm not sure to tell you, but it seems after updating to the latest firmware messages are sent with longer interval - it seems every ~ 15 seconds.

Tsury, 2024-11-20:
@rp2019433 How do I update it? Pair it to a Tuya hub and use the app on the phone?

rp2019433, 2024-11-20:
> @rp2019433 How do I update it? Pair it to a Tuya hub and use the app on the phone?

Yes, pair with Tuya hub and update it.

CodeFinder2, 2025-03-29:
> For those who are experiencing problems like this, especially after updating SDM01 to Tuya version 1.0.6, use the new external converter:
>
> ```
> const fz = [...]

Thanks for sharing this, helped me a lot. Got my device to report correct values again.

For others having the same problem (reported values are incorrect after upgrading to 1.0.6): the path for storing that new external converter, as of Z2M v2.1.3-1 (via HA add-on, newest as of 29/03/2025), is: `addon_configs/45df7312_zigbee2mqtt/zigbee2mqtt/external_converters/tuya-sdm01.js` (the name of the file doesn't matter). There's no need to change the `configuration.yaml`.

Really don't now why this isn't part of the docs. I figured this out by adding it dynamically via MQTT (also ChatGPT gave confusing and outdated answers).

Regarding update frequency after the update: I am not really sure where this has reduced the network spam. "Linkquality" is still updated quite often (sometimes, the value changes a few times per second). Other values seem to be updated less often. I am not sure and I even don't think its worth upgrading. For posterity, the "changelog" in the Tuya app says:

[image]

[image]