Koenkk/zigbee2mqtt #24152: [Feature request]: Allow hiding groups from Home Assistant

jackwilsdon, 2024-09-29:
### Is your feature request related to a problem? Please describe

I have some groups which are used for binding which I do not want to expose in Home Assistant.

### Describe the solution you'd like

Setting `homeassistant: null` on a group should hide it from Home Assistant.

### Describe alternatives you've considered

No obvious alternatives.

### Additional context

N/A

John1322, 2024-09-30:
Hey. I would like to help with this one!

jackwilsdon, 2024-09-30:
Relevant code is here: https://github.com/Koenkk/zigbee2mqtt/blob/403d3c090d71d06771b860abf988ae3450077ecb/lib/extension/homeassistant.ts#L1496-L1503

jackwilsdon, 2024-12-11:
I'll have a go at implementing this.

closed 2025-02-19 by github-actions[bot]