Koenkk/zigbee2mqtt #24154: zh:zstack:znp: Socket error using SLZB-06 and tcp/ethernet

indexample, 2024-09-30:
### What happened?

after starting the Zigbee2MQTT  addon for HA, it starts and opens socket successfully.
in log messages (INFO) indicate the 
usually within a few minutes, sometimes 30 minutes, it crashes

happens with:
- Zigbee2MQTT and Edge
- port: hardcoded IP and local dns name

below a snippet of the debug log. as you can see, it runs normally (first timestamp 13:04:39 about skipping discovery)
then crashes:

`
**[2024-09-30 13:04:39]** debug: 	z2m: Skipping discovery of 'sensor/0xa4c13878fc15b22e/linkquality/config', already discovered
**[2024-09-30 13:06:50]** debug: 	zh:zstack:unpi:parser: <-- [254,35,68,129,0,0,0,0,253,197,1,1,0,160,0,12,214,156,0,0,15,8,33,10,1,0,32,80,226,255,32,55,228,255,32,0,253,197,29,113]
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,35,68,129,0,0,0,0,253,197,1,1,0,160,0,12,214,156,0,0,15,8,33,10,1,0,32,80,226,255,32,55,228,255,32,0,253,197,29,113]
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,253,197,1,1,0,160,0,12,214,156,0,0,15,8,33,10,1,0,32,80,226,255,32,55,228,255,32,0,253,197,29] - 113
[2024-09-30 13:06:50] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":50685,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":160,"securityuse":0,"timestamp":10278412,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,33,10,1,0,32,80,226,255,32,55,228,255,32,0]}}
[2024-09-30 13:06:50] debug: 	zh:controller: Received payload: clusterID=0, address=50685, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=160, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":33,"commandIdentifier":10},"payload":[{"attrId":1,"dataType":32,"attrData":80},{"attrId":65506,"dataType":32,"attrData":55},{"attrId":65508,"dataType":32,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-09-30 13:06:50] debug: 	zh:controller:endpoint: ZCL command 0xa4c138997ce5df3d/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":33,"writeUndiv":false})
[2024-09-30 13:06:50] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138997ce5df3d:50685/1 (0,0,1)
[2024-09-30 13:06:50] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":50685,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":2,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,33,11,10,0]}}
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,253,197,1,1,0,0,2,0,30,5,24,33,11,10,0,51]
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-30 13:06:50] debug: 	z2m: Received Zigbee message from 'Kantoorlamp1', type 'attributeReport', cluster 'genBasic', data '{"65506":55,"65508":0,"appVersion":80}' from endpoint 1 with groupID 0
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-09-30 13:06:50] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,2,196]
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,2,196]
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196
[2024-09-30 13:06:50] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2}
[2024-09-30 13:06:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
**[2024-09-30 13:07:36] info: 	zh:zstack:znp: Socket error**
[2024-09-30 13:07:36] info: 	zh:zstack:znp: Port closed
[2024-09-30 13:07:36] debug: 	zh:controller: Adapter disconnected
[2024-09-30 13:07:36] info: 	zh:zstack:znp: closing
[2024-09-30 13:07:36] error: 	z2m: Adapter disconnected, stopping
[2024-09-30 13:07:36] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2024-09-30 13:07:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-09-30 13:07:36] info: 	z2m: Disconnecting from MQTT server
[2024-09-30 13:07:36] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-30 13:07:36] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-09-30 13:07:36] info: 	z2m: Stopped zigbee-herdsman
[2024-09-30 13:07:36] info: 	z2m: Stopped Zigbee2MQTT
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2024-09-30 13:07:38] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-30 13:07:38] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-09-30 13:07:38] info: 	z2m: Starting Zigbee2MQTT version 1.40.1 (commit #unknown)
[2024-09-30 13:07:38] info: 	z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-30 13:07:38] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":27209,\"extendedPanID\":[221,141,6,80,38,137,160,61],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"tcp://zigbee2mqtt-slzb06.local:6638\",\"adapter\":\"zstack\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false}}"'
[2024-09-30 13:07:38] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":27209,"extendedPanID":[221,141,6,80,38,137,160,61],"channelList":[11]},"serialPort":{"path":"tcp://zigbee2mqtt-slzb06.local:6638","adapter":"zstack"},"adapter":{"disableLED":false,"concurrent":null,"delay":null},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2024-09-30 13:07:38] info: 	zh:zstack:znp: Opening TCP socket with zigbee2mqtt-slzb06.local:6638
[2024-09-30 13:07:38] info: 	zh:zstack:znp: Socket error
[2024-09-30 13:07:38] error: 	z2m: Error while starting zigbee-herdsman
[2024-09-30 13:07:38] error: 	z2m: Failed to start zigbee
[2024-09-30 13:07:38] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-30 13:07:38] error: 	z2m: Exiting...
[2024-09-30 13:07:38] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:178:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:189:25)
    at Logger.info (/app/lib/util/logger.ts:202:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:108:16)
    at Object.onceWrapper (node:events:632:26)
    at Socket.emit (node:events:517:28)
    at TCP.<anonymous> (node:net:350:12)
`

### What did you expect to happen?

the addon keeps running / not crash

### How to reproduce it (minimal and precise)

run the addon for a few minutes

### Zigbee2MQTT version

1.40.1-1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

addon on home assistant, intel nuc x64 Hassio

### Debug log

[log (1).log](https://github.com/user-attachments/files/17188827/log.1.log)
[log.log](https://github.com/user-attachments/files/17188828/log.log)

magiva, 2024-10-09:
has anyone looked at this yet, i have the same issue

gcs8, 2024-10-12:
`[2024-10-12 06:58:46] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-10-12 06:58:46] info: 	z2m: Disconnecting from MQTT server
[2024-10-12 06:58:46] info: 	z2m: Stopping zigbee-herdsman...
[2024-10-12 06:58:57] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-10-12 06:58:57] info: 	zh:zstack:znp: closing
[2024-10-12 06:58:57] info: 	z2m: Stopped zigbee-herdsman
[2024-10-12 06:58:57] info: 	z2m: Stopped Zigbee2MQTT

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:96:16)
    at Object.onceWrapper (node:events:632:26)
    at Socket.emit (node:events:529:35)
    at TCP.<anonymous> (node:net:350:12)
[06:58:57] INFO: Preparing to start...
[06:58:58] INFO: Socat not enabled
[06:58:59] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-10-12 06:59:01] info: 	z2m: Logging to console, file (filename: log.log)
[2024-10-12 06:59:01] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #unknown)
[2024-10-12 06:59:01] info: 	z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-12 06:59:01] info: 	zh:zstack:znp: Opening TCP socket with 192.168.100.137:6638
[2024-10-12 06:59:01] info: 	zh:zstack:znp: Socket connected
[2024-10-12 06:59:01] info: 	zh:zstack:znp: Socket ready
[2024-10-12 06:59:01] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-10-12 06:59:02] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-10-12 06:59:02] info: 	z2m: zigbee-herdsman started (resumed)
[2024-10-12 06:59:02] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20221226,"transportrev":2},"type":"zStack3x0"}'
[2024-10-12 06:59:02] info: 	z2m: 0x00158d008afe16cf (0x00158d008afe16cf): Not supported (Unknown)
[2024-10-12 06:59:02] info: 	z2m: Currently 1 devices are joined.
[2024-10-12 06:59:02] info: 	z2m: Zigbee: disabling joining new devices.
[2024-10-12 06:59:02] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-10-12 06:59:02] info: 	z2m: Connected to MQTT server
[2024-10-12 06:59:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-10-12 06:59:02] info: 	z2m: Started frontend on port 8099
[2024-10-12 06:59:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-10-12 06:59:02] info: 	z2m: Zigbee2MQTT started!`

magiva, 2024-10-12:
> at TCP. (node:net:350:12)

I found my issue.
I chose to lock the network access to my local network only. which is fine
for accessing the Web page but maybe there is a bug in it allowing access
on sockets.

once I turned that off, connection was fine.

LintHart, 2024-10-14:
> I found my issue. on my szlb06 ..... I chose to lock the network access to my local network only. which is fine for accessing the Web page but maybe there is a bug in it allowing access on sockets. once I turned that off, connection was fine.
> […](#)

Hey @magiva, could you elaborate a little on the steps you took? Were you blocking access at your router level? or were all your adjustments made via the SLZB settings page/z2m settings etc.

just swtiched over to an SLZB-06 and also having this issue

cheers,

Lint

magiva, 2024-10-14:
slzb security page for ip whitelist
I turned that off and it was OK

LintHart, 2024-10-14:
> slzb security page for ip whitelist I turned that off and it was OK
> […](#)

Great, thank you :) I didn't have that on myself though I'd like too once things stabilise.

I ended up changing the port number late afternoon and it was still alive this morning which is a first in a few days :)

indexample, 2024-10-16:
i tried changing the port too (from 6638 to 6639) but i still get the same error after a few minutes:
```
[2024-10-16 13:54:07] info: 	zh:zstack:znp: Socket error
[2024-10-16 13:54:07] info: 	zh:zstack:znp: Port closed
[2024-10-16 13:54:07] info: 	zh:zstack:znp: closing
[2024-10-16 13:54:07] error: 	z2m: Adapter disconnected, stopping
[2024-10-16 13:54:07] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-10-16 13:54:07] info: 	z2m: Disconnecting from MQTT server
[2024-10-16 13:54:07] info: 	z2m: Stopping zigbee-herdsman...
[2024-10-16 13:54:07] info: 	z2m: Stopped zigbee-herdsman
[2024-10-16 13:54:07] info: 	z2m: Stopped Zigbee2MQTT
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2024-10-16 13:54:09] info: 	z2m: Logging to console, file (filename: log.log)
[2024-10-16 13:54:09] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #unknown)
[2024-10-16 13:54:09] info: 	z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-16 13:54:09] info: 	zh:zstack:znp: Opening TCP socket with 192.168.2.7:6639
[2024-10-16 13:54:09] info: 	zh:zstack:znp: Socket error
[2024-10-16 13:54:09] error: 	z2m: Error while starting zigbee-herdsman
[2024-10-16 13:54:09] error: 	z2m: Failed to start zigbee
[2024-10-16 13:54:09] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-10-16 13:54:09] error: 	z2m: Exiting...
[2024-10-16 13:54:09] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:166:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:96:16)
    at Object.onceWrapper (node:events:632:26)
    at Socket.emit (node:events:517:28)
    at TCP.<anonymous> (node:net:350:12)
```

Koenkk, 2024-12-10:
Could you check with the `.dev16` firmware for the SLZB-06 ESP? Smlight mentioned to me there are some improvements in there.

indexample, 2024-12-10:
Mine was fixed with #644

closed 2024-12-10 by indexample

schwalbe13, 2024-12-19:
I am running on the SLZB-06 Core Firmware version  v2.6.8.dev21 and still experience socket errors.

Here is the last bit of a Zigbee2MQTT log file:
```
[2024-12-19 19:03:18] error: 	zh:controller: Failed to keep permit join alive: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
[2024-12-19 19:03:27] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2024-12-19 19:03:27] info: 	zh:zstack:znp: Port closed
[2024-12-19 19:03:27] debug: 	zh:controller: Adapter disconnected
[2024-12-19 19:03:27] info: 	zh:zstack:znp: closing
[2024-12-19 19:03:27] error: 	z2m: Adapter disconnected, stopping
[2024-12-19 19:03:27] debug: 	z2m: Saving state to file /app/data/state.json
[2024-12-19 19:03:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-12-19 19:03:27] info: 	z2m: Disconnecting from MQTT server
[2024-12-19 19:03:27] info: 	z2m: Stopping zigbee-herdsman...
[2024-12-19 19:03:27] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-12-19 19:03:27] info: 	z2m: Stopped zigbee-herdsman
[2024-12-19 19:03:27] info: 	z2m: Stopped Zigbee2MQTT
```

Here the end of another Zigbee2MQTT log file:
```
[2024-12-19 19:31:05] debug: 	z2m: Device 'dHF1LWo1' requested OTA
[2024-12-19 19:31:29] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2024-12-19 19:31:29] info: 	zh:zstack:znp: Port closed
[2024-12-19 19:31:29] debug: 	zh:controller: Adapter disconnected
[2024-12-19 19:31:29] info: 	zh:zstack:znp: closing
[2024-12-19 19:31:29] error: 	z2m: Adapter disconnected, stopping
[2024-12-19 19:31:29] debug: 	z2m: Saving state to file /app/data/state.json
[2024-12-19 19:31:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-12-19 19:31:29] info: 	z2m: Disconnecting from MQTT server
[2024-12-19 19:31:29] info: 	z2m: Stopping zigbee-herdsman...
[2024-12-19 19:31:29] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-12-19 19:31:29] info: 	z2m: Stopped zigbee-herdsman
[2024-12-19 19:31:29] info: 	z2m: Stopped Zigbee2MQTT
```

Koenkk, 2024-12-21:
@schwalbe13 this is either an issue with the SLZB-06 (ESP32 side locking-up, known issue) or your WiFi/LAN network stability (not Zigbee), it cannot be fixed from Z2M.

kiloptero, 2025-01-20:
uncheck ip whitelist and everything is ok.

LeeTaylorX12, 2025-01-23:
### when applying the HAOS update it breaks.  i didn't have the whitelist on.  I have applied and removed without any change.

`[2025-01-23 09:08:36] info: 	zh:zstack:znp: Socket connected
[2025-01-23 09:08:36] info: 	zh:zstack:znp: Socket ready
[2025-01-23 09:08:36] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-23 09:08:36] debug: 	zh:zstack:unpi:writer: --> buffer [239]
[2025-01-23 09:08:37] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-01-23 09:08:37] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-23 09:08:37] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-23 09:08:37] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-23 09:08:37] error: 	z2m: Exiting...
[2025-01-23 09:08:37] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:530:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)
/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
    at Object.onceWrapper (node:events:633:26)
    at Socket.emit (node:events:518:28)
    at TCP.<anonymous> (node:net:343:12)`

rret, 2025-02-01:
this fixed it for me:

(as per user rbotteon)-
**Solved by activating Allow multi-threaded socket connection in SLZB-06 / Z2M and ZHA / ADVANCED: Socket connection options**

Release notes:
1. Active Sockets Multithreading Management: By default, 1 active socket is enabled (recommnded setting). User can change this in the “ZHA/Z2M” settings tab.
2. Improved Network Communication: A new Zigbee packet processing mode has been implemented. It groups all ZNP(ZSTACK), EMBER or EZSP packets and processes them at once (previously, only individual bytes were processed).
...

RaggedyDandy, 2025-03-22:
> I had the same problem with the **SLZB-06**.
> 
> Solved by activating `Allow multi-threaded socket connection` in `SLZB-06 / Z2M and ZHA / ADVANCED: Socket connection options:`
> 
> Is a new setting in firmware **v2.7.1**:
> 
> ```
> Release notes:
> 1. Active Sockets Multithreading Management: By default, 1 active socket is enabled (recommnded setting). User can change this in the “ZHA/Z2M” settings tab.
> 2. Improved Network Communication: A new Zigbee packet processing mode has been implemented. It groups all ZNP(ZSTACK), EMBER or EZSP packets and processes them at once (previously, only individual bytes were processed).
> ...
> ```

*** FINALLY! *** I've been trying to get Z2M to work for months, and this solved the 502: Bad Gateway issue for me.