Koenkk/zigbee2mqtt #24155: Zemismart Model ZM25RX-08/30 Zigbee Model TS0601 Zigbee Manufacturer _TZE200_7eue9vhc

yozsever, 2024-09-30:
### What happened?

I have 3 ZM25RX-08/30 which shows as 
Zigbee Model TS0601
Zigbee Manufacturer _TZE200_bv1jcqqu which works fine and can be controlled from home assistant.
I bought 2 new ZM25RX-08/30 one shows not supported and the shows as 
Zigbee Model TS0601
Zigbee Manufacturer _TZE200_7eue9vhc
but can not be controlled from home assistant, no response to up down or stop button

### Zigbee2MQTT version

 1.40.1 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Add-on Home Assistant inside Proxmox VM on intel nuc

### Debug log

[home-assistant_mqtt_2024-09-30T11-21-43.870Z.log](https://github.com/user-attachments/files/17188872/home-assistant_mqtt_2024-09-30T11-21-43.870Z.log)

markxroberts, 2024-09-30:
I thought I was the only one with this problem.  I didn't realise that it was model-specific.  See #23801.  I returned a _7eue9vhc as faulty.  It responded to Zigbee commands during the pairing phase but not thereafter.  It responded to the remote control at all times.  I've ordered another which shows up as _bv1jcqqu which operates perfectly.  I wonder whether there is a problem with a batch of _7eue9vhc.  I don't think this is a Z2M issue as I have the same problem with a Tuya hub - it pairs and is then not controllable.

yozsever, 2024-09-30:
Yes the 2 new motors can only be used from the remote control it works but not from the z2m. Maybe needs an firmware update but how?

closed 2024-12-19 by github-actions[bot]