Koenkk/zigbee2mqtt #24158: Ember Znet backup gives empty devices object

KipK, 2024-09-30:
### What happened?

When doing a dongle backup from z2m or using ember-zli, the coordinator_backup.json generated doesn't have any devices included in device object:

devices": []

So I have doubt we can restore the network using actual ember backups yet.

### What did you expect to happen?

should output the paired device in devices object:
```
devices": [
{
      "nwk_address": "803d",
      "ieee_address": "842e14fffe6cca58",
      "is_child": false
    },
]

```

### How to reproduce it (minimal and precise)

Ask for a backup in z2m , and check coordinator_backup.json

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

ember 8.0.1 ( tested with 7.4.4 too )

### Adapter

SonOff-E

### Setup

z2m in lxc container

Koenkk, 2024-10-01:
@Nerivec I believe this is expected since it does not have the TC link keys, right?

Nerivec, 2024-10-01:
It is expected indeed. `ember` doesn't have `devices` since app link keys are not used (hence empty).

closed 2024-10-01 by Nerivec

KipK, 2024-10-01:
Will backup / restore to another stick still works then?

Nerivec, 2024-10-01:
From an EmberZNet adapter to an EmberZNet adapter, it shouldn't be a problem.

For migrating to/from other stacks, as far as I know, stacks have different internal formats for app link keys anyway.
_Note: `ember` currently won't let you cross-restore unless you manually change the JSON (accepting the risks that go with it). `zstack` doesn't prevent it, but that's mostly because it never did and was "alone" before._
_Extensive testing is still required for cross-restoring, several stacks, many firmware versions, two directions... loads of data!_

KipK, 2024-10-02:
I've already successfully transferred my network from Zstack to EmberZnet few months ago. However I was using the previous znp implementation, and used zigpy for that task. Haven't tested the other way back yet.