Koenkk/zigbee2mqtt #24160: [New device support]: Tuya Zigbee IR Remote  TS1201 - Vendor: _TZ3290_u9xac5rv

gr1los, 2024-09-30:
### Link

https://www.aliexpress.com/item/1005007621983514.html

### Database entry

{"id":17,"type":"EndDevice","ieeeAddr":"0x0cae5ffffe76f752","nwkAddr":37883,"manufId":4098,"manufName":"_TZ3290_u9xac5rv","powerSource":"Battery","modelId":"TS1201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":61440,"inClusterList":[0,4,5,3,1,60672,57348,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS1201","manufacturerName":"_TZ3290_u9xac5rv","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1727712493418}

### Zigbee2MQTT version

1.40.1-1

### Comments

I purchased this zigbee IR to give it a try hoping that using an external definition based on [Moes UFO-R11](https://www.zigbee2mqtt.io/devices/UFO-R11.html#moes-ufo-r11) will be usable. After reading the  following issue https://github.com/Koenkk/zigbee2mqtt/issues/11633 I was able to end up on the following external definition below.

Also an image of the internal PCB
[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const Buffer = require('buffer').Buffer;
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ea = exposes.access;
const e = exposes.presets;

function nextSeq(entity) {
    entity.seq = ((entity.seq || -1)+1) % 0x10000;
    return entity.seq;
}

function messagesGet(entity, seq) {
    const info = entity.irMessageInfo;
    if (info.seq!=seq) {
        throw new Error(`Unexpected sequence value (expected: ${info.seq} current: ${seq}).`);
    }
    return info.data;
}
function messagesSet(entity, seq, data) {
    entity.irMessageInfo={seq: seq, data: data};
}

function messagesClear(entity, seq) {
    const info = entity.irMessageInfo;
    if (info.seq!=seq) {
        throw new Error(`Unexpected sequence value (expected: ${info.seq} current: ${seq}).`);
    }
    delete entity.irMessageInfo;
}

function calcArrayCrc(values) {
    return Array.from(values.values()).reduce((a, b)=>a+b)%0x100;
}

function calcStringCrc(str) {
    return str.split('').map((x)=>x.charCodeAt(0)).reduce((a, b)=>a+b)%0x100;
}

const fzZosung = {
    zosung_send_ir_code_01: {
        cluster: 'zosungIRTransmit',
        type: ['commandZosungSendIRCode01'],
        convert: (model, msg, publish, options, meta) => {
            meta.logger.debug(`"IR-Message-Code01" received (msg:${JSON.stringify(msg.data)})`);
            const seq = msg.data.seq;
            const irMsg = messagesGet(msg.endpoint, seq);
            meta.logger.debug(`IRCode to send: ${JSON.stringify(irMsg)} (seq:${seq})`);
[... 191 lines cut ...]
        model: 'ZS05',
        vendor: 'FLY',
        description: 'Universal Smart IR Remote Control',
        fromZigbee: [
            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
        ],
        toZigbee: [
            tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code,
        ],
        toZigbee: [
            tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code,
        ],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send()],
    },
];
```

### What does/doesn't work with the external definition?

With the following external definition I was able by requesting the `learn_ir_code` ( as described [here](https://www.zigbee2mqtt.io/devices/UFO-R11.html#switch-to-a-learning-mode) ) to make the led flashing on for couple of seconds. Pointing an IR source to it while pressing a button nothing happens and the `learned_ir_code` is empty.

This device is much smaller than the UFO-R11 and its white color makes it really good especially for AC/Mini Split Ductless Units/Heatpumps that they are mostly white.

My current setup is Z2M V1.40.1-1 on HaOs with Sonoff Zigbee Bridge-P Tasmotized

gr1los, 2024-09-30:
Made some progress with the following external definition:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS1201', manufacturerName: '_TZ3290_u9xac5rv'},
        ],
        model: 'ZS05',
        vendor: 'FLY',
        description: 'Universal smart IR remote control',
        fromZigbee: [
            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
            fz.battery,
        ],
        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send(), e.battery(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.batteryVoltage(endpoint);
        },
    },
];
```

Now it's able to read an IR code and sent it. 
Although I'm still tinkering with the battery that reports that it's at 1%.

hirenj, 2024-11-03:
I've managed to get this working on a device I received - it's reporting 100% battery, but they're fresh-ish batteries so who knows if this means the battery reporting is working or not!

The above code was missing the code to load the `exposes` module, which I added in here. This should go in a TS1201.js file, next to the configuration.yaml, and add the reference to the file in the external_converters section as per adding any other external converter.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS1201', manufacturerName: '_TZ3290_u9xac5rv'},
        ],
        model: 'ZS05',
        vendor: 'FLY',
        description: 'Universal smart IR remote control',
        fromZigbee: [
            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
            fz.battery,
        ],
        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send(), e.battery(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.batteryVoltage(endpoint);
        },
    },
];
```

arthurperezbessa, 2024-11-04:
Thank you! Finally got this working. I can finally migrate almost everything to Zigbee. Local Tuya and Tuya Local are sometimes a pain in the a**. Devices get unavailable too often.

My device doesn't have Battery, so follows below the code without ir

ATTENTION: the configuration.yaml that you should put "external_converters" is the Zigbee2Mqtt, not the HA main conf.

```
external_converters:
  - external_converters/TS1201.js
```

TS1201.js File.

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS1201', manufacturerName: '_TZ3290_8biqfysp'},
        ],
        model: 'ZS05',
        vendor: 'FLY',
        description: 'Universal smart IR remote control',
        fromZigbee: [
            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
        ],
        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg');
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        },
    },
];```

arthurperezbessa, 2024-11-04:
So. This external convertings is working for me, but it show an error when realoading Zigbee2MQTT

I think it has some to do with me deleting the battery functions.

`
error 2024-11-04 16:02:24z2m: Failed to configure 'Smart Zigbee IR', attempt 3 (TypeError: Cannot read properties of undefined (reading 'map') at Endpoint.ensureManufacturerCodeIsUniqueAndGet (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:849:24) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:434:53) at Object.configure (/app/data/extension/externally-loaded.js:26:28) at Configure.configure (/app/lib/extension/configure.ts:130:37) at /app/lib/extension/configure.ts:88:62 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:23) at EventEmitter.emit (node:events:529:35) at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:81:22) at Controller.<anonymous> (/app/lib/zigbee.ts:81:27) at Controller.emit (node:events:517:28))
`

19fabiom84, 2024-11-15:
Hello Giusy!
Thanks to you I can use my Zemismart ZM-18-USB.

I used this code for TS1201.js file:

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS1201', manufacturerName: '_TZ3290_uc8lwbi2'},
        ],
        model: 'ZM-18-USB',
        vendor: 'Zemismart',
        description: 'Universal smart IR remote control',
        fromZigbee: [
            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
            fz.battery,
        ],
        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send()],
    },
];
`

markonings, 2024-12-10:
Hi! For some reason, I'm not able to get my Tuya ZS05 device into learning mode. It keeps saying N/A when the toggle in zigbee2mqtt is set to "On". I do see the light turn white/blue on the actual device, and linkquality acutually gives back a value; but no IR code is displayed. Thoughts?

This is the discovery file im using. It's correctly discovered, but simply not showing any IR values. 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS1201', manufacturerName: '_TZ3290_u9xac5rv'},
        ],
        model: 'ZS05',
        vendor: 'FLY',
        description: 'Universal smart IR remote control',
        fromZigbee: [
            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
            fz.battery,
        ],
        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send(), e.battery(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.batteryVoltage(endpoint);
        },
    },
];
```

[image]
[image]

convixMaximus, 2024-12-12:
Hi. Please tell me. I have a code that I received. How can I integrate it into HA? Turn on TV, change programs, adjust volume, etc. If you write a script, then please show a small example. I will be very grateful for help.

convixMaximus, 2024-12-12:
Something like that.
[image]
[image]
[image]

convixMaximus, 2024-12-12:
> Hi! For some reason, I'm not able to get my Tuya ZS05 device into learning mode. It keeps saying N/A when the toggle in zigbee2mqtt is set to "On". I do see the light turn white/blue on the actual device, and linkquality acutually gives back a value; but no IR code is displayed. Thoughts?
> 
> This is the discovery file im using. It's correctly discovered, but simply not showing any IR values.
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const zosung = require('zigbee-herdsman-converters/lib/zosung');
> const fzZosung = zosung.fzZosung;
> const tzZosung = zosung.tzZosung;
> const ez = zosung.presetsZosung;
> 
> module.exports = [
>     {
>         fingerprint: [
>             {modelID: 'TS1201', manufacturerName: '_TZ3290_u9xac5rv'},
>         ],
>         model: 'ZS05',
>         vendor: 'FLY',
>         description: 'Universal smart IR remote control',
>         fromZigbee: [
>             fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
>             fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
>             fz.battery,
>         ],
>         toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
>         exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send(), e.battery(), e.battery_voltage()],
>         configure: async (device, coordinatorEndpoint, logger) => {
>             const endpoint = device.getEndpoint(1);
>             await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
>             await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
>             await reporting.batteryPercentageRemaining(endpoint);
>             await reporting.batteryVoltage(endpoint);
>         },
>     },
> ];
> ```
> 
> <img alt="Scherm­afbeelding 2024-12-10 om 21 16 09" width="471" src="https://private-user-images.githubusercontent.com/178611630/394438058-53a429df-2b33-4efd-884e-6d20dda15d87.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQwMDgxOTUsIm5iZiI6MTczNDAwNzg5NSwicGF0aCI6Ii8xNzg2MTE2MzAvMzk0NDM4MDU4LTUzYTQyOWRmLTJiMzMtNGVmZC04ODRlLTZkMjBkZGExNWQ4Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxMlQxMjUxMzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMTRkNjM5Mjc0Y2Q0OTEyYjA2ZDQ2NTU1YWRkMWIyYWEyMTFhZDE5NWJkMGRmYTdkZjc1NDBlNzk0YWVlNTk3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.NHDMKGe3GkCfb1YNZm7bTC-7hlyYlp0PtQOs9W_Az78"> <img alt="Scherm­afbeelding 2024-12-10 om 21 20 36" width="674" src="https://private-user-images.githubusercontent.com/178611630/394439559-4b2e013b-3e91-47e1-8e0a-94b009d6f26e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQwMDgxOTUsIm5iZiI6MTczNDAwNzg5NSwicGF0aCI6Ii8xNzg2MTE2MzAvMzk0NDM5NTU5LTRiMmUwMTNiLTNlOTEtNDdlMS04ZTBhLTk0YjAwOWQ2ZjI2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxMlQxMjUxMzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNDZiZWI3NmMxZGMyMmFmY2JlZDMwZTg1MmQzN2E0MTI0Yjc5MTdkNTUzNGU5NDM4YjcxNzE2YTEzMmE2YjljJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.V_ZuROor6QeJNSeS38zCI5p1-PvYm4dwF5FFh-bMp_M">

[TS1201.zip](https://github.com/user-attachments/files/18111810/TS1201.zip)
It s work

fabianosan, 2024-12-20:
> Hello guys! Thanks to you I can use now my Zemismart ZM-18-USB (USB IR dongle).
> 
> I used this code for TS1201.js file:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const zosung = require('zigbee-herdsman-converters/lib/zosung');
> const fzZosung = zosung.fzZosung;
> const tzZosung = zosung.tzZosung;
> const ez = zosung.presetsZosung;
> 
> module.exports = [
>     {
>         fingerprint: [
>             {modelID: 'TS1201', manufacturerName: '_TZ3290_uc8lwbi2'},
>         ],
>         model: 'ZM-18-USB',
>         vendor: 'Zemismart',
>         description: 'Universal smart IR remote control',
>         fromZigbee: [
>             fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
>             fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
>         ],
>         toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
>         exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send()],
>     },
> ];
> ```
> 
> Could someone help me to make it available for other people?

I used it here, the sensors and controls shows in z2m and home assistant, but it doens't work.
You can't learn a command through it, because it's not appear as remote control.

How can you use it with HA?

convixMaximus, 2024-12-22:
I wrote a script with the code from the remote control. You need to read each key.

asfm1964, 2024-12-24:
I have two instantes of zigbee2mqtt with the same file ts1201.js.
Instance 1: zigbee2mqtt version 1.41 work fine. On line model ZS05. Vendor FLY
Instance 2: zigbee2mqtt version 1.42. NO work. On line model TS1201 vendor unknow.
Anyone with the same problem?

OlofssonMagnus, 2025-01-04:
Zigbee2MQTT version 2.0.0-1 running as addon to home assistant
Zigbee USB usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_3cde681a481fef11999657d0639e525b-if00-port0 
Running firmware ncp-uart-sw_EZNet7.4.3_V1.0.0
Running in ember mode, overall it has been unstable. Any other mode recommended?

Moes UFO-R11, gatewayed via a IKEA light bulb.
Can turn Learn ir code ON/OFF. But if I send IR code with learning mode ON I get:

error 2025-01-04 18:37:06zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2025-01-04 18:37:06zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2025-01-04 18:37:06zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2025-01-04 18:37:06zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info 2025-01-04 18:37:06zh:ember:uart:ash: ASH COUNTERS since last clear:
info 2025-01-04 18:37:06zh:ember:uart:ash: Total frames: RX=70, TX=111
info 2025-01-04 18:37:06zh:ember:uart:ash: Cancelled : RX=0, TX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: DATA frames : RX=65, TX=39
info 2025-01-04 18:37:06zh:ember:uart:ash: DATA bytes : RX=1056, TX=515
info 2025-01-04 18:37:06zh:ember:uart:ash: Retry frames: RX=3, TX=2
info 2025-01-04 18:37:06zh:ember:uart:ash: ACK frames : RX=0, TX=69
info 2025-01-04 18:37:06zh:ember:uart:ash: NAK frames : RX=0, TX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: CRC errors : RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Comm errors : RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Length < minimum: RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Length > maximum: RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Bad controls : RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Bad lengths : RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Out of buffers : RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: Retry dupes : RX=3
info 2025-01-04 18:37:06zh:ember:uart:ash: Out of sequence : RX=0
info 2025-01-04 18:37:06zh:ember:uart:ash: ACK timeouts : RX=2
error 2025-01-04 18:37:15zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=37 Len=7].
error 2025-01-04 18:37:15z2m: Exception while calling fromZigbee converter: ZCL command 0x4c5bb3fffe83396a/1 zosungIRTransmit.zosungSendIRCode01({"zero":0,"seq":5,"length":504,"unk1":0,"unk2":57348,"unk3":1,"cmd":4,"unk4":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0x4c5bb3fffe83396a:41578 apsFrame={"profileId":260,"clusterId":60672,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0}] Failed to send request with status=FAIL.)}
info 2025-01-04 18:37:36zh:ember:uart:ash: Port closed.
info 2025-01-04 18:37:36zh:ember:uart:ash: ======== ASH stopped ========
info 2025-01-04 18:37:36zh:ember:ezsp: ======== EZSP stopped ========
info 2025-01-04 18:37:36zh:ember: ======== Ember Adapter Stopped ========
error 2025-01-04 18:37:36z2m: Adapter disconnected, stopping

hubpf, 2025-01-07:
It stops working under Zigbee2MQTT version 2.0.0, can you please help to fix it? Thanks

hubpf, 2025-01-08:
@arthurperezbessa Thank you so much, it works great now. By the way, how can we add the as officially supported device rather than using external converter?

gr1los, 2025-01-19:
Considering that this ticket is needed by a lot of people I will try in the next few days ( depending on my work load ) to create a proper PR adding this device in the official Z2M compatibility list

hubpf, 2025-01-21:
Perfect, thank you, @gr1los

dantol1, 2025-01-29:
> I wrote a script with the code from the remote control. You need to read each key. [#24160 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/24160#issuecomment-2538760718)

can you please share this script? no idea how to do it yet 😅

lattice0, 2025-02-21:
I tried this from @convixMaximus and it worked at the first time then gave errors like

`z2m: Exception while calling fromZigbee converter: Unexpected sequence value (expected: 9 current: 0).}
`

code

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');

const reporting = require('zigbee-herdsman-converters/lib/reporting');

const exposes = require('zigbee-herdsman-converters/lib/exposes');

const e = exposes.presets;

const zosung = require('zigbee-herdsman-converters/lib/zosung');

const fzZosung = zosung.fzZosung;

const tzZosung = zosung.tzZosung;

const ez = zosung.presetsZosung;

module.exports = [

    {

        fingerprint: [

            {modelID: 'TS1201', manufacturerName: '_TZ3290_u9xac5rv'},

        ],

        model: 'ZS05',

        vendor: 'FLY',

        description: 'Universal smart IR remote control',

        fromZigbee: [

            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,

            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,

            fz.battery,

        ],

        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],

        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send(), e.battery(), e.battery_voltage()],

        configure: async (device, coordinatorEndpoint, logger) => {

            const endpoint = device.getEndpoint(1);

            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);

            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);

            await reporting.batteryPercentageRemaining(endpoint);

            await reporting.batteryVoltage(endpoint);

        },

    },

];
```

error 2025-02-21 13:40:02z2m: Exception while calling fromZigbee converter: Unexpected sequence value (expected: 8 current: 0).}

lattice0, 2025-02-21:
more errors, now when learning a code:
```

zhc:zosung: Unexpected IR code position: {"zero":0,"seq":1,"position":55,"msgpart":{"type":"Buffer","data":[97,2,64,3,3,112,6,97,2,64,7,128,3,64,1,64,11,1,12,2,64,23,5,112,6,12,2,12,2,128,7,7,112,6,12,2,112,6,12,2]},"msgpartcrc":107}, expecting: 95.
zhc:zosung: Unexpected IR code position: {"zero":0,"seq":1,"position":0,"msgpart":{"type":"Buffer","data":[11,65,35,74,17,97,2,112,6,97,2,12,2,224,21,3,64,35,224,23,3,192,67,7,112,6,12,2,12,2,97,2,224,7,3,1,112,6,64,3,3,12,2,12,2,64,7,224,7,3,5,12,2,12,2]},"msgpartcrc":125}, expecting: 55.
```

arthurperezbessa, 2025-02-21:
> more errors, now when learning a code:
>
> ```
>
> zhc:zosung: Unexpected IR code position: [...]

I use this one. Can you try it?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const zosung = require('zigbee-herdsman-converters/lib/zosung');
const fzZosung = zosung.fzZosung;
const tzZosung = zosung.tzZosung;
const ez = zosung.presetsZosung;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS1201', manufacturerName: '_TZ3290_8biqfysp'},
        ],
        model: 'TS1201',
        vendor: 'Tuya',
        description: 'Universal smart IR remote control',
        fromZigbee: [
            fzZosung.zosung_send_ir_code_00, fzZosung.zosung_send_ir_code_01, fzZosung.zosung_send_ir_code_02,
            fzZosung.zosung_send_ir_code_03, fzZosung.zosung_send_ir_code_04, fzZosung.zosung_send_ir_code_05,
            fz.battery,
        ],
        toZigbee: [tzZosung.zosung_ir_code_to_send, tzZosung.zosung_learn_ir_code],
        exposes: [ez.learn_ir_code(), ez.learned_ir_code(), ez.ir_code_to_send(), e.battery(), e.battery_voltage()],
    },
];
```

lattice0, 2025-02-21:
@arthurperezbessa this is not the same model, however I changed here to my model and its working. At least for now. I got some strange errors some times but I will continue investigating.

Battery/Voltage is not being reported and the learning status not as well, I have to press on/off specifically.

Learning seems to be working.

Another little off topic question that I couldnt find help anywhere: home assistant refuses to send the same command if I already sent it before. I have to send another just to clear the buffer and then I can send tth other one again. Is there a way to fix this? It looks like it wont update the text it its already what its trying to update. The text in this case is the IR code

arthurperezbessa, 2025-02-24:
> [@arthurperezbessa](https://github.com/arthurperezbessa) this is not the same model, however I changed here to my model and its working. At least for now. I got some strange errors some times but I [...]

Mine device is not battery powered too, but I use this converter and works perfectly!

I don't have this problem with the same command, because my automation always send three commands in a row

lattice0, 2025-02-24:
@arthurperezbessa the device he shows in tth photo above is battery powered and so is mine. We are talking about slightly different models.

Anyways, how are your automations sending the command? Through entity or through mqtt publish? Are they from home assistant?

From home assistant through entity it should not send the same twice. From home assistant with mqtt publish it works perfectly

arthurperezbessa, 2025-02-24:
> [@arthurperezbessa](https://github.com/arthurperezbessa) the device he shows in tth photo above is battery powered and so is mine. We are talking about slightly different models.
>
> Anyways, how are [...]

```
action: mqtt.publish
data:
  evaluate_payload: false
  qos: 0
  retain: false
  topic: zigbee2mqtt/Smart IR Escritorio/set
  payload: {"ir_code_to_send": "CFQMtiYCAgUGAmAAAzICAgJAAwEFBkAFQACABwMFBgICQAMBAgKADwECAsAHQADgAwuAAEArQBVAB8AACwUGMgIFBjICAgICAg=="}
enabled: true
```

hubpf, 2025-03-02:
> Considering that this ticket is needed by a lot of people I will try in the next few days ( depending on my work load ) to create a proper PR adding this device in the official Z2M compatibility list

@gr1los Just a kind reminder, please don't forget to include this in the official Z2M compatibility list.

emintjeee, 2025-03-12:
Following, I got one of these sensors without checking if it was compatible. When was it first supported?

omega015, 2025-03-18:
Is it expected that there is a delay of 2-10~ seconds for this device to send the IR signal?

lattice0, 2025-03-18:
@omega015 same here on all my 5 devices. I think itś because of power saving. The wired version gors instantly, I also have some.

omega015, 2025-03-18:
> [@omega015](https://github.com/omega015) same here on all my 5 devices. I think itś because of power saving. The wired version gors instantly, I also have some.

Thanks for the confirmation, just good to know and can still be useful for some devices.

lattice0, 2025-03-19:
This .js has the following problem: the devices don't respond to ping, so they are marked offline unless you send a command to them, then they show up as online for a while The wired version which is similar responds to pings.

gr1los, 2025-05-19:
Fyi the support of this device is already included in upstream ZigBee2Mqtt.

omega015, 2025-05-20:
> Fyi the support of this device is already included in upstream ZigBee2Mqtt.

So we can now remove this custom code and still have full support/features?

gr1los, 2025-05-21:
> > Fyi the support of this device is already included in upstream ZigBee2Mqtt.
> 
> So we can now remove this custom code and still have full support/features?

Yes

closed 2025-05-21 by gr1los