Koenkk/zigbee2mqtt #24163: Inconsistent group state. Cover group state "closed" even if one device is "open"

Trexano99, 2024-09-30:
### What happened?

I have create a group with multiple BTicino cover shutter ([K4027C/L4027C/N4027C/NT4027C](https://www.zigbee2mqtt.io/devices/K4027C_L4027C_N4027C_NT4027C.html#bticino-k4027c%252Fl4027c%252Fn4027c%252Fnt4027c)) on Z2M and it works well, but the state sometimes is inconsistent. <br>The group state is presented as "**CLOSE**" while one of the devices "_Tapparella PF Cucina_" in the group is in state "**OPEN**".<br>What should be the problem?<br>
This is the group ("_Tutte Tapparelle_"):
[image]
and this is the log:
[image]
_P.S: The group off_state is "all_members_off"_

### What did you expect to happen?

The group "Tutte Tapparelle" to be "**OPEN**"

### How to reproduce it (minimal and precise)

It happen randomly.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

Zigbee 3.0 USB Dongle-E plus V2 (with ember driver)

### Setup

Home Assistant OS, x86-64

### Debug log

I have seen that in the log the state of "_Tutte Tapparelle_" continue switching from CLOSE to OPEN even if no action take place.
[2024-09-30.22-32-46_log.txt](https://github.com/user-attachments/files/17196598/2024-09-30.22-32-46_log.txt)

Trexano99, 2024-10-03:
Watching at the code I suspect the problem is that even if using the option `off_state: 'all_members_off'`, the state group changing is triggered iif the changed device is _OFF_ and all other devices are in state _OFF_. <br>

In case of groups of cover the change is _OPEN_ or _CLOSE_, and this (**_not ON - OFF_**) bring the group to change state every time a device change it's state or only broadcast it (as my situation).<br> 

According to that I have proposed a [fix on my fork](https://github.com/Trexano99/zigbee2mqtt/commit/38cb1c265b04ea93f40d718bc87a858098c2be21) managing also the state _OPEN_, _CLOSE_.<br>

The only problem is that I cannot test this code because I am running Z2M on HomeAssistant using the specific Add-On, and so I don't know how to propagate the changes to make my own tests. I have seen there's the part of [Testing changes locally](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt?tab=readme-ov-file#testing-changes-locally) in the documentation, but entering the docker container I can't find the src Z2M files (maybe because removed during the Docker build) but only the zigbee-herdsman-converters and few other parts. <br>

Sry for the tag @Koenkk but if you can help my with understanding how to test my changes i would really appreciate that and for sure I will also help to trouble other issues. Thx

Koenkk, 2024-10-04:
@Trexano99 you have to edit the equivalent `.js` files under the `dist` folder to test the changes locally.

closed 2024-11-20 by Trexano99