Koenkk/zigbee2mqtt #24169: SmartCrest HG07310: Values do not get updated

Wuehli, 2024-10-01:
### What happened?

I paired my SmartCrest HG07310, Firmware 2021.1.5 and they get recognized.
But after that the values for `Occupancy`, `Battery `low and `Tamper`
aren't updated and stay as "N/A", only `Battery `gets updated.
I tried two devices and re-paired them several times with no effect, in Deconz/Phoscon they are working fine.

I also tried this workaround https://github.com/Koenkk/zigbee2mqtt/issues/9794
but it doesn't change anything.

### What did you expect to happen?

The values get updated accordingly 

### Zigbee2MQTT version

1.40.1 commit: 403d3c09

### Adapter firmware version

20240710

### Adapter

Sonoff ZBDonlge-P

### Setup

Z2M plain on PiOS Lite x64, Pi4

Wuehli, 2024-10-11:
Anybody any ideas?

Wuehli, 2024-10-19:
@Koenkk Do you might have any idea how to debug this issue?

Koenkk, 2024-10-19:
Could you try the following:
- Force remove the device through the z2m frontend
- Restart z2m
- Enable debug logs, start z2m
- Re-pair device

Provide the debug logs here:

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Wuehli, 2024-10-20:
@Koenkk 

I have attached the full log here, the device is 0xcc86ecfffeeba4c1.
Hope it helps, unfortunately its telling me nothing.

[log.log](https://github.com/user-attachments/files/17451860/log.log)

Many thanks in advance.

Wuehli, 2024-10-25:
In the Meantime I setup a second z2m instance. Basically the same way (Pi 4, stock image with full upgrade, basic installation following the z2m guide, nothing else) using the same Sonoff Typ P USB Dongle (same firmware, same hardware revision), also using the same channel. 
To my supprise the device is recogniced and the missing values where updated (i used different channels incl. the same as my main instance to exclude this as a cause). 

So i applied my config backup to this instance, i even changed the ieee ID of the Sonoff stick to match my network. So basically a clone of my current environment. Everything is working as expected, all my devices work 100% with the clone. Except the HG07310, it isn't even recognised there, i always get an "interview failed". I tried 3 different HG07310 and tried it several times. So it must be something with my configuration is guess. But beside adding devices i haven't done anything fancy. 

🏳️

Wuehli, 2024-10-25:
In the meantime i also switched my clone to the DEV branch, same result 😭

Koenkk, 2024-10-26:
Could you try enabling joining on only the coordinator and pair it close to it?

Wuehli, 2024-10-26:
> Could you try enabling joining on only the coordinator and pair it close to it?

Already tried, even in diffrent order of enabeling joining (first coordinator, then device and vice versa). On my main instance it gets recognized but still the 3 values are missing.

Wuehli, 2024-10-26:
@Koenkk 
Found the solution after spending another 2h trying to understand logfiles, configuration files,... without really having a clue or any understanding of what im doing. The solution is more of a accidental find.
But if I add this:
[image]

The values are updated, I tried it with three devices and it worked every time I added this binding after the successful interview.
Why doesn't this happen from the beginning!?

Koenkk, 2024-10-27:
Interesting!

I've pushed a fix, can you check if it works?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Wuehli, 2024-11-01:
Hi,
sorry for my late Reply, i tried to test it on my DEV instance but couldn't get it to work at all. 
Now i installed the lastes version [1.41.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.41.0) commit: [cc31ceab](https://github.com/Koenkk/zigbee2mqtt/commit/cc31ceab) on my regular instance and it works great! 
I tried it with both devices, HG07310 and HG06335. The teach in is now much faster and works on first attempt now.
Also all values are now directly updated right from the start.
Many thanks @Koenkk

closed 2024-11-01 by Wuehli