Koenkk/zigbee2mqtt #24170: [Wrong device]: Moes ZM-104B-M

LeoeLeoeL, 2024-10-01:
### Link

https://moeshouse.com/products/zigbee-mini-diy-smart-light-switch-module-relay-timer-1-2-gang?variant=47687314243899

### Model

ZM-104B-M

### Description

Smart Switch Module 2 gang

### Vendor

Moes

### Picture (link)

https://moeshouse.com/cdn/shop/products/zigbee-mini-diy-smart-light-switch-module-relay-timer-12-gang-329624.jpg?v=1706013119&width=600

### Database entry

{"id":76,"type":"Router","ieeeAddr":"0xa4c1387dafa4a285","nwkAddr":59552,"manufId":4417,"manufName":"_TZ3000_qaa59zqd","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAA"}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":2}},"genBasic":{"attributes":{"65503":"�\u0011�.

### Notes

Now Z2M is reporting:
Zigbee Model: TS0002
Zigbee Manufacturer: _TZ3000_qaa59zqd
Manufacturer: Tuya
Model: TS0002_basic

LeoeLeoeL, 2024-12-04:
Up

sylarm27, 2024-12-18:
Same here.
Did you find a way to give the correct model?

LeoeLeoeL, 2024-12-18:
I think using the same way for MOES MS-104BZ.
[image]
[image]

townsen, 2024-12-26:
I have the same problem, does the fact that its the wrong device affect functionality? How did you get it to be correctly shown as a MS-104BZ?

pajenny, 2025-01-26:
I have the same problem, has the solution been found?

LeoeLeoeL, 2025-03-28:
First or then...................

joaopedros2, 2025-03-31:
Same problem here.

onurece, 2025-05-02:
This should be fixed in [2.3.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.3.0). "#3658 Detect _TZ3000_qaa59zqd as Moes ZM-104B-M (@Koenkk)"
Can someone please check?

LeoeLeoeL, 2025-05-02:
I do confirm.

[image]

[image]