Koenkk/zigbee2mqtt #24177: [Feature request]: Battery device monitoring and reporting/notification

Majestic7979, 2024-10-01:
### Is your feature request related to a problem? Please describe

It's a problem in the sense that I'm not being made aware of devices that are going offline due to dying batteries (low charge). It breaks automations. I'm disabled and rely on many automations including from battery devices to help me manage my disabilities and live a better life. When the devices die because I was unaware their battery was low, it causes hassles. I would like to be proactive and be informed by the software before problems occur/devices die/drop off from low charge. Retention is causing them to appear as if they're functional/have a charge when they don't. They appear on HA as if they do... Disabling retention causes other issues. I have around 100 Zigbee devices and a few Zwave that I depend on, so this is actually a huge issue for me as most of my devices are battery-powered :-(

### Describe the solution you'd like

I have 60 devices in my Z2M network, and another separate Hue Bridge running just my Hue lights. So in total about 100 Zigbee devices alone... Some of these devices use battery power, and while the Hue app is great for monitoring, I sadly can't say the same for Z2M (please note I said that not as a criticism because it would be unfair, Z2M works great for everything it does).

Some of my battery devices (IKEA PARASOLL DOOR/WINDOW SENSOR) seem to be garbage about reporting battery levels, so I find that with MQTT retention I'm actually seeing the status of a dead sensor a lot of the time.

And it is frustrating because it breaks the alarm system and other automations... So I would like to ask for a special section on Z2M to be created, this section would allow me to select which devices are included for monitoring, then, Z2M would send a command once a day to check for charge, and if the device does not respond it would assume it's dead and report as such, otherwise it would report the charge percentage of each device in an easy-to-use table, almost as if using Excel with Conditional Formatting, so I could see a Red/Amber/Green status for everything very easily and then replace the battery on those devices with low charge before they die completely and cause issues on Home Assistant.

Going even further, if this could then interact with Home Assistant to notify me of which devices have low charge (as a persistent notification on the HA sidebar) that would be amazing! Or, alternatively a custom card that would report the devices on RED category (battery almost dead/must replace battery).

I don't really know how Z2M interacts with the sensors, so I have a question to educate myself and would be grateful for a ELI5 explanation - does Z2M "ask" my sensors, including the PARASOLL contact sensors if they're alive? If so how often in a day? Because if Z2M does not "ask" for an update from the device, then a device with a very low charge may die and Z2M will never know because it didn't poll for a battery level status. For instance, I have a sensor in a garden box that almost never gets opened, and the only way I find this PARASOLL sensor is dead is if I open the box and see that there's no flashing red LED on the sensor (it flashes red when you either open or close, when the battery is providing enough charge). Is there any way to adjust polling, even if manually via config file, and how? Can I do this per device, or for devices obeying a certain rule/filter "MODEL contains PARASOLL"?

Thanks for your consideration, and if someone has seen a solution that already covers my basis I'd be very grateful and happy to be told about alternatives. Thanks!

### Describe alternatives you've considered

I considered just doing automation with regex on Home Assistant but it doesn't really solve the polling issue... I think logic in Z2M could be improved in terms of polling for battery level and how it updates and presents this data. I think it should be more proactive.

### Additional context

Thanks for considering my request. I look forward to an update to know if this is planned or will never happen. I think the battery level monitoring could be improved more, currently it relies only on the device to report but the device may die off from low charge before it ever has a chance to send an update... it seems to affect devices the further away they are from mains-powered devices/routers, which makes sense.

Koenkk, 2024-10-02:
Did you already consider the availability feature? This will mark devices as unavailable (also in HA) when they did not check-in for 25 hours. [docs](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html)

Majestic7979, 2024-10-03:
> Did you already consider the availability feature? This will mark devices as unavailable (also in HA) when they did not check-in for 25 hours. [docs](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html)

But the idea is exactly to avoid that... if they've gone unavailable it's too late, I want to be warned that there's an impending device dropout, so that I can go change the battery for the device before it has a chance to lose all charge, that way some automations won't break - sorry for the lengthy OG post, I can't blame you :)

Majestic7979, 2024-10-05:
I understand. If pinging/polling is not possible, is the rest possible?

Koenkk, 2024-10-06:
What do you mean with "the rest"? The only way to know if a device is gonna die before it actually dies is replacing it when the battery % gets low.

Majestic7979, 2024-10-06:
Never mind, I don't know how to explain again differently, but thank you for your attention on this so far, I hope you have a pleasant week :)

closed 2024-10-06 by Majestic7979

Great-Chart, 2024-10-09:
@Majestic7979 You might find what you are looking for here through Battery State Card via HACS.
https://github.com/maxwroc/battery-state-card
A number of Home Assistant YouTube video's cover this as a useful walkthrough - one such recent one is by smart Home Australia - https://www.youtube.com/watch?v=acpyrqda_Uw

I think what Koenkk was advising is that Z2M is bringing in battery levels but not really it's forte to process such.

The issue with Parasoll reporting battery levels (or not) is another matter that's not yet been fixed (although revisions have been nominated - https://github.com/Koenkk/zigbee2mqtt/discussions/23156).  But I think it got drowned out with the battery draining  issues that these experienced initially.

You'll also find that many coin cell batteries will drop like a stone and may stop working at 30-40%; but using Battery State Card to present all batteries and checking on one's that drop and looking at the long term statistics views for them allows you to build up a picture of which devices and battery types need closer monitoring.

Shout if you get stuck on anything and I can share examples

Majestic7979, 2024-10-09:
> @Majestic7979 You might find what you are looking for here through Battery State Card via HACS. https://github.com/maxwroc/battery-state-card A number of Home Assistant YouTube video's cover this as [...]

Thanks for the comprehensive response, I'm sure the resources will come useful to me after I review them soon, thanks again 🙏