Koenkk/zigbee2mqtt #24178: [New device support]: TS0601 _TZE204_9mjy74mp

glaseca, 2024-10-01:
### Link

https://es.aliexpress.com/item/1005006068601747.html?gatewayAdapt=glo2esp

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c1388273a0c123","nwkAddr":60945,"manufId":4417,"manufName":"_TZE204_9mjy74mp","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�E�.f�E�.\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":73,"modelId":"TS0601","manufacturerName":"_TZE204_9mjy74mp","powerSource":3,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1727820732808}

### Zigbee2MQTT version

1.40.2

### Comments

I have not been able to implement the external definition

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_9mjy74mp',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Paired, but not supported device

monteship, 2024-10-04:
https://github.com/Koenkk/zigbee2mqtt/issues/23206

theundefined, 2024-12-11:
any progress? :) added as external device works fine more or less.

glaseca, 2024-12-12:
> any progress? :) added as external device works fine more or less.

I think it will be added soon https://github.com/Koenkk/zigbee-herdsman-converters/pull/8418

Anyway, recently I found a better adaptation to add controls over the screen brightness from a new external converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [{
        // The model ID from: Device with modelID 'TS0601' is not supported
        // You may need to add \u0000 at the end of the name in some cases
        modelID: 'TS0601',
        // The manufacturer name from: Device with modelID 'TS0601' is not supported.
        manufacturerName: '_TZE204_9mjy74mp',
    }, ],
    model: 'TRV801Z',
    vendor: 'Moes',
    description: 'Moes old TRV/zigbee model',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(),
        e.child_lock(),
        e.max_temperature(),
        e.min_temperature(),
        e.position(),
        e.window_detection(),
        e.binary('window', ea.STATE, 'OPEN', 'CLOSE').withDescription('Window status closed or open '),
        e.binary('alarm_switch', ea.STATE, 'ON', 'OFF').withDescription('Thermostat in error state'),
        e
        .climate()
        .withLocalTemperature(ea.STATE)
        .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
        .withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
        .withPreset(
            ['auto', 'manual', 'off', 'on'],
            'MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
            'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
            'AUTO MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature. ' +
            'ON - In this mode, the thermostat stays open ' +
            'OFF - In this mode, the thermostat stays closed',
        )
        .withSystemMode(['auto', 'heat', 'off'], ea.STATE)
        .withRunningState(['idle', 'heat'], ea.STATE),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
        e.enum('display_brightness', ea.STATE_SET, ['high', 'medium', 'low']).withDescription('Display brightness'),
        e.enum('screen_orientation', ea.STATE_SET, ['up', 'right', 'down', 'left']).withDescription('Screen orientation'),
        e
        .enum('mode', ea.STATE_SET, ['comfort', 'eco'])
        .withDescription(
            'Hysteresis - comfort > switches off/on exactly at reached ' +
            'temperature with valve smooth from 0 to 100%, eco > 0.5 degrees above or below, valve either 0 or 100%',
        ),
    ],
    meta: {
        tuyaDatapoints: [
            [1, null, tuya.valueConverter.thermostatSystemModeAndPreset(null)],
            [1, 'system_mode', tuya.valueConverter.thermostatSystemModeAndPreset('system_mode')],
            [1, 'preset', tuya.valueConverter.thermostatSystemModeAndPreset('preset')],
            [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [6, 'running_state', tuya.valueConverterBasic.lookup({heat: 1, idle: 0})],
            [7, 'window', tuya.valueConverterBasic.lookup({OPEN: 1, CLOSE: 0})],
            [8, 'window_detection', tuya.valueConverter.onOff],
            [12, 'child_lock', tuya.valueConverter.lockUnlock],
            [13, 'battery', tuya.valueConverter.raw],
            [14, 'alarm_switch', tuya.valueConverter.onOff],
            [15, 'min_temperature', tuya.valueConverter.divideBy10],
            [16, 'max_temperature', tuya.valueConverter.divideBy10],
            [17, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [18, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [19, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [20, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [21, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [22, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [23, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [101, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
            [108, 'position', tuya.valueConverter.divideBy10],
            [111, 'display_brightness', tuya.valueConverterBasic.lookup({high: tuya.enum(0), medium: tuya.enum(1), low: tuya.enum(2)})],
            [
                113,
                'screen_orientation',
                tuya.valueConverterBasic.lookup({
                    up: tuya.enum(0),
                    right: tuya.enum(1),
                    down: tuya.enum(2),
                    left: tuya.enum(3),
                }),
            ],
            [114, 'mode', tuya.valueConverterBasic.lookup({comfort: tuya.enum(0), eco: tuya.enum(1)})],
        ],
    },

    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

glaseca, 2025-01-04:
Merged in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8415

closed 2025-01-04 by glaseca