Koenkk/zigbee2mqtt #24181: Won't start after coordinator firmware update (znp)

Ulrar, 2024-10-02:
### What happened?

I've updated my poe coordinator's firmware to the newer version, and Z2M is now failing to start.

With and without debug logs it doesn't really say much, the only error is that mqtt isn't connected, which seems unrelated to the firmware update.
Is that actually the issue ? If so, is there any way to make it log why it's failing to connect ? I didn't touch mqtt at all and I can connect from the Z2M pod using netcat so the network isn't blocking the connection.

With debug enabled it does seem the network is getting setup as expected. I tried re-flashing the firmware again to make sure a second time, same behavior.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20240710

### Adapter

TubeSZB cc2552p poe

### Setup

In kubernetes using an STS

### Debug log

Starting Zigbee2MQTT without watchdog.
[2024-10-02 14:55:56] info: 	z2m: Logging to console, file (filename: log.log)
[2024-10-02 14:55:56] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-02 14:55:56] info: 	z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-02 14:55:56] info: 	zh:zstack:znp: Opening TCP socket with tubeszb-cc2652-zigbee.home.lemonnier.ie:6638
[2024-10-02 14:55:56] info: 	zh:zstack:znp: Socket connected
[2024-10-02 14:55:56] info: 	zh:zstack:znp: Socket ready
[2024-10-02 14:55:56] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-10-02 14:55:57] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-10-02 14:58:24] error: 	z2m: Not connected to MQTT server!
[2024-10-02 14:58:24] error: 	z2m: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: '{"state":"offline"}
[2024-10-02 14:58:24] info: 	z2m: Disconnecting from MQTT server
[2024-10-02 14:58:24] info: 	z2m: Stopping zigbee-herdsman...
[2024-10-02 14:58:25] info: 	z2m: Stopped zigbee-herdsman
[2024-10-02 14:58:25] info: 	z2m: Stopped Zigbee2MQTT

Ulrar, 2024-10-02:
I tried flashing again the previous firmware, which was working fine before, but same issue. I really don't get why mqtt would suddenly be an issue

Ulrar, 2024-10-02:
Also tried downgrading to 1.40.1, in case that update was the issue but no luck. I have other devices connecting to the MQTT broker fine so it's definitely not the issue, there must be an actual error hidden somewhere

chris-1243, 2024-10-02:
> [2024-10-02 15:18:24] �[31merror�[39m: 	z2m: Not connected to MQTT server!

I would definitely check your user/password in `configuration.yaml` and all settings related to MQTT server. Also check the version used

```
    # Optional: MQTT protocol version (default: 4), set this to 5 if you
    # use the 'retention' device specific configuration
    version: 4
```

https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#server-connection

Ulrar, 2024-10-02:
I don't use a username / password, this is in cluster so I just have a network policy.
Both home assistant (also in-cluster with the exact same settings) and external devices are able to connect to the broker, while for Z2M nothing happens, no entries in the broker logs.

I've tried bumping the version to 5 but no change.
Is there a way to get an error out of z2m, is it a timeout or something else ?

chris-1243, 2024-10-02:
MQTT errors are always tricky from Z2M, especially connection to the server.

Would you provide the section in `configuration.yaml` related to this?

Ulrar, 2024-10-02:
```yaml
    mqtt:
      base_topic: zigbee2mqtt
      server: mqtt://mosquitto-service.mosquitto.svc:1883
      version: 5
```

Tried with and without version defined.
When I exec into the pod I'm able to netcat to that address / port with no issues, and it does show up in the mqtt logs when I do so.

Ulrar, 2024-10-02:
So, this is fun.
I copied the /app/data contents to a standalone machine, and started the exact same config pointing to the same broker there, and it worked just fine.

I then took the files from that and moved them back to the pod's PVC and then restarted it : bang, all good.
This makes no sense, but I guess it's fixed, so .. yay ?

chris-1243, 2024-10-02:
This is... interesting... At least it is working.

A file corrupted maybe? Glad you are back online.

Ulrar, 2024-10-03:
Yep, all good since. This will stay a mystery I suppose, thanks for your help !

closed 2024-10-03 by Ulrar