Koenkk/zigbee2mqtt #24182: NOUS DZ4 not recognized

DEFAYArnaud, 2024-10-02:
### What happened?

Hello,

I have just bought the NOUS DZ4 module (https://www.zigbee2mqtt.io/devices/D4Z.html).
But the problem is that it is not recognized and so it is “Not supported”.

[image]

How can I solve the problem please.

Thank you in advance

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.4

### Adapter

SKYCONNECT

### Setup

Home Assistant Green

closed 2024-10-02 by DEFAYArnaud

derspe, 2024-10-04:
Hi @DEFAYArnaud , I'm having the same problem here. How did you solve it. Thx in advance.

DEFAYArnaud, 2024-10-04:
Hi @derspe, here's the solution : [https://github.com/Koenkk/zigbee-herdsman-converters/issues/8064](url)

derspe, 2024-10-04:
Thx so much - made my day