Koenkk/zigbee2mqtt #24183: [New device support]: Müller Licht / tint Malea Ceiling Light

itkama, 2024-10-02:
### Link

https://hey-tint.de/collections/wohnraum-kueche/products/tint-smarte-led-deckenleuchte-malea-gray

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0xa4c13800aa453c98","nwkAddr":49663,"manufId":4635,"manufName":"MLI","powerSource":"Mains (single phase)","modelId":"Ceiling light","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,4111,65376],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"Ceiling light","manufacturerName":"MLI","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MDPANL502"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":555,"currentHue":92,"colorMode":1,"currentX":30295,"currentY":27192,"currentSaturation":254,"colorTemperature":370}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":112}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MDPANL502","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1727882838130}

### Zigbee2MQTT version

1.40.2

### Comments

Need to find out what to actually call the model. All other Müller Licht devices seem to use a 6 digit number there. How do I find this one out?

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Ceiling light'],
    model: 'Malea Ceiling light',
    vendor: 'MLI',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[153,555]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Everything seems to be working pretty much out of the box.

Koenkk, 2024-10-03:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/muller_licht.ts)?

TMan1695, 2024-12-17:
Any update on this @itkama? I have two of these in the mail waiting for Zigbee2MQTT and HA integration :-)

itkama, 2024-12-18:
@TMan1695 Sorry, time flew by, maybe over the holidays. The external converter should work already though. I just didn't have time to look into how to merge it with the existing devices, as they seem to do it a bit differently.

closed 2025-02-25 by github-actions[bot]