Koenkk/zigbee2mqtt #24187: USB adapter discovery error (No valid USB adapter found)

Koenkk, 2024-10-02:
### What happened?

d

### Zigbee2MQTT version

9

### Adapter firmware version

9

### Adapter

-

### Setup

-

### Debug log

-

aivult, 2024-10-03:
> Could this error be related to the functionality under discussion? After adding the `adapter: zstack` option the error remains
> 
> ```
> [2024-10-02 09:02:44] error: 	zh:zstack:znp: Failed to determine if path is valid: 'Error: spawn udevadm ENOENT'
> [2024-10-02 09:02:44] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
> ```
> 
> Debug log: [log.log](https://github.com/user-attachments/files/17241419/log.log)

Have the same problem. Adding adapter line doesn't help

Koenkk, 2024-10-03:
The change has not been merged yet. `Failed to validate path: 'Error: spawn udevadm ENOENT'` means z2m doesn't have udev permissions (for the addon it will be fixed in https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/pull/637)