Koenkk/zigbee2mqtt #24191: Install failing - NPM errors?

umbrelleman, 2024-10-02:
### What happened?

At the 'NPM CI' stage of install instructions, install never completes due to NPM error:

pi@pizigbee4:/opt/zigbee2mqtt $ sudo npm ci
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
⠼
npm error process terminated
npm error signal SIGINT
npm error A complete log of this run can be found in: /root/.npm/_logs/2024-10-02T20_19_34_668Z-debug-0.log

### How to reproduce it (minimal and precise)

Fresh install on brand new Raspberry Pi 4 with NODE v20.17.0 and NPM v10.8.2

### Zigbee2MQTT version

None

### Adapter firmware version

None

### Adapter

None

### Setup

Raspberry Pi 4

### Debug log

[2024-10-02T20_19_34_668Z-debug-0.log.log](https://github.com/user-attachments/files/17235775/2024-10-02T20_19_34_668Z-debug-0.log.log)

lux73, 2024-10-03:
https://github.com/Koenkk/zigbee2mqtt/issues/23238 😉

umbrelleman, 2024-10-03:
Does this mean I cannot actually install Zigbee2MQTT any more!?

lux73, 2024-10-03:
absolutely not - AFAIK you can safely ignore this, this modules are not used neither loaded by z2m

umbrelleman, 2024-10-03:
But the install fails, every time in the install it gets to

npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
⠼

This takes about 5-10 minutes to eventually time out and it says

npm error process terminated
npm error signal SIGINT

The install is not complete and it cannot be run?

Koenkk, 2024-10-03:
Looks like your system runs out of memory

umbrelleman, 2024-10-03:
> Looks like your system runs out of memory, try stopping other processes before executing it.

Unfortunately this is on a brand new Raspberry Pi 4 with a fresh install and nothing else running?

Jozefik, 2024-10-03:
Hi, i'm having similar issue, after running npm ci on Raspberry Pi Zero 2 W after a few minutes ssh timeout and i cannot even ping the rpi. I have tried new sdcard, 32/64bit lite raspberry pi os, even with node v22.x but still nothing. Until the latest update it was working well. It started happening when i tried the update.

WombatHollow, 2024-10-06:
FYI I have just **updated** my system to V1.40.2 and installed OK with only the 2 depreciation warnings listed above by the author of the issue.
Raspi 3B+ running
- "Raspbian GNU/Linux 11 (bullseye)
- Node.js v20.17.0
- npm v10.8.2

Updated using CLI lines
```
cd /opt/zigbee2mqtt
./update.sh
```

closed 2024-12-19 by github-actions[bot]