Koenkk/zigbee2mqtt #24193: Sonoff ZBDongle-E ember adapter fails with ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT

SoulofAkuma, 2024-10-03:
### What happened?

Hi,

I tried to setup a new ZBDongle-E controller using zigbee2mqtt with home assistant and mosquitto today.
I have tried multiple 7.4.x firmware versions (including the most recent one) and all of them caused the below issue for me.
The closest I got to a working zigbee2mqtt instance was this one.

I am a developer and would be happy to help (and contribute) regarding this issue, if that would help.

### What did you expect to happen?

I expected the adapter to work with the latest firmware on my new dongle.

### How to reproduce it (minimal and precise)

 I am using a new ZBDongle-E flashed with the latest firmware on a new Rasberry Pi 5. The ` compose.yaml`  I am using is the following:
```yaml
services:
  homeassistant:
    container_name: homeassistant
    image: "ghcr.io/home-assistant/home-assistant:stable"
    volumes:
      - /home/mallert2/smarthome/homeassistant:/config
      - /etc/localtime:/etc/localtime:ro
      - /run/dbus:/run/dbus:ro
    restart: unless-stopped
    privileged: true
    network_mode: host

  portainer:
    container_name: portainer
    image: portainer/portainer-ce
    restart: always
    ports:
      - "9000:9000/tcp"
    environment:
      - TZ=Europe/Berlin
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
      - /home/mallert2/smarthome/portainer:/data

  mosquitto:
    image: eclipse-mosquitto:2.0
    container_name: mosquitto
    restart: unless-stopped
    volumes:
      - /home/mallert2/smarthome/mosquitto:/mosquitto
      - /home/mallert2/smarthome/mosquitto/data:/mosquitto/data
      - /home/mallert2/smarthome/mosquitto/log:/mosquitto/log
    ports:
      - 1883:1883
      - 9001:9001

  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - ./zigbee2mqtt:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
```

My zigbee2mqtt configuration is the following:
[... 499 lines cut ...]
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=1
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=4
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash: Port closed.
[2024-10-03 02:14:00] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-10-03 02:14:00] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-10-03 02:14:00] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-10-03 02:14:00] error: 	z2m: Adapter disconnected, stopping
[2024-10-03 02:14:00] debug: 	z2m: Saving state to file /app/data/state.json
[2024-10-03 02:14:00] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-10-03 02:14:00] info: 	z2m: Disconnecting from MQTT server
[2024-10-03 02:14:00] info: 	z2m: Stopping zigbee-herdsman...
[2024-10-03 02:14:00] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-10-03 02:14:00] info: 	z2m: Stopped zigbee-herdsman
[2024-10-03 02:14:00] info: 	z2m: Stopped Zigbee2MQTT

jan0o0sch, 2024-10-03:
I have the same problem.
I am using the ZBDONGLE-E with 7.4.4 original firmware. Using HA with Z2M on latest updates, I can't get the dongle to work. It always stops with the same error message, metionend above.

I have round about 100 devices in my network -- most of them are TUYA Zigbee devices. I have dimmers, presence detectors, heating controllers and blind controllers in every room.

Help would be appreciated.
Kind regards ..

Area-me, 2024-10-03:
I have original the Same Problem. 
But i only have 4 devices in z2m 2 motion detectors (tradfri and tuya) and 2 tuya switches . 
I am thankful for help.

AURELB87, 2024-10-04:
Same problem here too.
It's not possible to make an update. It starts but failed after x seconds.

I have also a lot of this message: "zhc: Failed to apply calibration to 'pressure': 'pressure_calibration' is not a number, got string ()"

Scandiravian, 2024-10-23:
Seeing the same error when doing OTA updates, pairing new devices, and if I do several actions rapidly (pressing different buttons, toggling lights through the web interface, etc)

I've tried to eliminate potential causes through trial and error:

- with a single IKEA styrbar switch in the network I was able to do an OTA update if the switch was directly on top of the dongle (took approximately 80 minutes). This could just have been luck
- I could not do an OTA update from a few metres away with only a single device in the network
- with 10+ devices in the network I have not been able to update a different switch even when it was directly on top. It fails after a few seconds, with 2-3 devices it gets to around 40% before failing
- I've tried different combinations of devices in the network to eliminate a single device causing it
- I've moved the dongle to different locations to keep it away from sources of interference, which has no effect 
- the issue seems to become more prevalent the larger the network grows. At 20 devices (6 of them are routers) it takes 4-5 attempts to pair a new device to the network. Distance to dongle/router doesn't seem to matter
- I've tried with both 7.4.4 and 8.0.2 which didn't change the frequency of the error
- Switching channels based on ember-zli scans had no effect. I do however live in an area with a lot of congestion in the 2.4ghz spectrum, so it could be that it's just too much noise on all possible channels
- I didn't notice any difference in the error rate increasing the transmission power from 5 to 20, but I didn't do any metrics on the error rate

I used a deconz adapter until recently, but switched in an attempt to fix another bug (hue lights not updating their state when toggled in home assistant). I still had OTA issues with that, but pairing worked seamlessly

closed 2024-12-31 by github-actions[bot]