Koenkk/zigbee2mqtt #24195: 1.40.2 - Entities showing Unknown not None - Buttons exposing actions

woodmj74, 2024-10-03:
### What happened?

When an entity such as `sensor.[name]_action` stops receiving an action its status is set to Unknown.

### What did you expect to happen?

Previously, this entity when empty contained None. Expected behaviour to remain the same:

[image]
1.40.2 installed at 08:50 and difference in behaviour seen

### How to reproduce it (minimal and precise)

- Press right button
- MQTT action "right"
- entity state "right"
- Release button
- MQTT action "" 
- entity state "unknown"

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

HA add on

Skynetnexus, 2024-10-07:
Same issue, button action, all brands became unavailable after installing core 2024.10.01 reverted using "ha core update --version 2024.9.3" and back to normal.

alexsaas, 2024-10-21:
Same here since Update HA from 2024.9.3 to 2024.10.3.

woodmj74, 2024-10-21:
Okay thanks for the update @Koenkk 

Closing based on feedback.

closed 2024-10-21 by woodmj74

Skynetnexus, 2024-10-21:
> Seems to be some regression from HA, anyway, even if this can be fixed from Z2M we won't as the `action` sensor is going to be removed in Z2M 2.0.0 (#24198)

The Action Sensor is going? I have 13 buttons that I trigger off the action sensor, I will have to start using sensor.button_click instead of sensor.button_action but 2 of my buttons do not expose the click sensor ( Hue smart button )

Koenkk, 2024-10-22:
The click sensor is also deprecated, use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended)

alexsaas, 2024-10-22:
> The click sensor is also deprecated, use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended)

Hi, what about the new event entity you are discussing here:

https://github.com/Koenkk/zigbee2mqtt/discussions/24198#discussioncomment-10860861

will it be integrated into zigbee2MQTT? I think this seems to be a very good alternative.

Skynetnexus, 2024-10-23:
> The click sensor is also deprecated, use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended)

Being an average user, I have scratched my head over this and finally work out what is happening, in layman terms:-  
Buttons sensors like sensor.mybutton_action and sensor.mybutton_click will not be created by Z2M and available in HA the future.

The recommended solution mentioned is to use **MQTT device triggers**, but this looks like a lot of work as you need to find the device ID etc. 

Eventually I found this post ( https://github.com/Koenkk/zigbee2mqtt/issues/3033)  and it clicked and became easy. **Don't write the automation in yaml**, **"use the Home Assistant automation editor to respond to this event."**

P.S. in my travels, as a back up, I worked out that this trigger works:-
 
triggers:
  - topic: zigbee2mqtt/Doorbell/click
    trigger: mqtt
    payload: single
  - topic: zigbee2mqtt/Doorbell/click
    trigger: mqtt
    payload: double
  - topic: zigbee2mqtt/Doorbell/click
    trigger: mqtt
    payload: triple

P.P.S.  @ Koenkk  A huge thank you for all you work on zigbee2MQTT and Home Assistant and thank you for chiming in on this topic.