Koenkk/zigbee2mqtt #24197: [New device support]: Radiator valve TS0601 by _TZE284_ymldrmzx

sveatlo, 2024-10-03:
### Link

https://www.aliexpress.com/item/1005007725276755.html

### Database entry

{"id":30,"type":"EndDevice","ieeeAddr":"0xa4c13898e286f604","nwkAddr":56368,"manufId":4417,"manufName":"_TZE284_ymldrmzx","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"Y\u001a�.iZ\u001a�.i[\u001a�.f\\\u001a�.\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_ymldrmzx","powerSource":3,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1727939018397}

### Zigbee2MQTT version

1.40.2

### Comments

No features were automatically discovered.
Using the definition from Tuya's TRV601 (_TZE204_rtrmfadk) or TRV602 doesn't work - no reported value is correct and changing the settings is not reflected on the device.

I started playing with this and successfully got few of the settings to work correctly but I do not have a tuya zigbee hub, so I cannot easily find all the datapoints

These are all the datapoints sent after the devices boots:

```

[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 39 not defined for '_TZE284_ymldrmzx' with value true
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 101 not defined for '_TZE284_ymldrmzx' with value false
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 102 not defined for '_TZE284_ymldrmzx' with value 0
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 111 not defined for '_TZE284_ymldrmzx' with value 0
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 112 not defined for '_TZE284_ymldrmzx' with value 50
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 113 not defined for '_TZE284_ymldrmzx' with value false
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 114 not defined for '_TZE284_ymldrmzx' with value 0
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 115 not defined for '_TZE284_ymldrmzx' with value 0
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 116 not defined for '_TZE284_ymldrmzx' with value 170
[2024-10-03 11:10:25] debug: 	zhc:tuya: Datapoint 117 not defined for '_TZE284_ymldrmzx' with value 230
[2024-10-03 11:10:26] debug: 	zhc:tuya: Datapoint 103 not defined for '_TZE284_ymldrmzx' with value ?*Sdx���*�
[2024-10-03 11:10:26] debug: 	zhc:tuya: Datapoint 104 not defined for '_TZE284_ymldrmzx' with value ?*Sdx���*�
[2024-10-03 11:10:26] debug: 	zhc:tuya: Datapoint 105 not defined for '_TZE284_ymldrmzx' with value ?*Sdx���*�
[2024-10-03 11:10:26] debug: 	zhc:tuya: Datapoint 106 not defined for '_TZE284_ymldrmzx' with value ?*Sdx���*�
[2024-10-03 11:10:27] debug: 	zhc:tuya: Datapoint 107 not defined for '_TZE284_ymldrmzx' with value ?*Sdx���*�
[2024-10-03 11:10:27] debug: 	zhc:tuya: Datapoint 108 not defined for '_TZE284_ymldrmzx' with value I*P*dx���*�
[2024-10-03 11:10:27] debug: 	zhc:tuya: Datapoint 109 not defined for '_TZE284_ymldrmzx' with value I*P*dx���*�
[2024-10-03 11:10:27] debug: 	zhc:tuya: Datapoint 110 not defined for '_TZE284_ymldrmzx' with value c�\�gt
[2024-10-03 11:10:28] debug: 	zhc:tuya: Datapoint 118 not defined for '_TZE284_ymldrmzx' with value 0
[2024-10-03 11:16:00] debug: 	zhc:tuya: Datapoint 6 not defined for '_TZE284_ymldrmzx' with value 100
```

### External definition

```shell
const definition = {
  fingerprint: [{ modelID: "TS0601", manufacturerName: "_TZE284_ymldrmzx" }],
  model: "TRV603",
  vendor: "Tuya",
  description: "",
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
  configure: tuya.configureMagicPacket,
  exposes: [
    e
      .climate()
      .withLocalTemperature(ea.STATE)
      .withSetpoint("current_heating_setpoint", 5, 35, 0.5, ea.STATE_SET)
      .withLocalTemperatureCalibration(-10, 10, 0.5, ea.STATE_SET)
      .withPreset(["schedule", "holiday", "manual", "comfort", "eco"])
      .withSystemMode(["off", "heat"], ea.STATE),
    ...tuya.exposes.scheduleAllDays(
      ea.STATE_SET,
      "HH:MM/C HH:MM/C HH:MM/C HH:MM/C",
    ),
    e
      .binary("frost_protection", ea.STATE_SET, "ON", "OFF")
      .withDescription(
        "When the room temperature is lower than 5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.",
      ),

    e
      .enum("screen_orientation", ea.STATE_SET, ["up", "right", "down", "left"])
      .withDescription("Screen orientation"),
  ],
  meta: {
    tuyaDatapoints: [
      // preset: 2 - manual, 0 - auto, 1 - ??, ... - ??
      [
        2,
        "preset",
        tuya.valueConverterBasic.lookup({
          schedule: tuya.enum(0),
          manual: tuya.enum(2),
        }),
      ],
      [4, "current_heating_setpoint", tuya.valueConverter.divideBy10],
      [5, "local_temperature", tuya.valueConverter.divideBy10],
      // 36 - some booleany value. can be configured under P6, hint is "Fr". maybe frost protection?
      [36, "frost_protection", tuya.valueConverter.onOff],
      [
        49,
        "system_mode",
        tuya.valueConverterBasic.lookup({
          off: tuya.enum(0),
          heat: tuya.enum(1),
        }),
      ],
      [
        103,
        "schedule_monday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1),
      ],
      [
        104,
        "schedule_tuesday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2),
      ],
      [
        105,
        "schedule_wednesday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3),
      ],
      [
        106,
        "schedule_thursday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4),
      ],
      [
        107,
        "schedule_friday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5),
      ],
      [
        108,
        "schedule_saturday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6),
      ],
      [
        109,
        "schedule_sunday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7),
      ],
      [
        111,
        "screen_orientation",
        tuya.valueConverterBasic.lookup({
          up: tuya.enum(0),
          right: tuya.enum(1),
          down: tuya.enum(2),
          left: tuya.enum(3),
        }),
      ],
      [
        114,
        "local_temperature_calibration",
        tuya.valueConverter.localTempCalibration3,
      ],
    ],
  },
};
```

### What does/doesn't work with the external definition?

Works:

- setting heating setpoint
- setting auto/manual preset
- screen orientation
- local temperature calibration

Doesn't work:

- current status - heat/idle
- position
- battery
- window detection
- eco mode
- child lock
- schedule - I'm pretty sure 103-109 datapoints are daily schedules but the values are nonsensical when decoded (e.g. `16:63/1083.5 20:100/524.0 20:140/528.0 20:173/1097.2`)

persicocd, 2024-11-25:
Hey Sveatlo,

Thanks for this, have the same thermostat valve and was trying to figure out how to get it working with z2mqtt.

Saved your external definition as js and added it, etc., though it isn't getting paired to the device, and the device is remaining unsupported. Is your definition 1:1 what's in your .js file, or is there anything additional in there, either before or after what you posted above?

Any help is appreciated, thanks.

ESP32the2nd, 2024-12-06:
I got it to work.  I used the code from sveatlo but added the const definitions from the documentation and the Tuya devices. This is the entire content of my .js file:

```javascript

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
  fingerprint: [{ modelID: "TS0601", manufacturerName: "_TZE284_ymldrmzx" }],
  model: "TRV603",
  vendor: "Tuya",
  description: "",
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
  configure: tuya.configureMagicPacket,
  exposes: [
    e
      .climate()
      .withLocalTemperature(ea.STATE)
      .withSetpoint("current_heating_setpoint", 5, 35, 0.5, ea.STATE_SET)
      .withLocalTemperatureCalibration(-10, 10, 0.5, ea.STATE_SET)
      .withPreset(["schedule", "holiday", "manual", "comfort", "eco"])
      .withSystemMode(["off", "heat"], ea.STATE),
    ...tuya.exposes.scheduleAllDays(
      ea.STATE_SET,
      "HH:MM/C HH:MM/C HH:MM/C HH:MM/C",
    ),
    e
      .binary("frost_protection", ea.STATE_SET, "ON", "OFF")
      .withDescription(
        "When the room temperature is lower than 5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.",
      ),

    e
      .enum("screen_orientation", ea.STATE_SET, ["up", "right", "down", "left"])
      .withDescription("Screen orientation"),
  ],
  meta: {
    tuyaDatapoints: [
      // preset: 2 - manual, 0 - auto, 1 - ??, ... - ??
      [
        2,
        "preset",
        tuya.valueConverterBasic.lookup({
          schedule: tuya.enum(0),
          manual: tuya.enum(2),
        }),
      ],
      [4, "current_heating_setpoint", tuya.valueConverter.divideBy10],
      [5, "local_temperature", tuya.valueConverter.divideBy10],
      // 36 - some booleany value. can be configured under P6, hint is "Fr". maybe frost protection?
      [36, "frost_protection", tuya.valueConverter.onOff],
      [
        49,
        "system_mode",
        tuya.valueConverterBasic.lookup({
          off: tuya.enum(0),
          heat: tuya.enum(1),
        }),
      ],
      [
        103,
        "schedule_monday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1),
      ],
      [
        104,
        "schedule_tuesday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2),
      ],
      [
        105,
        "schedule_wednesday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3),
      ],
      [
        106,
        "schedule_thursday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4),
      ],
      [
        107,
        "schedule_friday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5),
      ],
      [
        108,
        "schedule_saturday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6),
      ],
      [
        109,
        "schedule_sunday",
        tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7),
      ],
      [
        111,
        "screen_orientation",
        tuya.valueConverterBasic.lookup({
          up: tuya.enum(0),
          right: tuya.enum(1),
          down: tuya.enum(2),
          left: tuya.enum(3),
        }),
      ],
      [
        114,
        "local_temperature_calibration",
        tuya.valueConverter.localTempCalibration3,
      ],
    ],
  },
};
module.exports = definition;
``` 

Regarding the scheduled mode, I believe the settings are working. Unfortunately the setpoint temperature value goes to 0° if preset mode was switched to "schedule".

[image]

Would be great if somebody could fix the non-working part.

closed 2025-02-13 by github-actions[bot]