Koenkk/zigbee2mqtt #24202: Error: Interview failed because can not get node descriptor

sridhar2rok, 2024-10-03:
### What happened?

I am currently using the latest version of Zigbee2MQTT along with an ESP32-C6 end device running the HA_ON_OFF_light example from Espressif. However, Zigbee2MQTT repeatedly attempts to interview the device without success, and it fails with the error shown below. The same end device works perfectly with the Sonoff Zigbee adapter.
(Note: The ConBee 3 is able to detect other standard Zigbee devices without issues.)

### What did you expect to happen?

It works well with other adapter, but not conbee 3

### How to reproduce it (minimal and precise)

- flash the esp32c6 development board with HA_ON_OFF_light example from expressif.
- use conbee 3 as adapter

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

deCONZ_ConBeeIII_0x26530900.bin.GCF

### Adapter

Conbee 3 stick

### Setup

on docker

### Debug log

10-03 22:26:47 [2024-10-03 22:26:47] debug: zh:deconz: sendZclFrameToAll - message send
2024-10-03 22:26:47 [2024-10-03 22:26:47] debug: zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 131 request id: 12
2024-10-03 22:26:47 [2024-10-03 22:26:47] debug: zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 12 status: 0
2024-10-03 22:26:47 [2024-10-03 22:26:47] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:frameparser: DEVICE_STATE changed: 100110
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:driver: query aps data confirm
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 132
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfffd request id: 12 confirm status: 0
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:driver: send write parameter request from queue. seqNr: 133 paramId: 33 param: 0
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:frameparser: write parameter response - parameter id: 33 - status: 0
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz: Permit joining on coordinator for 0 sec.
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 134 request id: 13
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 13 status: 0
2024-10-03 22:26:48 [2024-10-03 22:26:48] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:26:49 [2024-10-03 22:26:49] debug: zh:deconz:frameparser: DEVICE_STATE changed: 100110
2024-10-03 22:26:49 [2024-10-03 22:26:49] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:26:49 [2024-10-03 22:26:49] debug: zh:deconz:driver: query aps data confirm
2024-10-03 22:26:49 [2024-10-03 22:26:49] debug: zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 135
2024-10-03 22:26:49 [2024-10-03 22:26:49] debug: zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 13 confirm status: 0
2024-10-03 22:26:49 [2024-10-03 22:26:49] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:26:54 [2024-10-03 22:26:54] debug: z2m: Saving state to file /app/data/state.json
2024-10-03 22:26:54 [2024-10-03 22:26:54] debug: zh:deconz:driver: send read device state from queue. seqNr: 136
2024-10-03 22:26:54 [2024-10-03 22:26:54] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:26:54 [2024-10-03 22:26:54] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:27:04 [2024-10-03 22:27:04] debug: zh:deconz:driver: send read device state from queue. seqNr: 137
2024-10-03 22:27:04 [2024-10-03 22:27:04] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:27:04 [2024-10-03 22:27:04] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:27:08 [2024-10-03 22:27:08] debug: zh:deconz:driver: Timeout for aps request CMD: 0x12 seq: 129
2024-10-03 22:27:14 [2024-10-03 22:27:14] debug: zh:deconz:driver: send read device state from queue. seqNr: 138
2024-10-03 22:27:14 [2024-10-03 22:27:14] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:27:14 [2024-10-03 22:27:14] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:27:24 [2024-10-03 22:27:24] debug: zh:deconz:driver: send read device state from queue. seqNr: 139
2024-10-03 22:27:24 [2024-10-03 22:27:24] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:27:24 [2024-10-03 22:27:24] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:27:34 [2024-10-03 22:27:34] debug: zh:deconz:driver: send read device state from queue. seqNr: 140
2024-10-03 22:27:34 [2024-10-03 22:27:34] debug: zh:deconz:frameparser: device state: 100010
[... 19 lines cut ...]
2024-10-03 22:28:04 [2024-10-03 22:28:04] debug: zh:deconz:driver: send read device state from queue. seqNr: 145
2024-10-03 22:28:04 [2024-10-03 22:28:04] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:28:04 [2024-10-03 22:28:04] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:28:14 [2024-10-03 22:28:14] debug: zh:deconz:driver: send read device state from queue. seqNr: 146
2024-10-03 22:28:14 [2024-10-03 22:28:14] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:28:14 [2024-10-03 22:28:14] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:28:24 [2024-10-03 22:28:24] debug: zh:deconz:driver: send read device state from queue. seqNr: 147
2024-10-03 22:28:24 [2024-10-03 22:28:24] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:28:24 [2024-10-03 22:28:24] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:28:34 [2024-10-03 22:28:34] debug: zh:deconz:driver: send read device state from queue. seqNr: 148
2024-10-03 22:28:34 [2024-10-03 22:28:34] debug: zh:deconz:frameparser: device state: 100010
2024-10-03 22:28:34 [2024-10-03 22:28:34] debug: zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-10-03 22:28:39 [2024-10-03 22:28:39] debug: zh:controller:device: Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2024-10-03 22:28:39 [2024-10-03 22:28:39] debug: zh:controller:device: Interview - quirks did not match
2024-10-03 22:28:39 [2024-10-03 22:28:39] debug: zh:controller:device: Interview - node descriptor request failed for '0x543204fffe014400', attempt 6
2024-10-03 22:28:39 [2024-10-03 22:28:39] debug: zh:controller:device: Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2024-10-03 22:28:39 [2024-10-03 22:28:39] debug: zh:controller:device: Interview - quirks did not match
2024-10-03 22:28:39 [2024-10-03 22:28:39] debug: zh:controller:device: Interview - failed for device '0x543204fffe014400' with error 'Error: Interview failed because can not get node descriptor ('0x543204fffe014400')'
2024-10-03 22:28:39 [2024-10-03 22:28:39] debug: zh:controller:database: Writing database to '/app/data/database.db'
2024-10-03 22:28:39 [2024-10-03 22:28:39] error: zh:controller: Interview failed for '0x543204fffe014400 with error 'Error: Interview failed because can not get node descriptor ('0x543204fffe014400')'
2024-10-03 22:28:39 [2024-10-03 22:28:39] error: z2m: Failed to interview '0x543204fffe014400', device has not successfully been paired
[... cut ...] dly_name":"0x543204fffe014400"},"type":"pairing"}'
2024-10-03 22:28:39 [2024-10-03 22:28:39] info: z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x543204fffe014400","ieee_address":"0x543204fffe014400","status":"failed"},"type":"device_interview"}'
2024-10-03 22:28:44 [2024-10-03 22:28:44] debug: zh:deconz:driver: send read device state from queue. seqNr: 149
2024-10-03 22:28:44 [2024-10-03 22:28:44] debug: zh:deconz:frameparser: device state: 100010

877dev, 2024-10-06:
Same here with two new Hue lamps, latest version and other lamps of exact same model and out the same box are not recognised...

xamadevp, 2024-10-09:
Same issue here when pairing [Osram Plug](https://www.zigbee2mqtt.io/devices/AB3257001NJ.html) with Conbee 3. The device is added but the interview fails.

closed 2024-12-19 by github-actions[bot]

citelao, 2025-01-22:
For the ESP32-C6, specifically, I started hitting this after reflashing my C6 a couple of times. The devices would "appear" (sometimes with outdated metadata) but fail to interview. I think there are 2 issues here: ① ESP32's flash memory is not fully erased when flashed with new firmware and ② Z2M has some sort of cache.

I was able to get connection by doing some combination of:

* Restart the Zigbee radio used by Z2M (in my case, an SLZB-06)
* Restart Z2M
* In VS Code: `ESP-IDF: Erase Flash Memory From Device` (**this was the big one**)
* Re-deploy my code to the ESP32.

Now I am able to get my devices to reliably rejoin (after being deleted) by simply **erasing flash memory on the device**. If I make "major" changes (like changing manufacturer name), I _also_ must **restart Z2M**.  I hope that helps!