Koenkk/zigbee2mqtt #24204: Group is out of sync when changing states quickly

szymonkozak, 2024-10-03:
### What happened?

Group state exposed to Home Assistant is out of sync when changing some group devices states quickly. It's displayed as disabled even if one of the devices inside group are enabled. 

### What did you expect to happen?

Group state should be synced

### How to reproduce it (minimal and precise)

Add 2 devices to group (or one device with 2 endpoints) and then change their states randomly quickly. Group switch that is exposed to HA is out of sync and is displayed as disabled even when one of these devices is enabled. This is 100% reproducible every time.

Logs:
```
info 2024-10-03 23:53:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"OFF"}'
info 2024-10-03 23:53:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"ON"}'
info 2024-10-03 23:53:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"OFF"}'
info 2024-10-03 23:53:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"ON"}'
info 2024-10-03 23:53:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"OFF"}'
info 2024-10-03 23:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"ON"}'
info 2024-10-03 23:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"OFF"}'
info 2024-10-03 23:53:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"ON"}'
info 2024-10-03 23:53:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"OFF","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":102,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"ON","switch_type":"toggle"}'
info 2024-10-03 23:53:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"OFF"}'
info 2024-10-03 23:53:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":109,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"ON"}'
info 2024-10-03 23:53:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"OFF","switch_type":"toggle"}'
info 2024-10-03 23:53:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/new_all_zigbee_lights', payload '{"state":"OFF"}'
info 2024-10-03 23:53:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living room right switch', payload '{"color_mode":"xy","linkquality":105,"power_on_behavior":null,"power_on_behavior_l1":"previous","state_l1":"ON","state_l2":"OFF","switch_type":"toggle"}'
```

As you can see there is a group named `new_all_zigbee_lights` and there is Living room right switch with L1 and L2 buttons. At the end one of the buttons (L2) is OFF and one of them (L1) is ON, but the latest state of `new_all_zigbee_lights` is OFF (although it should be ON, no message with that information is delivered).

Devices and group properties are all default. Devices have only friendly name property. Switch device used to test is [TS0002_basic](https://www.zigbee2mqtt.io/devices/TS0002_basic.html#tuya-ts0002_basic) and works without any problems (it's very responsive).

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

Slaesh CC2652RB

### Setup

Add-on on Home assistant on RPi4

szymonkozak, 2024-12-11:
> This issue is stale because it has been open 180 days with no activity

It's not true.

mundschenk-at, 2025-01-18:
Have you checked whether it happens with a group of unrelated devices or only when two endpoints of the same device are part of the group? I've seen this as well, but only in groups that have at least two endpoints of the same device. The group state in HA is the state of the endpoint that last changed, without consideration of the state of the other endpoints in the group.

szymonkozak, 2025-01-28:
@mundschenk-at Looks like it doesn't matter in my case. I can have 2 different physical devices and manupulate them this way that after that the group state (where these 2 devices are added) is out of sync. However I am still use zigbee 1.40.1 because I didn't have time to update yet.

szymonkozak, 2025-03-31:
@mundschenk-at There is no such problem on my side on the latest z2m version.