Koenkk/zigbee2mqtt #24205: Lost Zigbee Config

ALPF5555, 2024-10-04:
### What happened?

I had a Blackout at my house 2 days ago and when the power came back on after 10 minutes none of the Zigbee devices would work.  I found out that the config file had reset back to the Default where the Server is mqtt://localhost and the location of the serial device is port: /dev/ttyACM0.    These are not the correct settings and not the settings I had saved 8 weeks ago when everything was working fine up to 2 days ago.  My saved settings should have been Server mqtt://192.168.1.10:1883 and the serial device should be port: tcp://192.168.1.147:6638  but when I try to change the settings and save the yaml file the changes default back to the original settings. 

### What did you expect to happen?

I would have expected the settings that I saved 8 weeks ago to be the setting that should be loaded and the Zigbee2mqtt to work with the settings that I had saved. 
I would not expect it to go back to the default settings and if it did then I would expect to be able to make the changes and save the file so that it will use the new settings not always default back to your original settings 

### How to reproduce it (minimal and precise)

# Home Assistant integration (MQTT discovery)
homeassistant: false

# Enable the frontend, runs on port 8080 by default
frontend: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

# Advanced settings
advanced:
  # Let Zigbee2MQTT generate a network key on first start
  network_key: GENERATE
  # Let Zigbee2MQTT generate a pan_id on first start
  pan_id: GENERATE
  # Let Zigbee2MQTT generate a ext_pan_id on first start
  ext_pan_id: GENERATE

This is the default file that I always get when I  try to save the settings 

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

Adapter Firmware2.5.6  Zigbee Coordinator Firmware 20240710  

### Adapter

slzb-06 POE 

### Setup

On Raspberry Pi 4B with 8GB 

### Debug log

[2024-10-04 03:40:30] info:     z2m: Logging to console, file (filename: log.log)
[2024-10-04 03:40:30] info:     z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
[2024-10-04 03:40:30] info:     z2m: Starting zigbee-herdsman (0.50.1)
[2024-10-04 03:40:31] info:     zh:zstack:znp: Opening TCP socket with 192.168.1.147:6638
[2024-10-04 03:40:31] info:     zh:zstack:znp: Socket connected
[2024-10-04 03:40:31] info:     zh:zstack:znp: Socket ready
[2024-10-04 03:40:31] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-10-04 03:40:32] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: - PAN ID: configured=12243, adapter=42971
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: - Extended PAN ID: configured=c6b33f73a0b0a3e0, adapter=37ef622e273bbdbf
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: - Network Key: configured=e94d6139fdfe1c3068949c86f1620301, adapter=1a28221d98b47cfa3150191aa94dee8b
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: - Channel List: configured=11, adapter=11
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: Please update configuration to prevent further issues.
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /opt/zigbee2mqtt/data/coordinator_backup.json.
[2024-10-04 03:40:32] error:    zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
[2024-10-04 03:40:32] error:    z2m: Error while starting zigbee-herdsman
[2024-10-04 03:40:32] error:    z2m: Failed to start zigbee
[2024-10-04 03:40:32] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-10-04 03:40:32] error:    z2m: Exiting...
[2024-10-04 03:40:32] error:    z2m: Error: startup failed - configuration-adapter mismatch - see logs above for more information

Nerivec, 2024-10-05:
Is it running on baremetal? I know some virtualization software will automatically generate a config file for z2m unless explicitly turned off.
Any chance you have a backup/restore app/script that could be messing with the file that is passed to z2m?

ALPF5555, 2024-10-07:
Hello Sir              Thank you for your prompt reply I have managed to get the system up and running and now have had time to look in the log file on my raspberry Pi.  I am using Bookworm as a Raspberry Pi operating system. I also found out that updating the SLZB-06P coordinator and the firmware for the Zigbee2mqtt made the system work. I looked in the backup and can find 4 files that are the configuration.yaml file then a configuration.yaml.saved file and a configutation.yaml.saved1 file file and a configuration.example.yaml file.
The configuration.yaml files appears to be correct at least the IP address and the coordinator are correct so my question is should I delete the other configuration files as it appears that maybe when it boots up after a power loss it tries to use a different configuration file then the one it should do.
Best Regards 
Alan Turnbull

Koenkk, 2024-10-07:
Z2M always uses the configuration.yaml, deleting the other files shouldnt be necessary (I have no clue where those come from)

ALPF5555, 2024-10-08:
Hello Sir               Thank you for your message I will close out the GitHub message 
Best regards
Alan Turnbull

closed 2024-10-08 by ALPF5555